project-chip/connectedhomeip #8445: Unable to flash nrf rcp image

ramakrishnanandyala, 2021-07-15:
Following below doc to flash nrf
 https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/openthread_rcp_nrf_dongle.md

I see below error while flashing

ramotbrrcp@ramotbrrcp-desktop:~/ot-nrf528xx$ nrfutil dfu usb-serial -pkg build/bin/ot-rcp.zip -p /dev/ttyACM0

Traceback (most recent call last):
  File "/home/ramotbrrcp/.local/bin/nrfutil", line 8, in <module>
    sys.exit(cli())
  File "/usr/lib/python3/dist-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/ramotbrrcp/.local/lib/python3.8/site-packages/nordicsemi/__main__.py", line 1014, in usb_serial
    do_serial(package, port, connect_delay, flow_control, packet_receipt_notification, baud_rate, serial_number, False,
  File "/home/ramotbrrcp/.local/lib/python3.8/site-packages/nordicsemi/__main__.py", line 970, in do_serial
    dfu.dfu_send_images()
  File "/home/ramotbrrcp/.local/lib/python3.8/site-packages/nordicsemi/dfu/dfu.py", line 127, in dfu_send_images
    self._dfu_send_image(self.manifest.application)
  File "/home/ramotbrrcp/.local/lib/python3.8/site-packages/nordicsemi/dfu/dfu.py", line 88, in _dfu_send_image
    self.dfu_transport.open()
  File "/home/ramotbrrcp/.local/lib/python3.8/site-packages/nordicsemi/dfu/dfu_transport_serial.py", line 217, in open
    self.__get_mtu()
  File "/home/ramotbrrcp/.local/lib/python3.8/site-packages/nordicsemi/dfu/dfu_transport_serial.py", line 366, in __get_mtu
    self.mtu = struct.unpack('<H', bytearray(response))[0]
TypeError: cannot convert 'NoneType' object to bytearray

ramakrishnanandyala, 2021-07-15:
@Damian-Nordic Can you please look in to this error

ramakrishnanandyala, 2021-07-16:
+ @mspang

Damian-Nordic, 2021-07-16:
Looks like the script had some issues with getting MTU from the device. Are you sure `/dev/ttyACM0` is your nRF52840 Dongle and that it's in the bootloader mode (the red LED is blinking)? Also, could you paste `nrfutil --version`?

ramakrishnanandyala, 2021-07-16:
Yes, /dev/ttyACM0 is my device but it is not blinking red, though I pressed reset button

otbrrcp@otbrrcp-desktop:~/ot-nrf528xx$ nrfutil version
nrfutil version 6.1.0

Damian-Nordic, 2021-07-16:
Are you sure you're pressing the reset button? It's a bit hidden. See the picture:
https://github.com/project-chip/connectedhomeip/blob/master/examples/platform/nrfconnect/doc/images/nRF52840_Dongle-medium.jpg

ramakrishnanandyala, 2021-07-16:
I'm using this nrf board 
https://github.com/project-chip/connectedhomeip/blob/master/examples/platform/nrfconnect/doc/images/nRF5340_DK_info-medium.jpg

usb is connected to segger j-link usb connecter and Pressing the reset button as mentioned in pic

Damian-Nordic, 2021-07-16:
Ok, so the instruction you're following is strictly for nRF52840 Dongle. In fact, the script for building the RCP firmware does not support nRF5340 DK yet: https://github.com/openthread/ot-nrf528xx/blob/main/script/build. Do you have access to nRF52840 DK nRF52840 Dongle? nRF5340 is a more powerful SoC and is rarely used for the RCP firmware, but if you don't have access to the nRF52840 I can try to help use it for the RCP as well.

ramakrishnanandyala, 2021-07-16:
I have my nRF52840 DK board with me now

Damian-Nordic, 2021-07-16:
Great, then please follow this instruction: https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/ug_thread_tools.html#configuring-a-radio-co-processor (use "nRF52840 Development kit (UART transport" variant).

ramakrishnanandyala, 2021-07-16:
Getting this error 
otbrrcp@otbrrcp-desktop:~/mychip-rcp/ot-nrf528xx$ nrfjprog -f nrf52 --chiperase --program build/bin/ot-rcp.hex --reset
-bash: /usr/bin/nrfjprog: cannot execute binary file: Exec format error

Damian-Nordic, 2021-07-16:
Have you installed `nRF Command Line Tools`: https://www.nordicsemi.com/Products/Development-tools/nRF-Command-Line-Tools/Download? Looks like `nrfjprog` has not been installed properly. Can you reinstall it using binaries for your operating system (Linux 64?)?

ramakrishnanandyala, 2021-07-16:
Can I install it on raspberry pi-4?

Damian-Nordic, 2021-07-16:
Oh, I guess that might be an issue. I'm afraid packages of nrfjprog for ARM are not available. But you can flash the RCP firmware on your laptop and then plug the DK in your RPi.

LuDuda, 2022-02-13:
I think we can close this issue - the wrong DK was used for particular instruction. `nrfjprog` is not supported on RaspberryPi. Please re-open if we need more discussion @ramakrishnanandyala.

closed 2022-02-13 by LuDuda