project-chip/connectedhomeip #8452: ZAP with UI is not running through ./scripts/tools/zap/configure.sh

PSONALl, 2021-07-16:
#### Problem
ZAP with UI is not running through ./scripts/tools/zap/configure.sh
 * expected behavior: Run ZAP with UI to configure endpoints and clusters
 * actual behavior: After running ./scripts/tools/zap/configure.sh getting error:
 ```
[Error: ENOENT: no such file or directory, open 'spa/hash.json'] {
  errno: -2,
  code: 'ENOENT',
  syscall: 'open',
  path: 'spa/hash.json'
}
```

 * system configuration: macOS BigSur v11.1
 * I have installed all the dependencies given in [instructions](https://github.com/project-chip/zap/blob/master/docs/instructions.md) of zap and zap [FAQ's](https://github.com/project-chip/zap/blob/master/docs/faq.md)

Damian-Nordic, 2021-07-22:
Hi, note that the script has been replaced with `./scipts/tools/zap/run_zaptool.sh` which additionally also verifies whether you have all node modules downloaded. Can you try to use it and see if that helps?

PSONALl, 2021-07-26:
> `./scipts/tools/zap/run_zaptool.sh` 
Thanks @Damian-Nordic, this helped.

closed 2021-07-26 by PSONALl