project-chip/connectedhomeip #8487: Figure out whether we can use micro-ecc as a backend for our CryptoPal

bzbarsky-apple, 2021-07-19:
#### Problem
We don't have a micro-ecc backend for our crypto PAL

 #### Proposed Solution
Maybe we should.

bzbarsky-apple, 2021-07-19:
I should note that micro-ecc does not provide all the bits our crypto PAL needs (e.g. does not include hashing), so we would need to see what we do with that.

tcarmelveilleux, 2021-07-20:
Yes, micro-ecc is not enough. You need another library such as libsodium or tinycrypt that provides the missing primitives

emargolis, 2021-07-20:
`micro-ecc` is sufficient to implement all the ECC functions in CryptoPal.
Other crypto primitives can be taken from libraries like `libsodium` or `tinycrypt` (which by the way uses micro-ecc).

The only component for which we may need a custom implementation is the CSRReq.

 from Intel is also using `micro-ecc` and adds AES-CCM, HMAC and SHA256

bzbarsky-apple, 2021-07-23:
> and does not allow alternate choice (e.g. "custom back-end, irrespective of library used")

Maybe we should just focus on this part, then.

And yes, I did also see some worries about how micro-ecc handles side-channel attacks....