project-chip/connectedhomeip #8494: resolve 0 558 :failed to update node address:4072

736676938, 2021-07-20:
ESP32+MAC

`zx@ubuntu:~/connectedhomeip$ git log
commit a8ef4197acd4e879141077820abcf5d9bc76535b (HEAD, tag: test_event_3_2021_06_10, origin/test_event_3.1)
Author: doru91 <doru-cristian.gucea@nxp.com>
Date:   Wed Jun 9 22:55:16 2021 +0300

    [K32W] Increase IM timeout (#7490)
    
    * [K32W] Increase IM timeout
    
    On K32W, chip-device-ctrl times out while waiting for the OpCSR.
    Double the IM timeout for the moment.
    
    See also:
    - https://github.com/project-chip/connectedhomeip/issues/7489
    
    Signed-off-by: Doru Gucea <doru-cristian.gucea@nxp.com>
    
    * Restyled by clang-format
    
    Co-authored-by: Restyled.io <commits@restyled.io>

commit 7b59b8fbcc5a0e9fe1694f6c35cd64ee44d28123
Author: Damian Królik <66667989+Damian-Nordic@users.noreply.github.com>`

ESP32:

`rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7072
load:0x40078000,len:13212
load:0x40080400,len:4568
0x40080400: _init at ??:?

entry 0x400806f4
I (27) boot: ESP-IDF v4.2.1-141-g1e3638390 2nd stage bootloader
I (27) boot: compile time 00:46:20
I (27) boot: chip revision: 3
I (31) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (48) boot.esp32: SPI Speed      : 40MHz
I (48) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 001e6400
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 3, min. application chip revision: 0
I (102) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x2c26c (180844) map
I (180) esp_image: segment 1: paddr=0x0003c294 vaddr=0x3ffbdb60 size=0x03d84 ( 15748) load
I (187) esp_image: segment 2: paddr=0x00040020 vaddr=0x400d0020 size=0xc867c (820860) map
0x400d0020: _stext at ??:?

I (500) esp_image: segment 3: paddr=0x001086a4 vaddr=0x3ffc18e4 size=0x00324 (   804) load
I (500) esp_image: segment 4: paddr=0x001089d0 vaddr=0x40080000 size=0x1d614 (120340) load
0x40080000: _WindowOverflow4 at /home/zx/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (575) boot: Loaded app from partition at offset 0x10000
I (575) boot: Disabling RNG early entropy source...
I (576) cpu_start: Pro cpu up.
I (579) cpu_start: Application information:
I (584) cpu_start: Project name:     chip-all-clusters-app
I (590) cpu_start: App version:      test_event_3_2021_06_10
I (596) cpu_start: Compile time:     Jul 16 2021 01:24:47
I (602) cpu_start: ELF file SHA256:  54234eeaee428d3c...
[... 291 lines cut ...]
zx@ubuntu:~/connectedhomeip/examples/all-clusters-app/esp32$ ifconfig
ens33: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.110  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::1037:7f7f:c16:4b9a  prefixlen 64  scopeid 0x20<link>
        ether 00:0c:29:2d:d4:72  txqueuelen 1000  (以太网)
        RX packets 1383052  bytes 1769573370 (1.7 GB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 512057  bytes 35193289 (35.1 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (本地环回)
        RX packets 7921  bytes 767400 (767.4 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 7921  bytes 767400 (767.4 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
`

https://static.dingtalk.com/media/lADPBFf_-nawCj_NA0_NBJ4_1182_847.jpg_720x720q90g.jpg?bizType=im

https://static.dingtalk.com/media/lADPBE1XfohY2xvNA43NBJM_1171_909.jpg_720x720q90g.jpg?bizType=im

https://static.dingtalk.com/media/lADPBFuNeRtz6hrNAvXNA4g_904_757.jpg_720x720q90g.jpg?bizType=im

jmartinez-silabs, 2021-07-20:
I am not familiar with the ESP platform. But I suggest using the same commit hash for both the app and the controller

736676938, 2021-07-21:
> 我对ESP平台不熟悉。但我建议对应用程序和控制器使用相同的提交哈希
Mac and ESP32 are the same versions。
Have you encountered this problem before?  I don't know how to solve it, or anyone knows about ESP32.

736676938, 2021-07-21:
@bzbarsky-apple

benk-remootio, 2021-07-22:
@736676938 It might be helpful if you could post the terminal log of chip-device-ctrl where you invoke the resolve 0 558

bzbarsky-apple, 2021-07-22:
Yes, it's not clear to me what exactly the problem here is.  The "MAC" part of the log shows a successful address resolution...

closed 2022-01-26 by woody-apple