project-chip/connectedhomeip #8495: Hash_SHA256_Stream is missing intermediate digest computation

tcarmelveilleux, 2021-07-20:
#### Problem
CASE spec calls for partial transcript in S2K/S3K computation, but full transcript for final keys. However `Hash_SHA256_Stream` can only give final digest and cannot give an intermediate result, which is required.

 #### Proposed Solution
Augment public API to have `GetDigest()` for current digest state in addition to `Finish()`.

emargolis, 2021-07-20:
Normally, there is no such thing in crypto as "intermediate digest" and introduction of new method to calculate that will be very confusing. I would like to avoid that if possible.

To ease on CASE processing, can we update CASE spec to calculate:
`TranscriptHash1 = CHIP_Crypto_Hash(message = Msg1)`
`TranscriptHash2 = CHIP_Crypto_Hash(message = TranscriptHashMsg1 || Msg2)`
`TranscriptHash3 = CHIP_Crypto_Hash(message = TranscriptHashMsg2 || Msg3)`

@balducci-apple @bzbarsky-apple @tcarmelveilleux  your opinion?

tcarmelveilleux, 2021-07-20:
"intermediate digest" is a misnomer.

I have seen many instances in crypto protocols in the past of obtaining transcripts at multiple points. The hard way is to keep all the data. The method I add here is not wrong and does not do anything unexpected. It is well documented, and it allows computing the digest to-spec without changing the spec or keeping copies of messages long-lost (which are problematic from a memory perspective).

emargolis, 2021-07-20:
After chatting with @tcarmelveilleux I agree that introduction of `GetDigest()` might be preferred option at the moment, without a need to change the spec.

PR #8496 (2021-07-21): Add intermediate digest computation to Hash_SHA256_Stream

#### Problem
Add intermediate digest computation to Hash_SHA256_Stream, required for to-spec case transcript computation.

Issue #4189
Fixes #8495

#### Change overview

- Add `GetDigest()` to `Hash_SHA256_Stream` and associated tests
- Make `Hash_SHA256_Stream` work with safe spans (partial fix
  to #4189)
- Fix all usages of `Hash_SHA256_Stream`
- Drive-by simplification of hex conversion in Rotating ID

#### Testing

- Cert tests (test_suite.sh) for both mbedTLS and OpenSSL
- gn check on host for both mbedTLS and OpenSSL
- cross-val [... cut]
closed 2021-07-21 by bzbarsky-apple