project-chip/connectedhomeip #8502: Android app failed to commission EFR32 Light App(CHIP over Thread device)

leasual, 2021-07-20:
Use CHIP-TOOl for android app, failed to connect EFR32 Light APP, but use iOS is success and i can on off the device light.
please help.

@jepenven-silabs @jmartinez-silabs 
Android Log:

```
2021-07-20 14:06:52.955 16322-16322/com.se.chip.chiptool D/ViewRootImpl@30f425a[AddDeviceActivity]: ViewPostIme pointer 0
2021-07-20 14:06:52.975 16322-16322/com.se.chip.chiptool D/ViewRootImpl@30f425a[AddDeviceActivity]: ViewPostIme pointer 1
2021-07-20 14:06:53.040 16322-16322/com.se.chip.chiptool W/ActivityThread: handleWindowVisibility: no activity for token android.os.BinderProxy@11e0757
2021-07-20 14:06:53.103 16322-16322/com.se.chip.chiptool I/MultiWindowDecorSupport: updateCaptionType >> DecorView@f295699[], isFloating: false, isApplication: true, hasWindowDecorCaption: false, hasWindowControllerCallback: true
2021-07-20 14:06:53.104 16322-16322/com.se.chip.chiptool D/MultiWindowDecorSupport: setCaptionType = 0, DecorView = DecorView@f295699[]
2021-07-20 14:06:53.118 16322-16322/com.se.chip.chiptool D/KeyValueStoreManager: Key 'ipAddress' not found in shared preferences
2021-07-20 14:06:53.119 16322-16322/com.se.chip.chiptool D/KeyValueStoreManager: Key 'port' not found in shared preferences
2021-07-20 14:06:53.119 16322-16322/com.se.chip.chiptool D/PairingViewModel: networkType= 1
2021-07-20 14:06:53.126 16322-16322/com.se.chip.chiptool D/CTL: newDeviceController() called
2021-07-20 14:06:53.129 16322-16322/com.se.chip.chiptool D/TOO: KVS: Getting key ExampleOpCredsCAKey
2021-07-20 14:06:53.131 16322-16322/com.se.chip.chiptool D/TOO: KVS: Getting key ExampleOpCredsICAKey
2021-07-20 14:06:53.132 16322-16322/com.se.chip.chiptool D/IN: TransportMgr initialized
2021-07-20 14:06:53.132 16322-16322/com.se.chip.chiptool D/DIS: Init admin pairing table with server storage
2021-07-20 14:06:53.132 16322-16322/com.se.chip.chiptool D/TOO: KVS: Getting key CHIPAdmin0
2021-07-20 14:06:53.132 16322-16322/com.se.chip.chiptool D/IN: Loading certs from KVS
2021-07-20 14:06:53.132 16322-16322/com.se.chip.chiptool D/IN: local node id is 0x000000000001B669
2021-07-20 14:06:53.132 16322-16322/com.se.chip.chiptool D/ZCL: Using ZAP configuration...
2021-07-20 14:06:53.134 16322-16322/com.se.chip.chiptool D/ZCL: deactivate report event
2021-07-20 14:06:53.134 16322-16322/com.se.chip.chiptool D/CTL: Getting operational keys
2021-07-20 14:06:53.134 16322-16322/com.se.chip.chiptool D/CTL: Generating credentials
2021-07-20 14:06:53.134 16322-16322/com.se.chip.chiptool D/CTL: Loaded credentials successfully
2021-07-20 14:06:53.134 16322-16322/com.se.chip.chiptool D/TOO: KVS: Getting key StartKeyID
2021-07-20 14:06:53.136 16322-16322/com.se.chip.chiptool D/PairingViewModel: get deviceInfo= CHIPDeviceInfo(version=0, vendorId=9050, productId=21324, discriminator=3840, setupPinCode=73141520, optionalQrCodeInfoMap={}, discoveryCapabilities=[BLE])
2021-07-20 14:06:53.143 16322-16322/com.se.chip.chiptool D/BluetoothAdapter: STATE_ON
2021-07-20 14:06:53.150 16322-16322/com.se.chip.chiptool I/chip.BluetoothManager: Starting Bluetooth scan
2021-07-20 14:06:53.151 16322-16322/com.se.chip.chiptool D/BluetoothAdapter: STATE_ON
2021-07-20 14:06:53.153 16322-16322/com.se.chip.chiptool D/BluetoothAdapter: STATE_ON
2021-07-20 14:06:53.154 16322-16322/com.se.chip.chiptool D/BluetoothAdapter: BLE support array set: 010011
2021-07-20 14:06:53.154 16322-16322/com.se.chip.chiptool D/BluetoothLeScanner: Start Scan with callback
2021-07-20 14:06:53.157 16322-16340/com.se.chip.chiptool D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=11 mScannerId=0
2021-07-20 14:06:53.185 16322-16322/com.se.chip.chiptool D/ViewRootImpl@5ed5c2f[PairingActivity]: setView = com.android.internal.policy.DecorView@f295699 TM=true MM=false
2021-07-20 14:06:53.187 16322-16322/com.se.chip.chiptool D/ViewRootImpl@30f425a[AddDeviceActivity]: MSG_WINDOW_FOCUS_CHANGED 0 1
2021-07-20 14:06:53.187 16322-16322/com.se.chip.chiptool D/InputMethodManager: prepareNavigationBarInfo() DecorView@78f9a11[AddDeviceActivity]
2021-07-20 14:06:53.187 16322-16322/com.se.chip.chiptool D/InputMethodManager: getNavigationBarColor() -855310
2021-07-20 14:06:53.207 16322-16473/com.se.chip.chiptool W/LOTTIE: Animation contains merge paths. Merge paths are only supported on KitKat+ and must be manually enabled by calling enableMergePathsForKitKatAndAbove().
2021-07-20 14:06:53.222 16322-16322/com.se.chip.chiptool D/ViewRootImpl@5ed5c2f[PairingActivity]: Relayout returned: old=(0,0,1080,2340) new=(0,0,1080,2340) req=(1080,2340)0 dur=14 res=0x7 s={true 491296743424} ch=true
2021-07-20 14:06:53.222 16322-16359/com.se.chip.chiptool D/OpenGLRenderer: createReliableSurface : 0x72d39d9b80, 0x7263916000
2021-07-20 14:06:53.224 16322-16359/com.se.chip.chiptool D/mali_winsys: EGLint new_window_surface(egl_winsys_display *, void *, EGLSurface, EGLConfig, egl_winsys_surface **, EGLBoolean) returns 0x3000
2021-07-20 14:06:53.258 16322-16322/com.se.chip.chiptool D/ViewRootImpl@5ed5c2f[PairingActivity]: MSG_WINDOW_FOCUS_CHANGED 1 1
2021-07-20 14:06:53.258 16322-16322/com.se.chip.chiptool D/InputMethodManager: prepareNavigationBarInfo() DecorView@f295699[PairingActivity]
2021-07-20 14:06:53.258 16322-16322/com.se.chip.chiptool D/InputMethodManager: getNavigationBarColor() -855310
2021-07-20 14:06:53.260 16322-16322/com.se.chip.chiptool D/InputMethodManager: prepareNavigationBarInfo() DecorView@f295699[PairingActivity]
2021-07-20 14:06:53.260 16322-16322/com.se.chip.chiptool D/InputMethodManager: getNavigationBarColor() -855310
2021-07-20 14:06:53.260 16322-16322/com.se.chip.chiptool V/InputMethodManager: Starting input: tba=com.se.chip.chiptool ic=null mNaviBarColor -855310 mIsGetNaviBarColorSuccess true , NavVisible : true , NavTrans : false
2021-07-20 14:06:53.260 16322-16322/com.se.chip.chiptool D/InputMethodManager: startInputInner - Id : 0
2021-07-20 14:06:53.260 16322-16322/com.se.chip.chiptool I/InputMethodManager: startInputInner - mService.startInputOrWindowGainedFocus
2021-07-20 14:06:53.276 16322-16322/com.se.chip.chiptool D/InputTransport: Input channel destroyed: 'ClientS', fd=87
2021-07-20 14:06:53.278 16322-16322/com.se.chip.chiptool D/ViewRootImpl@5ed5c2f[PairingActivity]: MSG_RESIZED: frame=(0,0,1080,2340) ci=(0,83,0,126) vi=(0,83,0,126) or=1
2021-07-20 14:06:53.402 16322-16322/com.se.chip.chiptool W/LOTTIE: Animation contains merge paths but they are disabled.
2021-07-20 14:06:53.501 16322-16322/com.se.chip.chiptool I/chip.BluetoothManager: Bluetooth Device Scanned Addr: 46:80:02:80:33:E6, Name EFR32_LIGHT
2021-07-20 14:06:53.507 16322-16322/com.se.chip.chiptool D/BluetoothAdapter: STATE_ON
2021-07-20 14:06:53.507 16322-16322/com.se.chip.chiptool D/BluetoothLeScanner: Stop Scan with callback
2021-07-20 14:06:53.514 16322-16322/com.se.chip.chiptool D/PairingViewModel: get deviceInfo= 46:80:02:80:33:E6
2021-07-20 14:06:53.515 16322-16322/com.se.chip.chiptool I/chip.BluetoothManager: Connecting
2021-07-20 14:06:53.516 16322-16322/com.se.chip.chiptool D/BluetoothAdapter: STATE_ON
2021-07-20 14:06:53.519 16322-16322/com.se.chip.chiptool D/BluetoothGatt: connect() - device: 46:80:02:80:33:E6, auto: false
2021-07-20 14:06:53.519 16322-16322/com.se.chip.chiptool D/BluetoothAdapter: isSecureModeEnabled
2021-07-20 14:06:53.521 16322-16322/com.se.chip.chiptool D/BluetoothGatt: registerApp()
2021-07-20 14:06:53.521 16322-16322/com.se.chip.chiptool D/BluetoothGatt: registerApp() - UUID=b1c4e6e0-52f0-4fd0-9646-a33d17312b30
2021-07-20 14:06:53.528 16322-16340/com.se.chip.chiptool D/BluetoothGatt: onClientRegistered() - status=0 clientIf=11
2021-07-20 14:06:53.603 16322-16359/com.se.chip.chiptool W/libEGL: EGLNativeWindowType 0x72d39dc610 disconnect failed
2021-07-20 14:06:53.611 16322-16322/com.se.chip.chiptool D/ViewRootImpl@30f425a[AddDeviceActivity]: Relayout returned: old=(0,0,1080,2340) new=(0,0,1080,2340) req=(1080,2340)8 dur=8 res=0x5 s={false 0} ch=true
2021-07-20 14:06:53.617 16322-16322/com.se.chip.chiptool D/ViewRootImpl@30f425a[AddDeviceActivity]: stopped(true) old=false
[... 198 lines cut ...]
2021-07-20 14:03:25.308063+0800 CHIPTool[9001:3797832] [all] 🔵 [1626761005308] [9001:3797832] CHIP: [IN] Sending secure msg on generic transport
2021-07-20 14:03:25.308475+0800 CHIPTool[9001:3797832] [all] 🔵 [1626761005308] [9001:3797832] CHIP: [IN] Secure msg send status No Error
2021-07-20 14:03:25.308496+0800 CHIPTool[9001:3797832] [all] 🟢 [1626761005308] [9001:3797832] CHIP: [DMG] ICR moving to [   Sending]
[... cut ...]
2021-07-20 14:03:28.775307+0800 CHIPTool[9001:3798445] [all] 🔵 [1626761008775] [9001:3798445] CHIP: [ZCL] DefaultResponse:
2021-07-20 14:03:28.775339+0800 CHIPTool[9001:3798445] [all] 🔵 [1626761008775] [9001:3798445] CHIP: [ZCL]   Transaction: 0x104e6e900
2021-07-20 14:03:28.775361+0800 CHIPTool[9001:3798445] [all] 🔵 [1626761008775] [9001:3798445] CHIP: [ZCL]   status: EMBER_ZCL_STATUS_SUCCESS (0x00)
2021-07-20 14:03:28.775425+0800 CHIPTool[9001:3798445] [all] 🟢 [1626761008775] [9001:3798445] CHIP: [DMG] ICR moving to [Uninitiali]

```

jmartinez-silabs, 2021-07-20:
Sorry, this is a known issue we have another ticket tracking this. We have someone working on fixing this, but It got delayed for multiple reason. It is comming soon. I wil link the tickets together so you know when this is fixed.

https://github.com/project-chip/connectedhomeip/issues/7078

leasual, 2021-07-21:
ok, thank you.

jmartinez-silabs, 2021-08-04:
Fixed with #8666

closed 2021-08-04 by jmartinez-silabs