project-chip/connectedhomeip #8533: manual pairing code generated is not encoded properly

cjandhyala, 2021-07-21:
The manual pairing code generated currently either in example apps or thru the 'setup-payload generate' in the python controller is not encoded properly. Since the pairing code is not correct , chip-tool could not provision the device using the same pairing code.

Steps to reproduce:

run the all-cluster-app example app on the accessory
$ sudo ./out/all-clusters-app/chip-all-clusters-app --wifi
[1626738927.239620][12299:12299] CHIP:DL: Device Configuration:
[1626738927.239679][12299:12299] CHIP:DL: Serial Number: TEST_SN
[1626738927.239704][12299:12299] CHIP:DL: Vendor Id: 9050 (0x235A)
[1626738927.239723][12299:12299] CHIP:DL: Product Id: 65279 (0xFEFF)
[1626738927.239744][12299:12299] CHIP:DL: Product Revision: 1
[1626738927.239764][12299:12299] CHIP:DL: Setup Pin Code: 20202021
[1626738927.239785][12299:12299] CHIP:DL: Setup Discriminator: 3840 (0xF00)
[1626738927.239805][12299:12299] CHIP:DL: Manufacturing Date: (not set)
[1626738927.239824][12299:12299] CHIP:DL: Device Type: 65535 (0xFFFF)
[1626738927.239877][12299:12299] CHIP:SVR: SetupQRCode: [MT:YNJV75HZ00KA0648G00]
[1626738927.239925][12299:12299] CHIP:SVR: Copy/paste the below URL in a browser to see the QR Code:
[1626738927.239944][12299:12299] CHIP:SVR: https://dhrishi.github.io/connectedhomeip/qrcode.html?data=MT%3AYNJV75HZ00KA0648G00
[1626738927.239984][12299:12299] CHIP:SVR: Manual pairing code: [34970112332]
[1626738927.240459][12299:12299] CHIP:ZCL: Using ZAP configuration...
[1626738927.240878][12299:12299] CHIP:ZCL: deactivate report event

pick the Manual pairing code 34970112332 and parse it using parse-manual utility , VIP/PID/FLow ..etc fields are shown as 0 instead of the correct values.
chip-device-ctrl > setup-payload parse-manual 34970112332
Version: 0
VendorID: 0
ProductID: 0
CommissioningFlow: 0
RendezvousInformation: 0
Discriminator: 3840
SetUpPINCode: 20202021

Paring with the same manual code using chip-tool throws error.

% sudo ./out/debug/chip-tool pairing manualcode 0 34970112332
[1626739215843] [58333:4692312] CHIP: [DL] _Init
[1626739215844] [58333:4692312] CHIP: [IN] TransportMgr initialized
[1626739215844] [58333:4692312] CHIP: [DIS] Init admin pairing table with server storage
[1626739215844] [58333:4692312] CHIP: [IN] Loading certs from KVS
[1626739215845] [58333:4692312] CHIP: [IN] local node id is 0x000000000001B669
[1626739215845] [58333:4692312] CHIP: [ZCL] Using ZAP configuration...
[1626739215845] [58333:4692312] CHIP: [ZCL] deactivate report event
[1626739215845] [58333:4692312] CHIP: [CTL] Getting operational keys
[1626739215845] [58333:4692312] CHIP: [CTL] Generating credentials
[1626739215846] [58333:4692312] CHIP: [CTL] Loaded credentials successfully
[1626739215846] [58333:4692317] CHIP: [TOO] Init Failure! PairDevice: CHIP Error 4047 (0x00000FCF): Invalid argument
[1626739215846] [58333:4692312] CHIP: [TOO] Run command failure: CHIP Error 4047 (0x00000FCF): Invalid argument
[1626739215846] [58333:4692312] CHIP: [CTL] Shutting down the commissioner
[1626739215846] [58333:4692312] CHIP: [CTL] Shutting down the controller
[1626739215846] [58333:4692312] CHIP: [DL] Inet Layer shutdown
[1626739215846] [58333:4692315] CHIP: [IN] Async DNS worker thread woke up.
[1626739215846] [58333:4692315] CHIP: [IN] Async DNS worker thread exiting.
[1626739215846] [58333:4692316] CHIP: [IN] Async DNS worker thread woke up.
[1626739215846] [58333:4692316] CHIP: [IN] Async DNS worker thread exiting.
[1626739215846] [58333:4692312] CHIP: [DL] BLE layer shutdown
[1626739215846] [58333:4692312] CHIP: [DL] System Layer shutdown

Damian-Nordic, 2021-07-21:
Note that VID and PID are mandatory for the manual pairing code only if CommissioningFlow != 0, so it's not a bug.

cjandhyala, 2021-07-23:
@Damian-Nordic , how come the commissioning flow is encoded as 0 on the default raspi apps, when those are used to provision over BLE. 

Aslo un-related to this, example apps on linux and nordic platforms generate 11 digit code whereas apps on M5 generate 21 digit code. Those 21 digit code seem to parse fine

Damian-Nordic, 2021-07-23:
According to the spec:
1. CommissioningFlow = 0 when no special action is required before the commissioning; 1 - when some action is needed (e.g. pressing a button to open the pairing window); 2 - when the commissioner must communicate with a vendor specific service before commissioning. 
2. Further in the section about Manual Pairing Code you can find info about VENDOR_ID field:

> Needed only to support devices that need custom flow
> before commissioning.
> If an accompanying QR code is present on the device with
> the Custom Flow flag set to a non-zero value, or if the
> device requires manufacturer-specific flow, this element
> SHALL be included.

...and PRODUCT_ID:

> * This element SHALL be included if and only if the
> VENDOR_ID element is present.

Thus both formats with and without VID & PID (11 and 21 digits) are correct according to the spec.

cjandhyala, 2021-08-20:
@Damian-Nordic @yufengwangca , The RendezvousInformation is parse as 0 , whereas we have Rendezvous flag to set to BLE in the AppMain.cpp code `PrintOnboardingCodes(chip::RendezvousInformationFlag::kBLE);` , which looks like not used while encoding the manual code ?

PR #9297 (2021-08-31): [Linux]: Fix for Device Configuration info does not match with the generated pairing code
#### Problem
What is being fixed?  Examples:
* This issue was reported from TE5, the dumped Device Configuration info does not match with the generated SetupQRCode and Manual pairing code on Linux platform.

```
[1626738927.239620][12299:12299] CHIP:DL: Device Configuration:
[1626738927.239679][12299:12299] CHIP:DL: Serial Number: TEST_SN
[1626738927.239704][12299:12299] CHIP:DL: Vendor Id: 9050 (0x235A)
[1626738927.239723][12299:12299] CHIP:DL: Product Id: 65279 (0xFEFF)
[1626738927.239744][12299:12299]  [... cut]
closed 2021-08-31 by andy31415