project-chip/connectedhomeip #8558: Use of Variable Length Arrays in generated code

mdamle, 2021-07-22:
#### Problem
variable length arrays  are used in generated code: CHIPClientCallbacks.cpp. For example:  https://github.com/project-chip/connectedhomeip/blob/master/src/controller/data_model/gen/CHIPClientCallbacks.cpp#L854 
`
uint16_t count = chip::Encoding::LittleEndian::Read16(message);
`
_AudioOutputInfo` data[count];
`
There are about 24 such instances. There are 2 issues with this usage:

1. Variable Length Arrays are not universally supported by all compilers since they are not part of c++ standard
2. The generated code uses over the wire value to allocate data on stack which is dangerous.

 * expected behavior
 
1. Check size before allocation
2. Allocate on heap instead of stack

 * actual behavior
 As explained in the problem section.
 * steps to reproduce
Compile code on top of tree and inspect generated CHIPClientCallbacks.cpp file.
 >

mdamle, 2021-07-22:
cc @tcarmelveilleux for visibility

bzbarsky-apple, 2021-07-22:
@vivien-apple @mrjerryjohns

tcarmelveilleux, 2021-07-22:
The root cause of this issue is wanted to pass "flat arrays" to callbacks when lists are used. I think these lists should be handled by iteration over the payload with an abstracted list iterator backed by a TLV reader, rather than copy to flat arrays. That would avoid alloc entirely, and avoid requiring to trust any sizes from the wire for an alloc. A temporary reader could be used to schema-validate, but access to the list elements should be abstracted with an iterator.

mlepage-google, 2021-07-22:
+1 for iterator access.

bzbarsky-apple, 2021-11-04:
Fixed in https://github.com/project-chip/connectedhomeip/pull/10681

closed 2021-11-04 by bzbarsky-apple