project-chip/connectedhomeip #8564: CreateNewUDCClientState is not safe to call more than once in a row

bzbarsky-apple, 2021-07-22:
#### Problem
Consider the following order of events:

1. `CreateNewUDCClientState` is called.
2. Some time passes (e.g. thread lost timeslice)
3. `CreateNewUDCClientState` is called again.
4. Consumer now tries to do something with the state from step 1.

This never happens in our code right now (as long as all access to the UDC client states is on a single thread) because we never store the pointers `CreateNewUDCClientState` returns across another call to `CreateNewUDCClientState`.  But this seems pretty fragile.

 #### Proposed Solution
Figure out a safer API.

chrisdecenzo, 2021-08-10:
Fixed in 8084, assigning to Boris to close

bzbarsky-apple, 2023-03-13:
And just to be clear, @chrisdecenzo this was not addressed in #8084