project-chip/connectedhomeip #8565: Various outstanding issues from PR 8084

bzbarsky-apple, 2021-07-22:
#### Problem
The code that landed in #8084 has various unresolved issues, none of which are caught because it is neither compiled nor run in CI:

* [ ] Is not tested in CI
* [ ] Probably doesn't compile, since `kAnyNodeId` is not a thing anymore.
* [ ] Never shuts down the `DeviceCommissioner` after calling `Init` on it.  It should be shut down after `RunEventLoop` returns.
* [ ] `DeviceCommissioner::Shutdown` never frees `mUdcServer`, so repeated init/shutdown cycles will leak more and more memory.
* [ ] Once the previous item is fixed, we should probably delete `mUdcTransportMgr` before deleting `mUdcServer`, so we don't have a dangling pointer for a bit there, with potential use-after-free.
* [ ] `UserDirectedCommissioningClient::SendUDCMessage` should probably be setting the (required) exchange id in the message header.  Possibly this should just get sent via an actual exchange instead of via a raw call on a transport manager....  @pan-apple any thoughts?

 #### Proposed Solution
Fix issues.

chrisdecenzo, 2021-08-10:
Issues addressed in 8739, assigning to Boris to close, thanks

bzbarsky-apple, 2021-08-11:
The last bullet point does not look fixed to me.  That is, this bit:

> UserDirectedCommissioningClient::SendUDCMessage should probably be setting the (required) exchange id in the message header. Possibly this should just get sent via an actual exchange instead of via a raw call on a transport manager....

mrjerryjohns, 2022-01-31:
The last bullet still looks to be un-addressed in the code on master.

mrjerryjohns, 2022-01-31:
Requesting secondary triage since it's not clear how important fixing that is.

bzbarsky-apple, 2022-02-01:
Do we want our UDC messages to look kind of like what the spec says?

That said, I don't know what the _practical_ impact of sending every single UDC message with an exchange id of 0 is.

chrisdecenzo, 2022-05-13:
These UDC messages are fire-and-forget with no accommodation for responses so I'm not sure what the value in setting a specific exchange id would be.

bzbarsky-apple, 2022-05-14:
The main idea of setting it would be so the other side doesn't consider them as part of an existing exchange.  Exchange id 0 is valid, last I checked, so there is potential for confusion and incorrect dispatch there....

bzbarsky-apple, 2022-05-14:
But maybe the fact that these are not operational messages and hence the source node id is not present means that does not matter in practice?

In that case, what we have is technically a spec violation (@turon can you please check whether that claim is true?) but might not matter in practice.

woody-apple, 2022-05-25:
SDK Spec Review: We do not believe this is a spec conformance issue. Second, we believe this also complete. Closing.

closed 2022-05-25 by woody-apple