project-chip/connectedhomeip #8566: ESP32 all-clusters-app long dispatch time during resolve  zcl commands

koreshx, 2021-07-22:
Hi everyone. My first attempt to get in with matter is not successive yet. I've tried to run this example with followed config
RPi4 ubuntu server 21.04 as a python controller
ESP32-S module as first node

Wi-Fi comissioning via BLE looks completed. However I can't call any of these basic commands without errors.
resolve 0 135246
zcl OnOff Off 135246 1 1

Both of them occure "Long dispatch time" error on the node's side.

RPi log:
```
chip-device-ctrl > ble-scan
2021-07-22 07:59:34,030 ChipBLEMgr   INFO     scanning started
2021-07-22 07:59:34,379 ChipBLEMgr   INFO     Name            = None
2021-07-22 07:59:34,379 ChipBLEMgr   INFO     ID              = ee421506-d55f-428e-ac8d-9f6abe3aac44
2021-07-22 07:59:34,381 ChipBLEMgr   INFO     RSSI            = -42
2021-07-22 07:59:34,382 ChipBLEMgr   INFO     Address         = 08:3A:F2:A9:2A:92
2021-07-22 07:59:34,386 ChipBLEMgr   INFO     Pairing State   = 0
2021-07-22 07:59:34,386 ChipBLEMgr   INFO     Discriminator   = 3840
2021-07-22 07:59:34,387 ChipBLEMgr   INFO     Vendor Id       = 9050
2021-07-22 07:59:34,387 ChipBLEMgr   INFO     Product Id      = 17729
2021-07-22 07:59:34,390 ChipBLEMgr   INFO     Adv UUID        = 0000fff6-0000-1000-8000-00805f9b34fb
2021-07-22 07:59:34,392 ChipBLEMgr   INFO     Adv Data        = 00000f5a234145
2021-07-22 07:59:34,392 ChipBLEMgr   INFO     
2021-07-22 07:59:44,152 ChipBLEMgr   INFO     scanning stopped
chip-device-ctrl > connect -ble 3840 20202021 135246
Device is assigned with nodeid = 135246
[1626940789.107622][6480:6483] CHIP:BLE: BLE removing known devices.
[1626940789.124799][6480:6483] CHIP:BLE: BLE initiating scan.
[1626940789.514715][6480:6483] CHIP:BLE: New device scanned: 08:3A:F2:A9:2A:92
[1626940789.514783][6480:6483] CHIP:BLE: Device discriminator match. Attempting to connect.
[1626940789.518390][6480:6483] CHIP:BLE: Scan complete notification without an active scan.
[1626940792.185465][6480:6487] CHIP:BLE: subscribe complete, ep = 0xffff851e7438
[1626940792.185868][6480:6487] CHIP:BLE: peripheral chose BTP version 3; central expected between 2 and 3
[1626940792.185928][6480:6487] CHIP:BLE: using BTP fragment sizes rx 20 / tx 128.
[1626940792.185968][6480:6487] CHIP:BLE: local and remote recv window size = 3
[1626940793.209572][6480:6487] CHIP:EM: Received message of type 33 and protocolId 0 on exchange 39017
[1626940796.719613][6480:6487] CHIP:EM: Received message of type 35 and protocolId 0 on exchange 39017
[1626940796.725138][6480:6487] CHIP:CTL: Sending root certificate to the device
[1626940796.725585][6480:6487] CHIP:IN: Encrypted message 0xffff7d7adc50 from 0x000000000001B669 to 0x000000000002104E of type 8 and protocolId 5 on exchange 39018.
[1626940796.725737][6480:6487] CHIP:IN: Sending msg 0xffff7d7adc50 to 0x000000000002104E at utc time: 1207805 msec
[1626940796.725799][6480:6487] CHIP:IN: Sending secure msg on generic transport
[1626940796.725863][6480:6487] CHIP:IN: Secure msg send status No Error
[1626940796.725923][6480:6487] CHIP:CTL: Sent root certificate to the device
[1626940799.401535][6480:6487] CHIP:IN: Secure transport received message destined to fabric 0, node 0x000000000001B669. Key ID 1
[1626940799.401702][6480:6487] CHIP:IN: Setting fabricID 2104E on admin.
[1626940799.401757][6480:6487] CHIP:IN: Since admin was modified, persisting changes to KVS
[1626940799.401869][6480:6487] CHIP:EM: Received message of type 9 and protocolId 5 on exchange 39018
SetCommandIndexStatus commandHandle=1 commandIndex=1
[1626940799.402930][6480:6487] CHIP:ZCL: DefaultResponse:
[1626940799.403079][6480:6487] CHIP:ZCL:   Transaction: 0xffff851e4c08
[1626940799.403395][6480:6487] CHIP:ZCL:   status: EMBER_ZCL_STATUS_SUCCESS (0x00)
[1626940799.403524][6480:6487] CHIP:CTL: Device confirmed that it has received the root certificate
[1626940799.404296][6480:6487] CHIP:IN: Encrypted message 0xffff7d7ad8b0 from 0x000000000001B669 to 0x000000000002104E of type 8 and protocolId 5 on exchange 39019.
[1626940799.404447][6480:6487] CHIP:IN: Sending msg 0xffff7d7ad8b0 to 0x000000000002104E at utc time: 1210484 msec
[1626940799.404555][6480:6487] CHIP:IN: Sending secure msg on generic transport
[1626940799.404652][6480:6487] CHIP:IN: Secure msg send status No Error
[1626940802.034213][6480:6487] CHIP:IN: Secure transport received message destined to fabric 0, node 0x000000000001B669. Key ID 1
[1626940802.034396][6480:6487] CHIP:EM: Received message of type 9 and protocolId 5 on exchange 39019
[1626940802.034668][6480:6487] CHIP:ZCL: OpCSRResponse:
[1626940802.034780][6480:6487] CHIP:ZCL:   CSR: 205
[1626940802.034866][6480:6487] CHIP:ZCL:   CSRNonce: 32
[1626940802.034949][6480:6487] CHIP:ZCL:   VendorReserved1: 0
[1626940802.035053][6480:6487] CHIP:ZCL:   VendorReserved2: 0
[1626940802.035134][6480:6487] CHIP:ZCL:   VendorReserved3: 0
[1626940802.035233][6480:6487] CHIP:ZCL:   Signature: 0
[1626940802.035342][6480:6487] CHIP:CTL: Received certificate signing request from the device
[1626940802.039182][6480:6487] CHIP:CTL: Getting intermediate CA certificate from the issuer
[1626940802.039662][6480:6487] CHIP:CTL: GetIntermediateCACertificate returned No Error
[... 563 lines cut ...]
I (235876) chip[EM]: Received message of type 48 and protocolId 0 on exchange 39023
V (235876) chip[EM]: ec id: 39023, Delegate: 0x0x3ffc528c
I (235886) chip[IN]: CASE Server received SigmaR1 message. Starting handshake. EC 0x3ffc4fe8
V (235886) chip[SC]: Waiting for SigmaR1 msg
V (235896) chip[SC]: Received SigmaR1 msg
V (235896) chip[SC]: Peer assigned session key ID 2
V (237666) chip[SC]: Sent SigmaR2 msg
E (237666) chip[DL]: Long dispatch time: 1790 ms
I (237826) chip[EM]: Received message of type 63 and protocolId 0 on exchange 39023
E (237826) chip[SC]: Received error (255) during CASE pairing process
I (237836) chip[IN]: CASE Session establishment failed: CHIP Error 54 (0x00000054): Invalid CASE parameter
I (249496) chip[EM]: Received message of type 48 and protocolId 0 on exchange 39024
V (249496) chip[EM]: ec id: 39024, Delegate: 0x0x3ffc528c
I (249506) chip[IN]: CASE Server received SigmaR1 message. Starting handshake. EC 0x3ffc4fe8
V (249506) chip[SC]: Waiting for SigmaR1 msg
V (249516) chip[SC]: Received SigmaR1 msg
V (249516) chip[SC]: Peer assigned session key ID 2
V (251356) chip[SC]: Sent SigmaR2 msg
E (251356) chip[DL]: Long dispatch time: 1860 ms
I (251366) chip[EM]: Received message of type 63 and protocolId 0 on exchange 39024
E (251366) chip[SC]: Received error (255) during CASE pairing process
I (251376) chip[IN]: CASE Session establishment failed: CHIP Error 54 (0x00000054): Invalid CASE parameter
```

Thanks a lot for any advice.

sweetymhaiske, 2021-07-22:
@koreshx  Is your host connected to the same access point to which you are trying to connect your esp32 device?

koreshx, 2021-07-22:
> @koreshx Is your host machine connected to the same access point to which you are trying to connect your esp32 device?

Yes. They both are connected to one AP. I can see starting the communication in ESP32 log

```
I (249496) chip[EM]: Received message of type 48 and protocolId 0 on exchange 39024
V (249496) chip[EM]: ec id: 39024, Delegate: 0x0x3ffc528c
I (249506) chip[IN]: CASE Server received SigmaR1 message. Starting handshake. EC 0x3ffc4fe8
V (249506) chip[SC]: Waiting for SigmaR1 msg
V (249516) chip[SC]: Received SigmaR1 msg
V (249516) chip[SC]: Peer assigned session key ID 2
V (251356) chip[SC]: Sent SigmaR2 msg
E (251356) chip[DL]: Long dispatch time: 1860 ms
```
But the communication fails with Long dispatch time: 1860 ms

sweetymhaiske, 2021-07-22:
@koreshx  Can you please share the Git HEAD on which you compiled the app?
Have you compiled the app and the controller with the same Git HEAD?

koreshx, 2021-07-22:
Thanks for the prompt reply. I compiled  the app and the controller with some days lag. But it seams those have identical git HEAD
APP:
```
cat .git/HEAD
ref: refs/heads/master
```
HOST:
```
cat .git/HEAD
ref: refs/heads/master
```

sweetymhaiske, 2021-07-22:
@koreshx  Can you please share the commit id, so that I can try on my side?

koreshx, 2021-07-22:
> @koreshx Can you please share the commit id, so that I can try on my side?

APP commint id: 
`6aab4c130f3809da2062b2bc7ab451e4c4467e49`

HOST commit id:
`3e57b5fda3f51912483278581feb3d4906e0562c`

sweetymhaiske, 2021-07-22:
@koreshx Commit id's are different. Can you please try commissioning with host and controller compiled on same commit id?

koreshx, 2021-07-22:
> @koreshx Commit id's are different. Can you please try commissioning with host and controller compiled on same commit id?
Of course I will try to update both of them and try again.

sweetymhaiske, 2021-07-22:
@koreshx Thanks.

koreshx, 2021-07-22:
Oh. I've sucessfully updated CHIP repo on the both systems, commit id are identical. I've already prepared host system but have a problem with app system.  I've recieved the follow error during source ./scripts/bootstrap.sh calling

"raise subprocess.CalledProcessError(err.returncode, err.cmd,
subprocess.CalledProcessError: Command '['ninja', '-C', '/home/koresh/tools/connectedhomeip/.environment/gn_out', '-v', ':python_packages.install']' returned non-zero exit status 1.@"

App system has python 3.8.10

koreshx, 2021-07-22:
The same problem after python upgarding to 3.9.5

sweetymhaiske, 2021-07-22:
If you are on a recent master branch commit head, `db59aa0ffa794c25f8e63390a96fcc74a93af7e4` then all-clusters-app will not compile fine. #8569 
However, you can try this commit af1cc26c1a055792452571fe929818642b2ef981` .
I tried bootstrap recently two days back, it worked fine for me.
Can you please clean the `.environment` folder and try running that script again?

koreshx, 2021-07-22:
I've finally crashed second system down )
WIll try your advice on the new VM asap. Thanks.

koreshx, 2021-07-23:
Bingo! I finally have success with this case. Led is turn on and turn off respectively commands
zcl OnOff Off 135246 1 1
zcl OnOff On 135246 1 1
I've compiled both of environments using af1cc26c1a055792452571fe929818642b2ef981 commit id.
Many thanks.
For now I've got followed behaviour of this example. This example sequence works properly. If I restart the ESP32 everything still works. However system breaks down after RPi controller reboots. I have followed error

```
(python_env) ubuntu@ubuntu:~/connectedhomeip$ chip-device-ctrl

[1627029512.680178][2392:2392] CHIP:IN: local node id is 0x000000000001B669
[1627029512.684789][2392:2392] CHIP:DL: Avahi client registered
[1627029512.686644][2392:2392] CHIP:ZCL: Using ZAP configuration...
[1627029512.686760][2392:2392] CHIP:ZCL: deactivate report event
[1627029512.686807][2392:2392] CHIP:CTL: Getting operational keys
[1627029512.687252][2392:2392] CHIP:CTL: Getting root certificate for the controller from the issuer
[1627029512.688381][2392:2392] CHIP:CTL: Generating operational certificate for the controller
[1627029512.689314][2392:2392] CHIP:CTL: Getting intermediate CA certificate from the issuer
[1627029512.689836][2392:2392] CHIP:CTL: GetIntermediateCACertificate returned ../../src/controller/ExampleOperationalCredentialsIssuer.cpp:129: No Error
[1627029512.690192][2392:2392] CHIP:CTL: Generating credentials
[1627029512.690749][2392:2392] CHIP:CTL: Loaded credentials successfully
[1627029512.703625][2392:2399] CHIP:DL: Platform main loop started.
Chip Device Controller Shell

chip-device-ctrl > resolve 0 135246
[1627029553.111807][2392:2399] CHIP:DL: Avahi resolve found
[1627029553.111932][2392:2399] CHIP:DIS: Node ID resolved for 0x000000000002104E to [10.0.0.76]:5540
Failed to update node address: 72
Exception ignored on calling ctypes callback function: <function ChipDeviceController.__init__.<locals>.HandleAddressUpdateComplete at 0xffff8af3be50>
Traceback (most recent call last):
  File "/home/ubuntu/connectedhomeip/out/python_env/lib/python3.9/site-packages/chip/ChipDeviceCtrl.py", line 114, in HandleAddressUpdateComplete
    self.state = DCState.IDLEHandleCommissioningComplete
  File "/usr/lib/python3.9/enum.py", line 405, in __getattr__
    raise AttributeError(name) from None
AttributeError: IDLEHandleCommissioningComplete
[1627029553.114177][2392:2399] CHIP:CTL: OperationalDiscoveryComplete for device ID 135246
```

Is it normal behaviour at current project stage?

sweetymhaiske, 2021-07-26:
@koreshx Yes, currently if we restart the controller, then we need to perform the steps again.

koreshx, 2021-07-26:
OK. I think I'll close this issue.

closed 2021-07-26 by koreshx