project-chip/connectedhomeip #8569: ESP32: All clusters app fails to boot up.

sweetymhaiske, 2021-07-22:
#### Problem
* All clusters app for esp32 fails to boot up, it throws below error
```
I (778) wifi:wifi driver task: 3ffdaa8c, prio:23, stack:6656, core=0
I (778) system_api: Base MAC address is not set
I (778) system_api: read default base MAC address from EFUSE
I (788) wifi:wifi firmware version: c7d0450
I (788) wifi:wifi certification version: v7.0
I (788) wifi:config NVS flash: enabled
I (788) wifi:config nano formating: disabled
I (808) wifi:Init data frame dynamic rx buffer num: 32
I (808) wifi:Init management frame dynamic rx buffer num: 32
I (818) wifi:Init management short buffer num: 32
I (818) wifi:Init dynamic tx buffer num: 32
I (818) wifi:Init static rx buffer size: 1600
                                                                                                                                                       I (828) wifi:Init static rx buffer num: 10
I (828) wifi:Init dynamic rx buffer num: 32
I (838) wifi_init: rx ba win: 6
I (838) wifi_init: tcpip mbox: 32
I (838) wifi_init: udp mbox: 6
I (848) wifi_init: tcp mbox: 6
I (848) wifi_init: tcp tx win: 5744
I (848) wifi_init: tcp rx win: 5744
I (848) wifi_init: tcp mss: 1440
I (858) wifi_init: WiFi IRAM OP enabled
I (868) wifi_init: WiFi RX IRAM OP enabled
E (878) chip[DL]: Configuration Manager initialization failed: Error 0x05001102
> E (888) all-clusters-app: device.Init() failed: Error 0x05001102
```

dhrishi, 2021-07-22:
@sweetymhaiske While you debug this, please find the commit that causes this issue. 

Note: This bootup failure happens after flash is erased (`idf.py erase_flash`) and then the firmware is flashed (`idf.py build flash`)

sweetymhaiske, 2021-07-22:
The commit id that introduced the issue `162366f07969412131464768ea519ea4f00f0d93`

sweetymhaiske, 2021-07-22:
This is the commit id which caused the issue 162366f07969412131464768ea519ea4f00f0d93

kpschoedel, 2021-07-22:
I don't see this myself — perhaps my device configuration is different. It looks like it could be the same root cause (I copy-pasted a bit of code and missed changing `NVS_READONLY` to `NVS_READWRITE` at `src/platform/ESP32/KeyValueStoreManagerImpl.cpp` line 63).

PR #8574 (2021-07-23): Fix ESP32 storage
#### Problem

Commit 162366f0 (#8469) broke persistent storage on ESP32.

#### Change overview

- Fixed return value of `ESP32Config::EnsureNamespace()`.
- Fixed a copy-paste mistake (`NVS_READONLY` for `NVS_READWRITE`)
  in `KeyValueStoreManagerImpl::_Put()`.
- Added string presentation of ESP32 errors to aid troubleshooting.

Fixes #8572 _ESP32 storage probably broken_
Fixes #8569 _ESP32: All clusters app fails to boot up._

#### Testing

Manual on M5Stack
commit 7386e5fe (2021-07-23): Fix ESP32 storage (#8574)

* Fix ESP32 storage

#### Problem

Commit 162366f0 (#8469) broke persistent storage on ESP32.

#### Change overview

- Fixed return value of `ESP32Config::EnsureNamespace()`.
- Fixed a copy-paste mistake (`NVS_READONLY` for `NVS_READWRITE`)
  in `KeyValueStoreManagerImpl::_Put()`.
- Added string presentation of ESP32 errors.

Fixes #8572 _ESP32 storage probably broken_
Fixes #8569 _ESP32: All clusters app fails to boot up._

#### Testing

Manual on M5Stack

* review

* review
closed 2021-07-23 by mspang