project-chip/connectedhomeip #8578: [Android] Replace finalizers with PhantomReference

austinh0, 2021-07-22:
Instead of overriding `finalize()` in Java to perform C++ cleanup, we should use `PhantomReference`.
See https://www.youtube.com/watch?v=7_caITSjk1k

gmarcosb, 2022-06-13:
So, the attached [video presentation says](https://youtu.be/7_caITSjk1k?t=1361):

"Arrange to execute periodically, possibly in a thread dedicated to such things"

I don't know that this gives us something better than using `finalize`, unless we plan to introduce a cleanup thread that runs periodically in matter (which we don't seem to have at the moment)

Given that `finalize` is unlikely to go away anytime soon & that it probably gives us better behavior than introducing a cleanup thread which periodically deletes unused refs, I'm going to push this out (it's definitely not important to have pre 1.0)

yunhanw-google, 2023-11-08:
obsolete, close

closed 2023-11-08 by yunhanw-google