project-chip/connectedhomeip #8606: Access to per-instance commissioning data needs a refactor

tcarmelveilleux, 2021-07-23:
#### Problem

Matter devices needing to be commissioned must rely on the following data, according to the spec, to get through the entire flow:

- Secure storage for a Device Attestation key pair
  - This has to be randomly generated per device with a secure random number generator. Common practice for this in IoT is for generation on-device during manufacturing, followed by injection of DAC after signature challenge by device.
- Storage for the DAC and PAI X.509v3 DER certificates passed to commissioner during commissioning
  - As above, usually done at factory for products that do not have existing alternate IoT protocols in parallel to Matter
- Storage for a signed Certification Declaration (CD) blob
  - This is directly associated with firmware, but is only available after certification of a given version.
- Storage for a SPAKE2+ Verifier (`w0`/`L` tuple and associated salt and iteration count)
  - Currently, the verifier is not stored, but rather passcode is stored, in such a way that the device locally computes the verifier using PBKDF2 on-device at every commissioning, which causes large amount of flash and runtime to be dedicated to generating the verifier that could be precomputed and stored.
  - Currently the PAKE salt is fixed (see `kSpake2pKeyExchangeSalt`), which goes against spec that calls for random-per-device and  16 bytes <= length <= 32 bytes
  - Currently the PAKE iteration counter is fixed to 100 (see `kSpake2p_Iteration_Count`), which goes against spec that calls for 1000..100000 to be used
- Storage of the discovery discriminator, which needs to be random across a distribution of devices, to avoid commissioning discovery collisions

Eventually, storage of the operational key pairs will also have to be done in such a way that they are securely handled to possibly support true storage in secure element/HSM/trustzone, rather than immediate bit storage with possibility of leakage or readout based on side-channel attacks or flash dumping, from devices that may have the ability to securely store them.

Many of the above settings are either not "directly" suitable for persistent storage, or need to vary per instance, in such a way that upon startup and commissionable announcement, the correct data is available and different per-device.

Furthermore, the current ConfigurationManager approach is a global singleton accessed in such a way that it would not be possible to host > 1 commissionable instances (e.g. software-implemented commissionable device, for test or production use) from a single process.

 #### Proposed Solution

This issue is meant to start a conversation, since Device Attestation procedure and PASE are major parts of the commissioning flow that cannot be implemented to-spec until we resolve the issue.

A possible strawman for a solution is:

1. Split some of the functionality of the ConfigurationManager public interface into some sub interfaces with getters, such as:
  - `PAKESecurityConfiguration` with access to the verifier, salt, iteration count and **secure signing primitive using DA secret key**
  - `AttestationCredentials` with access to signing with attestation private key, and access to the DAC, PAI and CD
2. Implement tooling and methodologies to exemplify the generation a-priori of an attestation key pair, and injection of a DAC and PAI, and PAKE config, for a given configuration instance, along with implementation of the accesssors, rather than keeping placeholder "common for all builds" default values.
3. Implement the additional interfaces above, for accessors to the per-instance security data
4. Rename `extern ConfigurationManager & ConfigurationMgr();` to a `GetInstance()` against the ConfigurationMgr.
5. Eventually obtain the `ConfigurationManager` instance from injected Matter stack context (i.e. per-node context) to avoid needing to steer the singleton access to the multiple possible instances based on heuristics or coarse implicit identifiers such as thread IDs.

Steps 1 and 2 can occur in parallel.

Step 3 is the part that will eventually enable to-spec secure PASE and device attestation.

@mleisner @pan-apple @mspang @yyzhong-g @bzbarsky-apple @andreilitvin @woody-apple @msandstedt

tcarmelveilleux, 2021-07-23:
Solving this issue would help with #7738.

tcarmelveilleux, 2021-07-23:
Today's ConfigurationManager has the following refactorable methods that could be used for and DAC, PAI:

```
    CHIP_ERROR GetManufacturerDeviceCertificate(uint8_t * buf, size_t bufSize, size_t & certLen);
    CHIP_ERROR GetManufacturerDeviceIntermediateCACerts(uint8_t * buf, size_t bufSize, size_t & certsLen);
```
These do assume DAC chain length > 3, so some minor refactor is needed for sure.

The following method:
```
CHIP_ERROR GetManufacturerDevicePrivateKey(uint8_t * buf, size_t bufSize, size_t & keyLen);
```
is both not used, and not safe for use: it extricates a private key, and if that private key is held in a secure element or similar trusted security module, this is impossible.

andy31415, 2021-07-27:
@harimau-qirex is working on stabilizing our storage logic (we currently have KVS/PersistedStorage/ConfigurationManager each with own functionality (binary vs non-binary, fixed vs variable keys). Ideally any interface should not extend/depend on a specific item since we plan to have a single way of storing persistent data.

If feasable, I would also expect storage interfaces to be injected instead of global singletons, however I would do this only if it does not couple things too much (propagating a 'storage' class everywhere sounds painful)

tcarmelveilleux, 2021-07-27:
> Ideally any interface should not extend/depend on a specific item since we plan to have a single way of storing persistent data.

I disagree with this completely. I think the problem we have right now is there is a single way of storing persistent data that both prevents some optimizations and also prevents hardware assist of some features such as security.

msandstedt, 2021-07-27:
> @harimau-qirex is working on stabilizing our storage logic (we currently have KVS/PersistedStorage/ConfigurationManager each with own functionality (binary vs non-binary, fixed vs variable keys). [...]

The problem is that, for some security collateral, it's untenable for the sdk to be involved in storage / retrieval.

This was called out as an example:

```
CHIP_ERROR GetManufacturerDevicePrivateKey(uint8_t * buf, size_t bufSize, size_t & keyLen);
CHIP_ERROR StoreManufacturerDevicePrivateKey(const uint8_t * key, size_t keyLen);
```

This implies that the sdk is responsible for storing a key into and retrieving a key from the environment.  That doesn't work.  For cases like this, we need to invert this relationship and instead define APIs for the sdk to use security collateral owned by the platform environment, e.g. 'sign this thing', 'decrypt this thing', 'encrypt this thing'.

mspang, 2021-07-27:
I would view storage as an application concern and believe that the core library should decouple itself from storage. CHIP abstractions should merely exchange data with the application in a way that conforms to protocol requirements without assuming whether or how the data is "stored". For example, an application may be able to reconstitute necessary data from other sources rather than directly store it.

andy31415, 2022-01-31:
Note that #7738 is V1 and is a subset of this. As far as I understand, the reason this did not get a V1 was because of title saying "refactor" but the requirements seem to be V1 requirements:

Need ability to store factory data and read it as such during commissioning.

tcarmelveilleux, 2022-06-28:
The core of this has been fixed overall.

closed 2022-06-28 by tcarmelveilleux