project-chip/connectedhomeip #8636: 2nd CASE exchange fails at HandleSigmaR2 with CHIP_ERROR_NO_MEMORY

mspang, 2021-07-27:
Attempting a 2nd CASE exchange fails at

https://github.com/project-chip/connectedhomeip/blob/a85a567c44251b4f474d20d3e04b107bc530331a/src/credentials/CHIPCert.cpp#L233

with the following stack

```
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff7c4f537 in __GI_abort () at abort.c:79
#2  0x00007ffff5d33a02 in chip::Credentials::ChipCertificateSet::LoadCert(chip::TLV::TLVReader&, chip::BitFlags<chip::Credentials::CertDecodeFlags, unsigned char>, chip::Span<unsigned char const>)
    (this=0xde0518, reader=..., decodeFlags=..., chipCert=...) at ../../src/credentials/CHIPCert.cpp:233
#3  0x00007ffff5d33cd2 in chip::Credentials::ChipCertificateSet::LoadCerts(chip::TLV::TLVReader&, chip::BitFlags<chip::Credentials::CertDecodeFlags, unsigned char>) (this=0xde0518, reader=..., decodeFlags=...) at ../../src/credentials/CHIPCert.cpp:276
#4  0x00007ffff5d33c10 in chip::Credentials::ChipCertificateSet::LoadCerts(unsigned char const*, unsigned int, chip::BitFlags<chip::Credentials::CertDecodeFlags, unsigned char>)
    (this=0xde0518, chipCerts=0x7fffedff9630 "\025\060\001\001\001$\002\001\067\003$\023\001$\025", chipCertsLen=247, decodeFlags=...) at ../../src/credentials/CHIPCert.cpp:262
#5  0x00007ffff5d58fe2 in chip::CASESession::Validate_and_RetrieveResponderID(unsigned char const*, unsigned short, chip::Crypto::P256PublicKey&) (this=
    0xdc6628, responderOpCert=0x7fffedff9630 "\025\060\001\001\001$\002\001\067\003$\023\001$\025", responderOpCertLen=247, responderID=...) at ../../src/protocols/secure_channel/CASESession.cpp:1215
#6  0x00007ffff5d55ad3 in chip::CASESession::HandleSigmaR2(chip::System::PacketBufferHandle&) (this=0xdc6628, msg=...)
    at ../../src/protocols/secure_channel/CASESession.cpp:772
#7  0x00007ffff5d5486b in chip::CASESession::HandleSigmaR2_and_SendSigmaR3(chip::System::PacketBufferHandle&)
    (this=0xdc6628, msg=...) at ../../src/protocols/secure_channel/CASESession.cpp:636
#8  0x00007ffff5d5a1a2 in chip::CASESession::OnMessageReceived(chip::Messaging::ExchangeContext*, chip::PacketHeader const&, chip::PayloadHeader const&, chip::System::PacketBufferHandle&&)
    (this=0xdc6628, ec=0xe13790, packetHeader=..., payloadHeader=..., msg=...)
    at ../../src/protocols/secure_channel/CASESession.cpp:1399
#9  0x00007ffff5ca4e28 in chip::Messaging::ExchangeContext::HandleMessage(chip::PacketHeader const&, chip::PayloadHeader const&, chip::Transport::PeerAddress const&, chip::BitFlags<chip::Messaging::MessageFlagValues, unsigned int>, chip::System::PacketBufferHandle&&) (this=0xe13790, packetHeader=..., payloadHeader=..., peerAddress=..., msgFlags=..., msgBuf=...)
    at ../../src/messaging/ExchangeContext.cpp:460
#10 0x00007ffff5ca78b7 in operator()<chip::Messaging::ExchangeContext>(chip::Messaging::ExchangeContext*) const
    (__closure=0x7fffedffa5e0, ec=0xe13790) at ../../src/messaging/ExchangeMgr.cpp:227
#11 0x00007ffff5ca78f7 in chip::BitMapObjectPool<chip::Messaging::ExchangeContext, 8>::LambdaProxy<chip::Messaging::ExchangeManager::OnMessageReceived(const chip::PacketHeader&, const chip::PayloadHeader&, chip::SecureSessionHandle, const chip::Transport::PeerAddress&, chip::SecureSessionMgrDelegate::DuplicateMessage, chip::System::PacketBufferHandle&&)::<lambda(auto:3*)> >::Call(void *, void *) (context=0x7fffedffa5e0, target=0xe13790) at ../../src/lib/support/Pool.h:139
#12 0x00007ffff5d7ad66 in chip::StaticAllocatorBitmap::ForEachActiveObjectInner(void*, bool (*)(void*, void*))
    (this=0xe13760, context=0x7fffedffa5e0, lambda=0x7ffff5ca78d4 <chip::BitMapObjectPool<chip::Messaging::ExchangeContext, 8>::LambdaProxy<chip::Messaging::ExchangeManager::OnMessageReceived(const chip::PacketHeader&, const chip::PayloadHeader&, chip::SecureSessionHandle, const chip::Transport::PeerAddress&, chip::SecureSessionMgrDelegate::DuplicateMessage, chip::System::PacketBufferHandle&&)::<lambda(auto:3*)> >::Call(void *, void *)>) at ../../src/lib/support/Pool.cpp:96
#13 0x00007ffff5ca7636 in chip::BitMapObjectPool<chip::Messaging::ExchangeContext, 8>::ForEachActiveObject<chip::Messaging::ExchangeManager::OnMessageReceived(const chip::PacketHeader&, const chip::PayloadHeader&, chip::SecureSessionHandle, const chip::Transport::PeerAddress&, chip::SecureSessionMgrDelegate::DuplicateMessage, chip::System::PacketBufferHandle&&)::<lambda(auto:3*)> >(struct {...} &&) (this=0xe13760, function=...) at ../../src/lib/support/Pool.h:128
#14 0x00007ffff5ca700b in chip::Messaging::ExchangeManager::OnMessageReceived(chip::PacketHeader const&, chip::PayloadHeader const&, chip::SecureSessionHandle, chip::Transport::PeerAddress const&, chip::SecureSessionMgrDelegate::DuplicateMessage, chip::System::PacketBufferHandle&&)
    (this=0xe13630, packetHeader=..., payloadHeader=..., session=..., source=..., isDuplicate=chip::SecureSessionMgrDelegate::DuplicateMessage::No, msgBuf=...) at ../../src/messaging/ExchangeMgr.cpp:216
#15 0x00007ffff5cbe1f6 in chip::SecureSessionMgr::MessageDispatch(chip::PacketHeader const&, chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&) (this=0xe12c20, packetHeader=..., peerAddress=..., msg=...)
    at ../../src/transport/SecureSessionMgr.cpp:316
#16 0x00007ffff5cbe115 in chip::SecureSessionMgr::OnMessageReceived(chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&) (this=0xe12c20, peerAddress=..., msg=...) at ../../src/transport/SecureSessionMgr.cpp:305
#17 0x00007ffff5cbfc8e in chip::TransportMgrBase::HandleMessageReceived(chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&) (this=0xdfd400, peerAddress=..., msg=...) at ../../src/transport/TransportMgrBase.cpp:57
#18 0x00007ffff5d4e34b in chip::Transport::Base::HandleMessageReceived(chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&) (this=0xdfd470, source=..., buffer=...) at ../../src/transport/raw/Base.h:92
#19 0x00007ffff5d4e957 in chip::Transport::UDP::OnUdpReceive(chip::Inet::IPEndPointBasis*, chip::System::PacketBufferHandle&&, chip::Inet::IPPacketInfo const*)
    (endPoint=0x7ffff5dc77e8 <chip::Inet::UDPEndPoint::sPool+104>, buffer=..., pktInfo=0x7fffedffabc0)
    at ../../src/transport/raw/UDP.cpp:112
#20 0x00007ffff5d6bf0c in chip::Inet::IPEndPointBasis::HandlePendingIO(unsigned short)
    (this=0x7ffff5dc77e8 <chip::Inet::UDPEndPoint::sPool+104>, aPort=5541) at ../../src/inet/IPEndPointBasis.cpp:1140
#21 0x00007ffff5d73a1d in chip::Inet::UDPEndPoint::HandlePendingIO()
    (this=0x7ffff5dc77e8 <chip::Inet::UDPEndPoint::sPool+104>) at ../../src/inet/UDPEndPoint.cpp:921
#22 0x00007ffff5d739b2 in chip::Inet::UDPEndPoint::HandlePendingIO(chip::System::WatchableSocket&) (socket=...)
    at ../../src/inet/UDPEndPoint.cpp:912
#23 0x00007ffff5cb8708 in chip::System::WatchableSocketBasis<chip::System::WatchableSocket>::InvokeCallback()
    (this=0x7ffff5dc7808 <chip::Inet::UDPEndPoint::sPool+136>) at ../../src/system/WatchableSocket.h:263
#24 0x00007ffff5cb8661 in chip::System::WatchableEventManager::HandleEvents()
    (this=0x7ffff5dc4898 <chip::DeviceLayer::SystemLayer+24>) at ../../src/system/WatchableEventManagerSelect.cpp:223
--Type <RET> for more, q to quit, c to continue without paging--
#25 0x00007ffff5ccbdd1 in chip::DeviceLayer::Internal::GenericPlatformManagerImpl_POSIX<chip::DeviceLayer::PlatformManagerImpl>::_RunEventLoop() (this=0x7ffff5dc5288 <chip::DeviceLayer::PlatformManagerImpl::sInstance+8>)
    at ../../src/include/platform/internal/GenericPlatformManagerImpl_POSIX.cpp:175
#26 0x00007ffff5b9e6d6 in chip::DeviceLayer::PlatformManager::RunEventLoop()
    (this=0x7ffff5dc5280 <chip::DeviceLayer::PlatformManagerImpl::sInstance>)
    at ../../src/include/platform/PlatformManager.h:284
#27 0x00007ffff5b9e70d in (anonymous namespace)::PlatformMainLoop(void*) ()
    at ../../src/controller/python/chip/native/StackInit.cpp:36
#28 0x00007ffff7f94ea7 in start_thread (arg=<optimized out>) at pthread_create.c:477
#29 0x00007ffff7d27def in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

It appears that in Validate_and_RetrieveResponderID we always append to the same credentials set and run out of space after the first exchange:

https://github.com/project-chip/connectedhomeip/blob/a85a567c44251b4f474d20d3e04b107bc530331a/src/protocols/secure_channel/CASESession.cpp#L1214-L1217

@pan-apple

mspang, 2021-07-27:
This can be reproduced with chip-device-ctrl, but not chip-tool because it can only do one exchange and then exits.

Damian-Nordic, 2021-07-27:
Duplicate of https://github.com/project-chip/connectedhomeip/issues/8207?

pan-apple, 2021-07-27:
tag: @jpk233

mspang, 2021-07-27:
Yes, duplicate of #8207

Shame that we knew about this and I ended up debugging again. Who's working on this?

closed 2021-07-27 by mspang