project-chip/connectedhomeip #8637: [Android] Impossible to BLE connect with the MatterDevice

jmeg-sfy, 2021-07-27:
BLE Cras

07-27 10:41:30.041  2051  2171 I chatty  : uid=1000(system) android.anim expire 9 lines
07-27 10:41:30.258  2051  2170 I chatty  : uid=1000(system) android.display expire 21 lines
07-27 10:41:32.486  2051  2368 I chatty  : uid=1000(system) UEventObserver expire 16 lines
07-27 10:41:37.494 17062 17062 E AndroidRuntime: FATAL EXCEPTION: main
07-27 10:41:37.494 17062 17062 E AndroidRuntime: Process: com.google.chip.chiptool, PID: 17062
07-27 10:41:37.494 17062 17062 E AndroidRuntime: chip.devicecontroller.ChipDeviceControllerException: Error 0x0000002F
07-27 10:41:37.494 17062 17062 E AndroidRuntime: 	at chip.devicecontroller.ChipDeviceController.newDeviceController(Native Method)
07-27 10:41:37.494 17062 17062 E AndroidRuntime: 	at chip.devicecontroller.ChipDeviceController.<init>(ChipDeviceController.java:35)
07-27 10:41:37.494 17062 17062 E AndroidRuntime: 	at com.google.chip.chiptool.ChipClient.getDeviceController(ChipClient.kt:35)
07-27 10:41:37.494 17062 17062 E AndroidRuntime: 	at com.google.chip.chiptool.provisioning.DeviceProvisioningFragment$startConnectingToDevice$1.invokeSuspend(DeviceProvisioningFragment.kt:82)
07-27 10:41:37.494 17062 17062 E AndroidRuntime: 	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
07-27 10:41:37.494 17062 17062 E AndroidRuntime: 	at kotlinx.coroutines.DispatchedTask.run(Dispatched.kt:241)
07-27 10:41:37.494 17062 17062 E AndroidRuntime: 	at android.os.Handler.handleCallback(Handler.java:938)
07-27 10:41:37.494 17062 17062 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:99)
07-27 10:41:37.494 17062 17062 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:250)
07-27 10:41:37.494 17062 17062 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:7755)
07-27 10:41:37.494 17062 17062 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
07-27 10:41:37.494 17062 17062 E AndroidRuntime: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:592)
07-27 10:41:37.494 17062 17062 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:958)
07-27 10:41:37.496  2051 17660 I chatty  : uid=1000(system) Binder:2051_3 expire 1 line
07-27 10:41:37.496  2051  5964 I chatty  : uid=1000(system) Binder:2051_13 expire 2 lines
07-27 10:41:37.510  2051  2184 I chatty  : uid=1000(system) ActivityManager expire 2 lines
07-27 10:41:37.561  2051  6881 I chatty  : uid=1000(system) Binder:2051_1E expire 1 line
07-27 10:41:37.562  2051  6871 I chatty  : uid=1000(system) Binder:2051_19 expire 2 lines
07-27 10:41:37.562  2051  2579 I chatty  : uid=1000(system) NsdService expire 2 lines
07-27 10:41:37.607  2051  9909 I chatty  : uid=1000(system) Binder:2051_1F expire 1 line
07-27 10:42:00.403  2051  2318 I chatty  : uid=1000(system) PowerManagerSer expire 12 lines
07-27 10:42:06.407  2051  2051 I chatty  : uid=1000(system) Binder:2051_3 expire 10 lines
07-27 10:42:06.933  2051  2264 I chatty  : uid=1000(system) batterystats-wo expire 9 lines
07-27 10:42:07.144  2051  2165 I chatty  : uid=1000(system) android.fg expire 1 line
07-27 10:42:07.235  2051  2094 I chatty  : uid=1000(system) Binder:2051_2 expire 3 lines
07-27 10:42:07.324  2051  3065 I chatty  : uid=1000(system) Binder:2051_B expire 5 lines
07-27 10:42:51.438  2051  2167 I chatty  : uid=1000(system) android.ui expire 2 lines
07-27 10:42:52.629  2638  2659 V FingerprintManager: onAuthenticationFailed
07-27 10:42:52.630  2638  2638 V FingerprintManager: handleMessage:failed
07-27 10:42:53.221  2051  2051 I chatty  : uid=1000(system) Binder:2051_3 expire 6 lines
07-27 10:42:53.234  2638  3143 V FingerprintManager: onAuthenticationSucceeded
07-27 10:42:53.235  2638  2638 V FingerprintManager: handleMessage:successed
07-27 10:42:53.248  2051  2318 I chatty  : uid=1000(system) PowerManagerSer expire 6 lines
07-27 10:42:53.250  2051  2165 I chatty  : uid=1000(system) android.fg expire 3 lines
07-27 10:42:53.273  2051  2051 I chatty  : uid=1000(system) Binder:2051_3 expire 1 line
07-27 10:42:53.296  2051  2094 I chatty  : uid=1000(system) Binder:2051_2 expire 1 line
07-27 10:42:53.296  2051  3065 I chatty  : uid=1000(system) Binder:2051_B expire 1 line
07-27 10:42:53.302  2051  6881 I chatty  : uid=1000(system) Binder:2051_1E expire 3 lines
07-27 10:42:53.302  2051  2185 I chatty  : uid=1000(system) ActivityManager expire 16 lines
07-27 10:42:53.305  2051  9909 I chatty  : uid=1000(system) Binder:2051_1F expire 1 line
07-27 10:42:53.306  2051  6874 I chatty  : uid=1000(system) Binder:2051_1B expire 2 lines
07-27 10:42:53.371  2051  2184 I chatty  : uid=1000(system) ActivityManager expire 6 lines
07-27 10:42:53.375  2051  2171 I chatty  : uid=1000(system) android.anim expire 8 lines
07-27 10:42:53.500  2051  2358 I chatty  : uid=1000(system) Binder:2051_4 expire 5 lines
07-27 10:42:53.612  2051  2577 I chatty  : uid=1000(system) ConnectivitySer expire 3 lines
07-27 10:42:54.419  2051  2368 I chatty  : uid=1000(system) UEventObserver expire 4 lines
07-27 10:42:54.457  2051  5940 I chatty  : uid=1000(system) Binder:2051_F expire 2 lines
07-27 10:42:54.633  2051  2567 I chatty  : uid=1000(system) Binder:2051_6 expire 3 lines
07-27 10:42:55.538  2051  2170 I chatty  : uid=1000(system) android.display expire 3 lines
07-27 10:42:55.755  2051  2264 I chatty  : uid=1000(system) batterystats-wo expire 23 lines
07-27 10:42:57.538  2051  6005 I chatty  : uid=1000(system) Binder:2051_17 expire 6 lines
07-27 10:42:57.910  2051  5979 I chatty  : uid=1000(system) Binder:2051_16 expire 6 lines
07-27 10:42:57.977  2051  3068 I chatty  : uid=1000(system) Binder:2051_C expire 3 lines
07-27 10:42:58.075  2051  6006 I chatty  : uid=1000(system) Binder:2051_18 expire 2 lines
07-27 10:43:10.684  2051  5961 I chatty  : uid=1000(system) Binder:2051_10 expire 12 lines
07-27 10:43:10.896  2051 10071 I chatty  : uid=1000(system) Binder:2051_20 expire 1 line
07-27 10:43:11.260  2051  5961 I chatty  : uid=1000(system) Binder:2051_10 expire 5 lines
07-27 10:43:12.176  2051  2567 I chatty  : uid=1000(system) Binder:2051_6 expire 11 lines
07-27 10:43:12.176  2051  2577 I chatty  : uid=1000(system) ConnectivitySer expire 5 lines
07-27 10:43:12.284  2051  6005 I chatty  : uid=1000(system) Binder:2051_17 expire 1 line
07-27 10:43:12.285  2051  3065 I chatty  : uid=1000(system) Binder:2051_B expire 6 lines
07-27 10:43:14.187  2051  2170 I chatty  : uid=1000(system) android.display expire 9 lines
[... 227 lines cut ...]
07-27 10:53:59.199  2051  2170 I chatty  : uid=1000(system) android.display expire 5 lines
07-27 10:54:13.298  2051  2171 I chatty  : uid=1000(system) android.anim expire 11 lines
07-27 10:54:16.442 20888 20888 E AndroidRuntime: FATAL EXCEPTION: main
07-27 10:54:16.442 20888 20888 E AndroidRuntime: Process: com.google.chip.chiptool, PID: 20888
07-27 10:54:16.442 20888 20888 E AndroidRuntime: java.lang.IllegalStateException: Unable to get connected device with nodeId 2
07-27 10:54:16.442 20888 20888 E AndroidRuntime: 	at com.google.chip.chiptool.ChipClient$getConnectedDevicePointer$2$1.onConnectionFailure(ChipClient.kt:56)
07-27 10:54:16.442 20888 20888 E AndroidRuntime: 	at chip.devicecontroller.ChipDeviceController.getConnectedDevicePointer(Native Method)
07-27 10:54:16.442 20888 20888 E AndroidRuntime: 	at chip.devicecontroller.ChipDeviceController.getConnectedDevicePointer(ChipDeviceController.java:108)
07-27 10:54:16.442 20888 20888 E AndroidRuntime: 	at com.google.chip.chiptool.ChipClient.getConnectedDevicePointer(ChipClient.kt:45)
07-27 10:54:16.442 20888 20888 E AndroidRuntime: 	at com.google.chip.chiptool.clusterclient.OnOffClientFragment.sendLevelCommandClick(OnOffClientFragment.kt:139)
07-27 10:54:16.442 20888 20888 E AndroidRuntime: 	at com.google.chip.chiptool.clusterclient.OnOffClientFragment$onCreateView$$inlined$apply$lambda$6$1.invokeSuspend(OnOffClientFragment.kt:69)
07-27 10:54:16.442 20888 20888 E AndroidRuntime: 	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
07-27 10:54:16.442 20888 20888 E AndroidRuntime: 	at kotlinx.coroutines.DispatchedTask.run(Dispatched.kt:241)
07-27 10:54:16.442 20888 20888 E AndroidRuntime: 	at android.os.Handler.handleCallback(Handler.java:938)
07-27 10:54:16.442 20888 20888 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:99)
07-27 10:54:16.442 20888 20888 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:250)
[... cut ...]
07-27 10:54:16.442 20888 20888 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:958)
07-27 10:54:16.446  2051 20987 I chatty  : uid=1000(system) Binder:2051_3 expire 1 line
07-27 10:54:16.503  2051  2264 I chatty  : uid=1000(system) batterystats-wo expire 9 lines
07-27 10:54:16.560  2051  2167 I chatty  : uid=1000(system) android.ui expire 2 lines
07-27 10:54:18.433  2051  6006 I chatty  : uid=1000(system) Binder:2051_18 expire 4 lines
07-27 10:54:18.765  2051  6871 I chatty  : uid=1000(system) Binder:2051_19 expire 3 lines

austinh0, 2021-07-27:
These logs seem like they only include `INFO` level and above. Can you copy the `VERBOSE` and above logs?

In case the logs are not enough to tell, it could also be helpful if you could narrow down the point where `CHIP_ERROR_INCORRECT_STATE` (0x00000003) is returned.

austinh0, 2021-08-04:
Fixed by #8666