project-chip/connectedhomeip #8645: CommandHandler::ProcessCommandDataElement broken for group paths

bzbarsky-apple, 2021-07-27:
#### Problem
`CommandHandler::ProcessCommandDataElement` does:
```
    err = commandPath.GetEndpointId(&endpointId);
    SuccessOrExit(err);
```
but this will fail for group paths.  What should instead happen is that if there is no endpoint id we check for a group id (which is NOT in the command path; it's in the packet header, iirc), and then loop over all endpoints in the group doing the command on each endpoint.  

 #### Proposed Solution
Fix things so group commands work, like they used to work in the Ember code.

@yunhanw-google @erjiaqing

yunhanw-google, 2021-07-27:
OK, we would handle it, we also need to remove group id from IM command path in MessageDef

bzbarsky-apple, 2021-07-28:
@mrjerryjohns should probably confirm what the encoding side looks like before we start changing it.

bzbarsky-apple, 2022-01-26:
Fixed in #13833.

closed 2022-01-26 by bzbarsky-apple