project-chip/connectedhomeip #8650: Move to using constructor/destructor for ReadHandler

bzbarsky-apple, 2021-07-27:
#### Problem
We have public APIs on `ReadHandler` (like `Init`, and parts of `OnReadRequest`) that are expected to only be called when the handler tests true for `IsFree`.   This is pretty fragile.

 #### Proposed Solution
Switch to having an actual constructor and destructor for `ReadHandler`.  The constructor would take a delegate and exchange context as arguments.  `Shutdown` would call the destructor.  We'd then need some out-of-band tracking of whether things are free or not, but we have helper classes for object pools that work that way.

That would also lay the groundwork for doing heap-allocation of `ReadHandler`s as needed, instead of pool-allocation.

mrjerryjohns, 2021-12-14:
Working on this already.

holbrookt, 2022-02-01:
@mrjerryjohns any update for this? Does this need to be in for 1.0?

bzbarsky-apple, 2022-02-01:
This was fixed in #13066.

closed 2022-02-01 by bzbarsky-apple