project-chip/connectedhomeip #8655: Android CHIPTool pairing fail during "PROVISION CHIP DEVICE WITH WI-FI"

joonhaengHeo, 2021-07-28:
#### Problem
<I try to paring between Android CHIPTool and chip-all-clusters-app(using Raspberry Pi). But, CHIPTool can't pair completed.
 * Paring complete
 * Android CHIPTool : Pixel 4, Android 11(RQ3A.210705.001)
 * Raspberry Pi : Ubuntu 20.10. bluez version 5.55
 >

When I rolled back the fix (#5593), I confirmed that it works normally. However, it is not working in its current state.

 #### Proposed Solution
<suggested fix, suggested enhancement>

joonhaengHeo, 2021-07-28:
Android Log
07-28 11:06:41.148 D 6832     6853     chip.BluetoothManager:                                MATTER-3584.onCharacteristicWrite: 18ee2ef5-263d-4559-959f-4f9c429f9d11 -> 0 
07-28 11:06:41.148 D 6832     6853     CTL                                                       : Received SubscribeCharacteristic 
07-28 11:06:41.148 D 6832     6853     CTL                                                       : Calling Java SubscribeCharacteristic 
07-28 11:06:41.148 D 6832     6853     BluetoothGatt:                                        setCharacteristicNotification() - uuid: 18ee2ef5-263d-4559-959f-4f9c429f9d12 enable: true 
07-28 11:06:41.171 D 6832     6853     chip.BluetoothManager:                                MATTER-3584.onDescriptorWrite: 00002902-0000-1000-8000-00805f9b34fb -> 253 
07-28 11:06:41.171 E 6832     6853     AndroidChipStack:                                     onDescriptorWrite for 00002902-0000-1000-8000-00805f9b34fb failed with status: 253 
07-28 11:06:41.171 D 6832     6853     BLE                                                       : subscribe complete, ep = 0x79f35b75a8 
07-28 11:06:41.208 D 6832     6853     BluetoothGatt:                                        onConnectionUpdated() - Device=00:1A:7D:DA:71:10 interval=36 latency=0 timeout=500 status=0 
07-28 11:06:56.139 E 6832     6865     BLE                                                       : connect handshake timed out, closing ep 0x79f35b75a8 
07-28 11:06:56.140 E 6832     6865     IN                                                         : Failed to establish BLE connection: Ble Error 0x00000415: Connect handshake timed out 

all-clusters-app Log
[1627367517.248101][14378:14381] CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 517
[1627367517.248288][14378:14381] CHIP:BLE: Write request received debug 0xffff94048500
[1627367517.248537][14378:14378] CHIP:BLE: local and remote recv window sizes = 3
[1627367517.248667][14378:14378] CHIP:BLE: selected BTP version 3
[1627367517.248869][14378:14378] CHIP:BLE: using BTP fragment sizes rx 128 / tx 20.
[1627367532.250743][14378:14378] CHIP:BLE: receive handshake timed out, closing ep 0xaaaab9e1d8d0
[1627367532.250835][14378:14378] CHIP:IN: Clearing BLE pending packets.

austinh0, 2021-08-05:
Dup of #8774

closed 2021-08-05 by austinh0