project-chip/connectedhomeip #8702: CASE causes abort on ESP32

austinh0, 2021-07-29:
CASE causes abort on all-clusters-app/ESP32 after #8137. Tested using Android/chip-tool as controllers.

```
I (26676) chip[IN]: CASE Server received SigmaR1 message. Starting handshake. EC 0x3ffc70a8
V (26686) chip[SC]: Waiting for SigmaR1 msg
I (26686) chip[IN]: CASE Server disabling CASE session setups
V (26696) chip[SC]: Received SigmaR1 msg
V (26696) chip[SC]: Peer assigned session key ID 18
I (26746) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
assertion "uxSemaphoreGetCount(engine_state) == 0 && "SHA engine should be locked"" failed: file "/Users/austinhsieh/tools/esp-idf/components/mbedtls/port/esp32/sha.c", line 283, function: esp_sha_block

abort() was called at PC 0x401841bf on core 0
0x401841bf: __assert_func at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

Backtrace:0x4008fed6:0x3ffe1cf0 0x400904a1:0x3ffe1d10 0x40096966:0x3ffe1d30 0x401841bf:0x3ffe1da0 0x4013517e:0x3ffe1dd0 0x4013293c:0x3ffe1e00 0x40132989:0x3ffe1e20 0x401329e1:0x3ffe1e40 0x400ea601:0x3ffe1e60 0x400f85e5:0x3ffe1e90 0x400f8695:0x3ffe1f90 0x400f9262:0x3ffe1fb0 0x400f7616:0x3ffe1fd0 0x400f7639:0x3ffe1ff0 0x400ed755:0x3ffe2010 0x400ede75:0x3ffe2050 0x400f05a5:0x3ffe20b0 0x400f091b:0x3ffe2120 0x400f0fb5:0x3ffe2170 0x400f6c47:0x3ffe2200 0x400fd15f:0x3ffe2250 0x400fdbbd:0x3ffe22a0 0x400fd382:0x3ffe22c0 0x400ef02e:0x3ffe22f0 0x400f2b21:0x3ffe2310 0x400f2b72:0x3ffe2330 0x400f2d97:0x3ffe2350 0x400f2dc1:0x3ffe23b0 0x400904a9:0x3ffe23d0
0x4008fed6: panic_abort at /Users/austinhsieh/tools/esp-idf/components/esp_system/panic.c:341

0x400904a1: esp_system_abort at /Users/austinhsieh/tools/esp-idf/components/esp_system/system_api.c:106

0x40096966: abort at /Users/austinhsieh/tools/esp-idf/components/newlib/abort.c:46

0x401841bf: __assert_func at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

0x4013517e: esp_sha_block at /Users/austinhsieh/tools/esp-idf/components/mbedtls/port/esp32/sha.c:282 (discriminator 1)

0x4013293c: mbedtls_internal_sha256_process at /Users/austinhsieh/tools/esp-idf/components/mbedtls/port/esp32/esp_sha256.c:226

0x40132989: mbedtls_sha256_update_ret at /Users/austinhsieh/tools/esp-idf/components/mbedtls/port/esp32/esp_sha256.c:323

0x401329e1: mbedtls_sha256_update_ret at /Users/austinhsieh/tools/esp-idf/components/mbedtls/port/esp32/esp_sha256.c:307

0x400ea601: chip::Crypto::Hash_SHA256_stream::AddData(chip::Span<unsigned char const>) at /Users/austinhsieh/repos/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/crypto/CHIPCryptoPALmbedTLS.cpp:234

0x400f85e5: chip::CASESession::SendSigmaR2() at /Users/austinhsieh/repos/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/protocols/secure_channel/CASESession.cpp:568

0x400f8695: chip::CASESession::HandleSigmaR1_and_SendSigmaR2(chip::System::PacketBufferHandle&) at /Users/austinhsieh/repos/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/protocols/secure_channel/CASESession.cpp:379
 (inlined by) chip::CASESession::HandleSigmaR1_and_SendSigmaR2(chip::System::PacketBufferHandle&) at /Users/austinhsieh/repos/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/protocols/secure_channel/CASESession.cpp:376

0x400f9262: chip::CASESession::OnMessageReceived(chip::Messaging::ExchangeContext*, chip::PacketHeader const&, chip::PayloadHeader const&, chip::System::PacketBufferHandle&&) at /Users/austinhsieh/repos/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/protocols/secure_channel/CASESession.cpp:1308

0x400f7616: chip::CASEServer::OnMessageReceived(chip::Messaging::ExchangeContext*, chip::PacketHeader const&, chip::PayloadHeader const&, chip::System::PacketBufferHandle&&) at /Users/austinhsieh/repos/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/protocols/secure_channel/CASEServer.cpp:99

0x400f7639: non-virtual thunk to chip::CASEServer::OnMessageReceived(chip::Messaging::ExchangeContext*, chip::PacketHeader const&, chip::PayloadHeader const&, chip::System::PacketBufferHandle&&) at ??:?

0x400ed755: chip::Messaging::ExchangeContext::HandleMessage(chip::PacketHeader const&, chip::PayloadHeader const&, chip::Transport::PeerAddress const&, chip::BitFlags<chip::Messaging::MessageFlagValues, unsigned int>, chip::System::PacketBufferHandle&&) at /Users/austinhsieh/repos/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/messaging/ExchangeContext.cpp:468

0x400ede75: chip::Messaging::ExchangeManager::OnMessageReceived(chip::PacketHeader const&, chip::PayloadHeader const&, chip::SecureSessionHandle, chip::Transport::PeerAddress const&, chip::SecureSessionMgrDelegate::DuplicateMessage, chip::System::PacketBufferHandle&&) at /Users/austinhsieh/repos/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/messaging/ExchangeMgr.cpp:293

0x400f05a5: chip::SecureSessionMgr::MessageDispatch(chip::PacketHeader const&, chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&) at /Users/austinhsieh/repos/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/transport/SecureSessionMgr.cpp:316
 (inlined by) chip::SecureSessionMgr::MessageDispatch(chip::PacketHeader const&, chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&) at /Users/austinhsieh/repos/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/transport/SecureSessionMgr.cpp:309

0x400f091b: chip::SecureSessionMgr::OnMessageReceived(chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&) at /Users/austinhsieh/repos/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/transport/SecureSessionMgr.cpp:305

0x400f0fb5: chip::TransportMgrBase::HandleMessageReceived(chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&) at /Users/austinhsieh/repos/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/transport/TransportMgrBase.cpp:57

0x400f6c47: chip::Transport::UDP::OnUdpReceive(chip::Inet::IPEndPointBasis*, chip::System::PacketBufferHandle&&, chip::Inet::IPPacketInfo const*) at /Users/austinhsieh/repos/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/transport/raw/Base.h:92
 (inlined by) chip::Transport::UDP::OnUdpReceive(chip::Inet::IPEndPointBasis*, chip::System::PacketBufferHandle&&, chip::Inet::IPPacketInfo const*) at /Users/austinhsieh/repos/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/transport/raw/UDP.cpp:112

0x400fd15f: chip::Inet::IPEndPointBasis::HandleDataReceived(chip::System::PacketBufferHandle&&) at /Users/austinhsieh/repos/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/inet/IPEndPointBasis.cpp:610

0x400fdbbd: chip::Inet::UDPEndPoint::HandleDataReceived(chip::System::PacketBufferHandle&&) at /Users/austinhsieh/repos/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/inet/UDPEndPoint.cpp:778

0x400fd382: chip::Inet::InetLayer::HandleInetLayerEvent(chip::System::Object&, int, unsigned int) at /Users/austinhsieh/repos/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/inet/InetLayer.cpp:1066
[... cut ...] atformManager.h:284
 (inlined by) chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::EventLoopTaskMain(void*) at /Users/austinhsieh/repos/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.cpp:204

0x400904a9: vPortTaskWrapper at /Users/austinhsieh/tools/esp-idf/components/freertos/xtensa/port.c:143
```

austinh0, 2021-07-29:
With @tcarmelveilleux, added some logs:
```
diff --git a/components/mbedtls/port/sha/parallel_engine/sha.c b/components/mbedtls/port/sha/parallel_engine/sha.c
index a87ced376..edfb440cb 100644
--- a/components/mbedtls/port/sha/parallel_engine/sha.c
+++ b/components/mbedtls/port/sha/parallel_engine/sha.c
@@ -39,6 +39,9 @@
 #include "soc/hwcrypto_periph.h"
 #include "driver/periph_ctrl.h"
 
+#include "esp_log.h"
+
+#define TAG "sha"
 /*
      Single spinlock for SHA engine memory block
 */
@@ -148,6 +151,7 @@ static bool esp_sha_lock_engine_common(esp_sha_type sha_type, TickType_t ticks_t
         return false;
     }
 
+    ESP_LOGI(TAG, "engine_lock: %s", pcTaskGetTaskName(xTaskGetCurrentTaskHandle()));
     portENTER_CRITICAL(&engines_in_use_lock);
 
     if (engines_in_use == 0) {
```

Which results in
```
I (87025) chip[EM]: Received message of type 0x30 with vendorId 0x0000 and protocolId 0x0000 on exchange 44255
W (87035) wifi:V (87035) chip[EM]: ec id: 44255, Delegate: 0x0x3ffc820c
<ba-add>idx:1 (ifx:0, b0:95:75:8c:cc:0b), tid:0, ssn:0, winSize:64
I (87045) chip[IN]: CASE Server received SigmaR1 message. Starting handshake. EC 0x3ffc7f68
I (87055) sha: engine_lock: CHIP
I (87055) sha: engine_lock: CHIP
V (87075) chip[SC]: Waiting for SigmaR1 msg
I (87075) chip[IN]: CASE Server disabling CASE session setups
V (87075) chip[SC]: Received SigmaR1 msg
I (87085) sha: engine_lock: CHIP
V (87085) chip[SC]: Peer assigned session key ID 24
I (88085) sha: engine_lock: CHIP
I (88085) sha: engine_lock: CHIP
I (88085) sha: engine_lock: CHIP
I (88085) sha: engine_lock: CHIP
I (88095) sha: engine_lock: CHIP
I (88095) sha: engine_lock: CHIP
assertion "uxSemaphoreGetCount(engine_state) == 0 && "SHA engine should be locked"" failed: file "/Users/austinhsieh/tools/esp-idf/components/mbedtls/port/sha/parallel_engine/sha.c", line 217, function: esp_sha_block

abort() was called at PC 0x4018cdc3 on core 1
0x4018cdc3: __assert_func at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

Backtrace:0x4009189a:0x3ffe6250 0x40091fc5:0x3ffe6270 0x4009867a:0x3ffe6290 0x4018cdc3:0x3ffe6300 0x4013aa2b:0x3ffe6330 0x40138cf4:0x3ffe6350 0x40138d41:0x3ffe6370 0x40138d99:0x3ffe6390 0x400ed889:0x3ffe63b0 0x400fbab1:0x3ffe63e0 0x400fbb61:0x3ffe64e0 0x400fc72e:0x3ffe6500 0x400faae2:0x3ffe6520 0x400fab05:0x3ffe6540 0x400f09fe:0x3ffe6560 0x400f1135:0x3ffe65a0 0x400f387d:0x3ffe6600 0x400f3bf3:0x3ffe6670 0x400f428d:0x3ffe66c0 0x400fa113:0x3ffe6750 0x401011db:0x3ffe67a0 0x40101c39:0x3ffe67f0 0x401013fe:0x3ffe6810 0x400f22f6:0x3ffe6840 0x400f5ff1:0x3ffe6860 0x400f6042:0x3ffe6880 0x400f625a:0x3ffe68a0 0x400f6281:0x3ffe6900 0x4009518d:0x3ffe6920
0x4009189a: panic_abort at /Users/austinhsieh/tools/esp-idf/components/esp_system/panic.c:367

0x40091fc5: esp_system_abort at /Users/austinhsieh/tools/esp-idf/components/esp_system/system_api.c:112
```

With `MBEDTLS_HARDWARE_SHA` set to false in the `menuconfig`, this crash is not observed.

tcarmelveilleux, 2021-07-29:
The issue is:

1. The ESP SHA HW only supports one SHA stream at a time, which needs to be protected by a semaphore, but not always called.
2. CASE requires at some point 2x SHA streams at a time (when extracting a digest)
3. Therefore, "single stream" HW accelerated SHA cannot be used to support CASE to-spec

Solution is two-fold:

1- Short term: disable HW SHA256 on ESP32, so that arbitrarily many running SHA streams can be used. The CPU costs are very low.
2- Consider spec changes to transcript computation for CASE so that 2 SHA contexts at a time are not needed (e.g., no start before a finish, and no SHA context cloning.

Solution 2 may actually cause more/different issues.

@dhrishi Is there a way to disable `MBEDTLS_HARDWARE_SHA` by default on all examples for now?

PR #8718 (2021-07-30): ESP32: Workaround to abort in CASE
#### Problem
What is being fixed?  Examples:
* Fixes #8702 

#### Change overview
Temporarily disable hardware SHA for mbedTLS in all relevant ESP32 examples. Analysis here: https://github.com/project-chip/connectedhomeip/issues/8702#issuecomment-889426630

#### Testing
* Tested that CASE works fine using all-clusters-app
PR #8838 (2021-08-11): ESP32: CASE causes abort with ESP SHA Hardware
#### Problem
This PR fixes #8702.

#### Change overview
1. Added the fix for copying the SHA context.
2. Revert the changes in PR #8718 

#### Testing
Compiled the all-clusters-app with SHA hardware enabled and performed commissioning and cluster control.
closed 2021-07-30 by andy31415