project-chip/connectedhomeip #8716: EnableNetwork fails to use proper SSID

fdelacourt-wyze, 2021-07-30:
#### Problem
on RPi, using Test_event_4 (but I had the same problem for a long time on other branches), when I enable the network using the commands:
zcl NetworkCommissioning AddWiFiNetwork 12344321 0 0 ssid=str:TESTSSID credentials=str:TESTPASSWD breadcrumb=0 timeoutMs=1000
zcl NetworkCommissioning EnableNetwork 12344321 0 0 networkID=str:TESTSSID breadcrumb=0 timeoutMs=1000

The EnableNetwork command fails with the following error:
[1627611143.045050][2652:2652] CHIP:NP: LinuxNetworkProvisioningDelegate: SSID: TESTSSID (process:2652): GLib-GIO-CRITICAL **: 02:12:23.045: g_dbus_proxy_call_sync_internal: assertion 'G_IS_DBUS_PROXY (proxy)' failed
[1627611143.045752][2652:2652] CHIP:DL: wpa_supplicant: failed to add network: TESTSSID: unknown error
[1627611143.045808][2652:2652] CHIP:NP: Failed to connect to WiFi network: CHIP Error 4172 (0x0000104C): Internal error

BUT, if I use the same command with a truncated SSID: TESTSS or TESTSSI, then the command succeeds in both cases and I'm able to resolve and execute zclread properly.

It seems that the matching of the parameters is not verifying that the comparison is exact and seems to fail when it is the same size.

 #### Proposed Solution
Verify and address the potential discrepancies in the comparison of the SSID passed as a parameter with the SSID declared during the AddWiFiNetwork.

erjiaqing, 2021-07-30:
From the log

```
[1627611143.045050][2652:2652] CHIP:NP: LinuxNetworkProvisioningDelegate: SSID: TESTSSID (process:2652): GLib-GIO-CRITICAL **: 02:12:23.045: g_dbus_proxy_call_sync_internal: assertion 'G_IS_DBUS_PROXY (proxy)' failed
```

Seems it is passing the correct SSID to gdbus, I wonder if we met some issues with the dbus side.

Can you use some tools to capture dbus traffic to see if the message is expected on dbus?

fdelacourt-wyze, 2021-07-30:
I can try but I'm not very fluent with with those tools...

erjiaqing, 2021-07-30:
Here is a brief guide of the dbus-monitor https://wiki.ubuntu.com/DebuggingDBus

However, you can try just run `sudo dbus-monitor --system`

And it will print all dbus traffic, which contains the argument of each dbus signal and method call.

fdelacourt-wyze, 2021-07-30:
Yes, that's what I tried but I can't decrypt the content:
```
signal time=1627621624.779310 sender=org.freedesktop.DBus -> destination=:1.24 serial=2 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameAcquired
   string ":1.24"
signal time=1627621624.779601 sender=org.freedesktop.DBus -> destination=:1.24 serial=4 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameLost
   string ":1.24"
signal time=1627621626.041658 sender=:1.19 -> destination=(null destination) serial=408 path=/chipoble/0906/service/c1; interface=org.freedesktop.DBus.Properties; member=PropertiesChanged
   string "org.bluez.GattCharacteristic1"
   array [
      dict entry(
         string "Value"
         variant             array of bytes [
               08 91 7a
            ]
      )
   ]
   array [
   ]
method call time=1627621628.624218 sender=:1.12 -> destination=:1.19 serial=471 path=/chipoble/0906/service/c2; interface=org.bluez.GattCharacteristic1; member=Confirm
signal time=1627621629.797259 sender=:1.19 -> destination=(null destination) serial=409 path=/chipoble/0906/service/c1; interface=org.freedesktop.DBus.Properties; member=PropertiesChanged
   string "org.bluez.GattCharacteristic1"
   array [
      dict entry(
         string "Value"
         variant             array of bytes [
               0d 92 7b 57 00 25 11 08 00 00 00 69 b6 01 00 00 00 00 00 01 5c
               bc 00 00 00 00 00 00 00 1f 75 3c 03 cd fe 62 c0 07 2b bc d3 5b
               97 64 bf ce 03 dc 1e d4 79 0e 0e 70 60 2b 1a 3f 78 fe 0c e5 a7
               4c e4 d3 28 42 df 35 cd f4 ab 23 69 93 6a 33 c9 f2 7a e1 1b 25
               c0 cb 7e 7c dd cb fa 3b
            ]
      )
   ]
   array [
   ]
method call time=1627621629.892372 sender=:1.12 -> destination=:1.19 serial=472 path=/chipoble/0906/service/c2; interface=org.bluez.GattCharacteristic1; member=Confirm
method call time=1627621629.989395 sender=:1.12 -> destination=:1.19 serial=473 path=/chipoble/0906/service/c2; interface=org.bluez.GattCharacteristic1; member=Confirm
signal time=1627621629.992384 sender=:1.19 -> destination=(null destination) serial=410 path=/chipoble/0906/service/c1; interface=org.freedesktop.DBus.Properties; member=PropertiesChanged
   string "org.bluez.GattCharacteristic1"
   array [
      dict entry(
         string "Value"
         variant             array of bytes [
               08 94 7c
            ]
      )
   ]
   array [
   ]
method call time=1627621630.087418 sender=:1.12 -> destination=:1.19 serial=474 path=/chipoble/0906/service/c2; interface=org.bluez.GattCharacteristic1; member=Confirm
method call time=1627621630.184294 sender=:1.12 -> destination=:1.19 serial=475 path=/chipoble/0906/service/c2; interface=org.bluez.GattCharacteristic1; member=Confirm
signal time=1627621630.186647 sender=:1.19 -> destination=(null destination) serial=411 path=/chipoble/0906/service/c1; interface=org.freedesktop.DBus.Properties; member=PropertiesChanged
   string "org.bluez.GattCharacteristic1"
   array [
      dict entry(
         string "Value"
         variant             array of bytes [
               08 96 7d
            ]
      )
   ]
   array [
   ]
method call time=1627621630.282307 sender=:1.12 -> destination=:1.19 serial=476 path=/chipoble/0906/service/c2; interface=org.bluez.GattCharacteristic1; member=Confirm
signal time=1627621632.769240 sender=:1.19 -> destination=(null destination) serial=412 path=/chipoble/0906/service/c1; interface=org.freedesktop.DBus.Properties; member=PropertiesChanged
   string "org.bluez.GattCharacteristic1"
   array [
      dict entry(
         string "Value"
         variant             array of bytes [
               08 97 7e
            ]
      )
   ]
   array [
   ]
```

erjiaqing, 2021-07-30:
> I've tried the dbus-monitor but not getting anything, when I sudo it, I get a "Failed to open connection to session bus: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11" because I'm running over SSH.
> 
> Doing sudo bus-monitor --system gives me the following, which I'm not able to decrypt.

Seems this does not emit anything related to wpa_supplicant.

If you stated the dbus-monitor before running tests, you should be able to see some messages from / to wpa_supplicant.

@kghost Seem this is an issue of dbus? Do you have some take a look at this issue?

kghost, 2021-07-30:
@fdelacourt-wyze would you please verify that ConnectivityManagerImpl::StartWiFiManagement is called properly ?

BTW, do you have full chip log ? It could be helpful to diagnosis the problem.

fdelacourt-wyze, 2021-07-30:
I don't have any of the development tool and I'm using the image from ts4 directly so I'm not sure I can verify those points.

kghost, 2021-07-30:
Can you access the chip log by chip process on the device ?

closed 2022-01-26 by woody-apple