project-chip/connectedhomeip #8725: Linux unit tests can stomp on each other's config files

bzbarsky-apple, 2021-07-30:
Looking at the log at https://github.com/project-chip/connectedhomeip/pull/8138/checks?check_run_id=3198431176 it looks to me like we run unit tests in parallel:
```
[147/224] touch obj/src/lib/support/tests/TestTimeUtils_run.stamp
[148/224] touch obj/src/lib/support/tests/tests_run.stamp
[149/224] python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../../src/platform/tests --default-toolchain=//build/toolchain/host:linux_x64_gcc --current-toolchain=//build/toolchain/host:linux_x64_gcc --touch gen/src/platform/tests/TestCHIPoBLEStackMgr_run.pw_pystamp --capture-output -- ../../third_party/pigweed/repo/pw_unit_test/py/pw_unit_test/test_runner.py --runner ../../third_party/pigweed/repo/targets/host/run_test --test tests/TestCHIPoBLEStackMgr
[150/224] python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../../src/messaging/tests --default-toolchain=//build/toolchain/host:linux_x64_gcc --current-toolchain=//build/toolchain/host:linux_x64_gcc --touch gen/src/messaging/tests/TestExchangeMgr_run.pw_pystamp --capture-output -- ../../third_party/pigweed/repo/pw_unit_test/py/pw_unit_test/test_runner.py --runner ../../third_party/pigweed/repo/targets/host/run_test --test tests/TestExchangeMgr
[151/224] touch obj/src/platform/tests/TestCHIPoBLEStackMgr_run.stamp
[152/224] touch obj/src/messaging/tests/TestExchangeMgr_run.stamp
```
But the Linux storage impl uses fixed filenames in /tmp, and two copies of the stack starting up at the same time can stomp on each other's config files. This causes failures like this:
```
'#0:','PlatformMgr tests'
'#2:','Setup                                                ','PASSED'
[1627653806.820523][17983:17983] CHIP:DL: writing settings to file (/tmp/chip_factory.ini.tmp)
[1627653806.820564][17983:17983] CHIP:DL: failed to rename (/tmp/chip_factory.ini.tmp), No such file or directory (2)
[1627653806.820577][17983:17983] CHIP:DL: Configuration Manager initialization failed: ../../src/platform/Linux
```
in https://github.com/project-chip/connectedhomeip/pull/8138/checks?check_run_id=3202506405 and https://github.com/project-chip/connectedhomeip/pull/8138/checks?check_run_id=3198431176 and https://github.com/project-chip/connectedhomeip/runs/3197311566?check_suite_focus=true and I have seen other ones recently.

Can we somehow serialize the test runs? Or better yet make each test use a different set of filenames? @mspang

kpschoedel, 2021-07-30:
#6811 would solve this.

bzbarsky-apple, 2021-07-30:
> #6811 would solve this.

I don't think so.  That bug is basically about moving the storage elsewhere, not about not sharing storage between different app invocations....

mspang, 2021-07-30:
The "right" thing to do is not to use /tmp.

mspang, 2021-07-30:
Or to at least not to use the _same_ files in /tmp.

bzbarsky-apple, 2021-07-30:
Right, it's the sameness that's a problem.  It doesn't matter where the files are if we are using files and always using the same file....

mspang, 2021-07-31:
"not use /tmp" in my head was "why bother writing the store in unit tests". It's not like anyone is going to read it.

I think we should
- Remove the default paths
- Add a mandatory initializer that either sets the filesystem path or turns off the persistence (in-memory only)

bzbarsky-apple, 2021-09-03:
Might get fixed by https://github.com/project-chip/connectedhomeip/pull/9409

closed 2022-08-26 by bzbarsky-apple