project-chip/connectedhomeip #8741: Device leaks secure session allocated in OnSessionEstablished

mspang, 2021-07-31:
The session allocated at

https://github.com/project-chip/connectedhomeip/blob/e175721458a56423c73872bc329f82fdb6482672/src/controller/CHIPDevice.cpp

is not freed by any of the following APIs

- Device::Reset
- DeviceController::ReleaseDevice
- DeviceCommissioner::UnpairDevice

The only API that frees it is

- Device::CloseSession

After pairing and unpairing 16 times the supply of PeerConnectionStates will be exhausted and further connections fail with CHIP_ERROR_NO_MEMORY (the limit is given by CHIP_CONFIG_PEER_CONNECTION_POOL_SIZE). 

It looks like there are other paths that could result in leaks even if this is fixed, e.g. SendMessage resets mState to `ConnectionState::NotConnected` without closing so if LoadSecureSessionParameters fails it would leak.

@andreilitvin @pan-apple @kghost @bzbarsky-apple Do you understand the intent here?

Also, it seems confusing that "handle" implies ownership in PacketBufferHandle but not with "SecureSessionHandle". We have some quite error prone patterns here..

kghost, 2021-07-31:
It looks like that no one have taken the consideration of the situation that session can be closed.

The `SecureSessionHandle` is used broadly to grab a reference to a session, when the session is closed the handle become a dangling handle. It will break all running exchanges and applications which is holding the dangling handle.

There is part of the reason why I wrote Channel (src/channel), it introduced a `ChannelHandle` which behaves like a `SessionHandle`, but with extra notification by `ChannelDelegate` about the session closed event. It will abort all running exchanges , and notify all applications which is bond to the session, that the underlying session has been closed.

mspang, 2021-07-31:
> It looks like that no one have taken the consideration of the situation that session can be closed.
> 
> The `SecureSessionHandle` is used broadly to grab a reference to a session, when the session is closed the handle become a dangling handle. It will break all running exchanges and applications which is holding the dangling handle.
> 
> There is part of the reason why I wrote Channel (src/channel), it introduced a `ChannelHandle` which behaves like a `SessionHandle`, but with extra notification using `ChannelDelegate` about the session closed event. It will abort all running exchanges , and notify all applications which is bond to the session, that the underlying session has been closed.

Are we going to move to the channels?

mspang, 2021-07-31:
> It looks like that no one have taken the consideration of the situation that session can be closed.
>
> The `SecureSessionHandle` is used broadly to grab a reference to a session, when the session is [...]

@Damian-Nordic implemented a manual close API and hooks for python.. but I do not think he realized that release was also not closing it.

kghost, 2021-07-31:
> Are we going to move to the channels?

I want to have a discussion with Boris and Pankaj about how to enable it, we are hitting more and more problems which can be solved by the channels.

mspang, 2021-07-31:
I believe there are other leaks, e.g. in DeviceCommissioner. SecureSessionHandle first arrives via OnNewConnection but there are a number of instances where that function returns early without storing its argument. If we need to manually call ExpirePairing on every new handle then these are also leaks.

@pan-apple You are the other of these APIs. Can you please describe the design intent here?

cecille, 2022-02-01:
Marking this for secondary because I'm concerned this could lead to crashes. Also because this may no longer have an owner if Pankaj is no longer on matter sdk.

woody-apple, 2022-02-07:
Keeping memory bugs in 1.0

woody-apple, 2022-06-07:
@mspang what work remains here?

bzbarsky-apple, 2022-06-08:
This is still an issue, but the session eviction work @mrjerryjohns is doing should address it...

woody-apple, 2022-11-02:
Issue Scrub: Assigning to @bzbarsky-apple to confirm if this is still an issue or not.

closed 2022-11-02 by mspang