project-chip/connectedhomeip #8743: EFR32 4180B Support

kai0989, 2021-07-31:
#### Problem
As described on Silicon Labs website https://www.silabs.com/development-tools/wireless/slwrb4180a-efr32xg21-wireless-gecko-radio-board the 4180A-board is  obsolet.
The successor is the 4180B which is currently the only board to get. 

 #### Proposed Solution
Could you add board-support for this hardware?

kai0989, 2021-07-31:
I tried also to run the example for 4180A on my 4180B, but i am running in a error:

##RTT Client: ************************************************************ 
###RTT Client: *                                                          * 
###RTT Client: *       (c) 2012 - 2016  SEGGER Microcontroller GmbH       * 
###RTT Client: *                                                          * 
###RTT Client: *     www.segger.com     Support: support@segger.com       * 
###RTT Client: *                                                          * 
###RTT Client: ************************************************************ 
###RTT Client: *                                                          * 
###RTT Client: * SEGGER J-Link RTT Client   Compiled Jul  8 2021 18:23:05 * 
###RTT Client: *                                                          * 
###RTT Client: ************************************************************ 

###RTT Client: -----------------------------------------------
###RTT Client: Connecting to J-Link RTT Server via localhost:19021 ...
###RTT Client: Connected.

SEGGER J-Link V7.50a - Real time terminal output
Silicon Labs J-Link Pro OB compiled Sep 16 2020 17:10:58 V4.0, SN=440228771
Process: JLinkExe
<efr32 > ==================================================
<efr32 > chip-efr32-lighting-example starting
<efr32 > ==================================================
<efr32 > Init CHIP Stack
<detail> [-] Initializing key value store
<detail> [-] KeyValueStore init complete: active keys 0, deleted keys 0, sectors 4, logical sector size 8192 bytes
<info  > [DL] Setting device name to : "EFR32_LIGHT"
<efr32 > Initializing OpenThread stack
<info  > [DL] OpenThread started: OK
<info  > [DL] Setting OpenThread device type to ROUTER
<efr32 > Starting Platform Manager Event Loop
<efr32 > Starting OpenThread task
<efr32 > Starting App Task
<efr32 > Starting FreeRTOS scheduler
<error > HardFault:

SCB->CFSR   0x01000000
SCB->HFSR   0x40000000
SCB->MMFAR  0xe000ed34
SCB->BFAR   0xe000ed38
SP          0x20009720
R0          0x20009794

R1          0x00000002

R2          0x000163af

R3          0xa5a5a5a5

R12         0xa5a5a5a5

LR          0x000172a1

PC          0x0001657c

PSR         0x01

jmartinez-silabs, 2022-05-12:
4180 board is from MG21 family and it lacks RAM memory to support matter applications. We have stopped matter support for that family a while ago.

closed 2022-05-12 by jmartinez-silabs