project-chip/connectedhomeip #8744: EFR32 lighting-app build fails for 4180A

kai0989, 2021-07-31:
#### Problem

On current master the lighting-app build for BRD4180A fails with error message:

hainahouwei, 2021-08-01:
@jmartinez-silabs

jmartinez-silabs, 2021-08-02:
Noted, but BRD4180A cannot currently run matter examples as the RAM usage exceed the available 96kB of ram of the EFR32MG21

kai0989, 2021-08-02:
I tested the BRD4180B with the exampel from last test event, is this also broken due RAM-issue?

jmartinez-silabs, 2021-08-02:
Yes both of those boards use MG21 which only has 96kB of RAM. Can I ask you for what company you work for? Do you have an applications engineer assigned at silabs?

kai0989, 2021-08-02:
I'm working for trilux,  i'm also signed in silabs.

jmartinez-silabs, 2022-01-27:
We stopped supporting BRD4180B (*EFR32MG21 families*) in Matter due to their RAM limitations. Unclear if/when those parts will be supported again in the near future as matter ram usage is way past 96kb currently

closed 2022-01-27 by jmartinez-silabs