project-chip/connectedhomeip #8762: Python controller failed to receive BLE message: CHIP Error 4167 (0x00001047): Version mismatch

Minan-Tuyasmart, 2021-08-03:
@jmartinez-silabs
 #### Problem
python chip-device-ctrl unable to establish BLE connection to EFR32MG12 running lighting-app example, 
Python controller sends :   Failed to receive BLE message: CHIP Error 4167 (0x00001047): Version mismatch 

 #### log
The log of EFR32 lighting example(see the attachment for details)
[lighting example.txt](https://github.com/project-chip/connectedhomeip/files/6923129/lighting.example.txt)
The log of controller 
[MobaXterm_192.168.31.101ubuntu_20210803_144212.txt](https://github.com/project-chip/connectedhomeip/files/6923131/MobaXterm_192.168.31.101ubuntu_20210803_144212.txt)

```
ubuntu@ubuntu:~/connectedhomeip/src/controller/python/out/host/controller/python$ chip-device-ctrl
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 0x000000000001b669
CHIP:ZCL: Using ZAP configuration...
CHIP:ZCL: deactivate report event
CHIP:DL: CHIP task running
CHIP:DL: Platform main loop started.
Chip Device Controller Shell

chip-device-ctrl > ble-scan
2021-08-03 06:41:08,319 ChipBLEMgr   INFO     scanning started
2021-08-03 06:41:08,799 ChipBLEMgr   INFO     Name            = EFR32_LIGHT
2021-08-03 06:41:08,800 ChipBLEMgr   INFO     ID              = d558231f-fb05-389c-bae2-6b797b364d28
2021-08-03 06:41:08,801 ChipBLEMgr   INFO     RSSI            = -23
2021-08-03 06:41:08,803 ChipBLEMgr   INFO     Address         = 5A:44:6F:BE:9D:9C
2021-08-03 06:41:08,807 ChipBLEMgr   INFO     Pairing State   = 0
2021-08-03 06:41:08,808 ChipBLEMgr   INFO     Discriminator   = 3840
2021-08-03 06:41:08,808 ChipBLEMgr   INFO     Vendor Id       = 9050
2021-08-03 06:41:08,808 ChipBLEMgr   INFO     Product Id      = 21324
2021-08-03 06:41:08,811 ChipBLEMgr   INFO     Adv UUID        = 0000fff6-0000-1000-8000-00805f9b34fb
2021-08-03 06:41:08,812 ChipBLEMgr   INFO     Adv Data        = 00000f5a234c53
2021-08-03 06:41:08,813 ChipBLEMgr   INFO
2021-08-03 06:41:19,308 ChipBLEMgr   INFO     scanning stopped
chip-device-ctrl > connect -ble 3840 73141520 1232
Device is assigned with nodeid = 1232
CHIP:BLE: Assigned local session key ID 0
CHIP:DL: TRACE: Bluez mainloop starting Thread
CHIP:EM: SessionEstablishmentExchangeDispatch::SendMessageImpl  mTransportMgr 0x2e952540
CHIP:DL: TRACE: Bus acquired for name C-0fd0
CHIP:IN: Message appended to BLE send queue
CHIP:BLE: Sent PBKDF param request
CHIP:DL: PlatformBlueZInit init success
CHIP:DL: Long dispatch time: 136 ms
CHIP:BLE: BLE removing known devices.
CHIP:BLE: BLE initiating scan.
CHIP:BLE: Device 43:51:15:BA:79:1A does not look like a CHIP device.
CHIP:BLE: Device 10:5A:17:0A:DC:4F does not look like a CHIP device.
CHIP:BLE: Device DC:23:4D:0D:6F:47 does not look like a CHIP device.
CHIP:BLE: Device 50:8A:06:F5:D8:EA does not look like a CHIP device.
CHIP:BLE: Device 7F:D9:2A:F2:BE:68 does not look like a CHIP device.
CHIP:BLE: Device 10:5A:17:0A:D9:6B does not look like a CHIP device.
CHIP:BLE: Device DC:23:4D:0D:79:16 does not look like a CHIP device.
CHIP:BLE: New device scanned: 5A:44:6F:BE:9D:9C
CHIP:BLE: Device discriminator match. Attempting to connect.
CHIP:BLE: Scan complete notification without an active scan.
CHIP:DL: ConnectDevice complete
CHIP:DL: Char1 /org/bluez/hci0/dev_5A_44_6F_BE_9D_9C/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_5A_44_6F_BE_9D_9C/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_5A_44_6F_BE_9D_9C/service000e
CHIP:DL: Char1 /org/bluez/hci0/dev_5A_44_6F_BE_9D_9C/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_5A_44_6F_BE_9D_9C/service000e
CHIP:DL: Char1 /org/bluez/hci0/dev_5A_44_6F_BE_9D_9C/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_5A_44_6F_BE_9D_9C/service000e
CHIP:DL: Char1 /org/bluez/hci0/dev_5A_44_6F_BE_9D_9C/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_5A_44_6F_BE_9D_9C/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_5A_44_6F_BE_9D_9C/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_5A_44_6F_BE_9D_9C/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_5A_44_6F_BE_9D_9C/service0018
CHIP:DL: New BLE connection 0xffff700adb10, device 5A:44:6F:BE:9D:9C, path /org/bluez/hci0/dev_5A_44_6F_BE_9D_9C
CHIP:DL: HandlePlatformSpecificBLEEvent 16385
CHIP:IN: BleConnectionComplete: endPoint 0xffff851acab0
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: HandlePlatformSpecificBLEEvent 16388
CHIP:BLE: subscribe complete, ep = 0xffff851acab0
CHIP:DL: Indication received, conn = 0xffff700adb10
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:BLE: peripheral chose BTP version 3; central expected between 2 and 3
CHIP:BLE: using BTP fragment sizes rx 20 / tx 128.
CHIP:BLE: local and remote recv window size = 3
CHIP:IN: BLE EndPoint 0xffff851acab0 Connection Complete
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0xffff700adb10
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: Indication received, conn = 0xffff700adb10
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0xffff700adb10
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: Indication received, conn = 0xffff700adb10
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0xffff700adb10
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:IN: Failed to receive BLE message: CHIP Error 4167 (0x00001047): Version mismatch
CHIP:DL: Indication received, conn = 0xffff700adb10
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0xffff700adb10
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0xffff700adb10
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0xffff700adb10
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
zcl NetworkCommissioning AddThreadNetwork 1232 0 0 operationalDataset=hex:0e080000000000010000000300000f35060004001fffe0020811111111222222220708fdc0697aa8a1d132051000112233445566778899aabbccddeeff030f4f70656e5468726561642d306236330102123404109bd408608c9859c8abf352575d3958f70c0402a0fff8 breadcrumb=0 timeoutMs=1000CHIP:DL: Indication received, conn = 0xffff700adb10
CHIP:DL: HandlePlatformSpecificBLEEvent 16389

CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0xffff700adb10
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
Failed to establish secure session to device: 4050
CHIP Error 4050 (0x00000FD2): Timeout
chip-device-ctrl > zcl NetworkCommissioning AddThreadNetwork 1232 0 0 operationalDataset=hex:0e080000000000010000000300000f35060004001fffe0020811111111222222220708fdc0697aa8a1d132051000112233445566778899aabbccddeeff030f4f70656e5468726561642d306236330102123404109bd408608c9859c8abf352575d3958f70c0402a0fff8 breadcrumb=0 timeoutMs=1000
An exception occurred during process ZCL command:
CHIP Error 4072 (0x00000FE8): Not connected
chip-device-ctrl > CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0xffff700adb10
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0xffff700adb10
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0xffff700adb10
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
zcl NetworkCommissioning EnableNetwork 1232 0 0 networkID=hex:1111111122222222 breadcrumb=0 timeoutMs=1000
An exception occurred during process ZCL command:
CHIP Error 4072 (0x00000FE8): Not connected
chip-device-ctrl > CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0xffff700adb10
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0xffff700adb10
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0xffff700adb10
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0xffff700adb10
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
```

jmartinez-silabs, 2021-08-04:
Point to either your lighting-app  or python controller were build on different commits. Rebuit both on the same commit and you won't see that issue

Minan-Tuyasmart, 2021-08-05:
@jmartinez-silabs  The commit ids of python controller and lighting-app are the same 903e149bd636a5cd1a8821864cb073b4f7a9d3d8, but  Python controller still sends : Failed to receive BLE message: CHIP Error 4167 (0x00001047): Version mismatch

jmartinez-silabs, 2021-08-05:
Can you try building and runnning the controller as follow and retry ? Never seen this version mismatch error before

```
cd connectedhomeip
scripts/build_python.sh -m platform -d true
source ./out/python_env/bin/activate
chip-device-ctrlscripts/build_python.sh -m platform -d true
source ./out/python_env/bin/activate
chip-device-ctrl
```

closed 2022-01-26 by woody-apple