project-chip/connectedhomeip #8774: Android App failed to commission lighting app - Connect hands hake timed out

fengli, 2021-08-04:
#### Problem
On test-event-4 branch, I was testing android example app to commission lighting app and it failed with `BLE     : connect handshake timed out, closing ep 0x7a059745f0`.

**Setup:**

- Tested with test-event-4 branch.
- Android example app was deployed to Pixel XL with Android 8.1.0.
- Lighting app was built and ran on Raspberry-PI with ubuntu server.

**Relevant output from Android example app:**
Full log see: [device_log.txt](https://github.com/project-chip/connectedhomeip/files/6927911/device_log.txt)

```
08-03 17:34:40.412   956   956 D CTL     : newDeviceController() called
08-03 17:34:40.412   956   956 D TOO     : KVS: Getting key ExampleOpCredsCAKey
08-03 17:34:40.412   956   956 D TOO     : KVS: Getting key afeng ExampleOpCredsCAKey
08-03 17:34:40.412   956   956 D TOO     : KVS: Getting key error 0
08-03 17:34:40.412   956   956 D TOO     : KVS: Getting key ExampleOpCredsICAKey
08-03 17:34:40.412   956   956 D TOO     : KVS: Getting key afeng ExampleOpCredsICAKey
08-03 17:34:40.412   956   956 D TOO     : KVS: Getting key error 0
08-03 17:34:40.413   956   956 D IN      : TransportMgr initialized
08-03 17:34:40.413   956   956 D DIS     : Init admin pairing table with server storage
08-03 17:34:40.413   956   956 D TOO     : KVS: Getting key CHIPAdmin0
08-03 17:34:40.413   956   956 D TOO     : KVS: Getting key afeng CHIPAdmin0
08-03 17:34:40.413   956   956 D TOO     : KVS: Getting key error 0
08-03 17:34:40.413   956   956 D IN      : Loading certs from KVS
08-03 17:34:40.413   956   956 D IN      : local node id is 0x000000000001B669
08-03 17:34:40.413   956   956 D ZCL     : Using ZAP configuration...
08-03 17:34:40.413   956   956 D ZCL     : deactivate report event
08-03 17:34:40.413   956   956 D CTL     : Getting operational keys
08-03 17:34:40.413   956   956 D CTL     : Generating credentials
08-03 17:34:40.413   956   956 D CTL     : Loaded credentials successfully
08-03 17:34:40.413   956   956 D TOO     : KVS: Getting key StartKeyID
08-03 17:34:40.413   956   956 D TOO     : KVS: Getting key afeng StartKeyID
08-03 17:34:40.413   956   956 D TOO     : KVS: Getting key error 0

...

08-03 17:34:44.337   956  1024 D BluetoothGatt: onClientConnectionState() - status=0 clientIf=8 device=DC:A6:32:EC:31:F1
08-03 17:34:44.338   956  1024 I chip.BluetoothManager|onConnectionStateChange: Discovering Services...
08-03 17:34:44.338   956  1024 D BluetoothGatt: discoverServices() - device: DC:A6:32:EC:31:F1

...

08-03 17:34:45.800   956  1024 D BluetoothGatt: onConfigureMTU() - Device=DC:A6:32:EC:31:F1 mtu=131 status=0
08-03 17:34:45.800   956  1024 D chip.BluetoothManager: MTU changed: connecting to CHIP device
08-03 17:34:45.819   956   956 D ChipDeviceController: Bluetooth connection added with ID: 1
08-03 17:34:45.820   956   956 D ChipDeviceController: Pairing device with ID: 2
08-03 17:34:45.820   956   956 D CTL     : pairDevice() called with device ID, connection object, and pincode
08-03 17:34:45.820   956   956 D TOO     : KVS: Getting key ListPairedDevices0
08-03 17:34:45.820   956   956 D TOO     : KVS: Getting key afeng ListPairedDevices0
08-03 17:34:45.820   956   956 D KeyValueStoreManager: Key 'ListPairedDevices0' not found in shared preferences
08-03 17:34:45.820   956   956 D TOO     : KVS: Getting key error 4016
08-03 17:34:45.822   956   956 D IN      : BleConnectionComplete: endPoint 0x7a059745f0
08-03 17:34:45.822   956   956 D CTL     : Received GetMTU
08-03 17:34:45.822   956   956 D CTL     : Calling Java onGetMTU
08-03 17:34:45.822   956   956 D AndroidChipStack: Android Manufacturer: (Google)
08-03 17:34:45.822   956   956 D AndroidChipStack: Android Model: (Pixel XL)
08-03 17:34:45.822   956   956 D CTL     : Received SendCharacteristic
08-03 17:34:45.822   956   956 D CTL     : Calling Java SendCharacteristic
08-03 17:34:45.824   956   956 D SC      : Assigned local session key ID 3
08-03 17:34:45.824   956   956 D IN      : Message appended to BLE send queue
08-03 17:34:45.824   956   956 D SC      : Sent PBKDF param request
08-03 17:34:45.824   956   956 D TOO     : KVS: Setting key StartKeyID
08-03 17:34:45.846   956  1024 D chip.BluetoothManager: MATTER-3840.onCharacteristicWrite: 18ee2ef5-263d-4559-959f-4f9c429f9d11 -> 0
08-03 17:34:45.846   956  1024 D CTL     : Received SubscribeCharacteristic
08-03 17:34:45.846   956  1024 D CTL     : Calling Java SubscribeCharacteristic
08-03 17:34:45.846   956  1024 D BluetoothGatt: setCharacteristicNotification() - uuid: 18ee2ef5-263d-4559-959f-4f9c429f9d12 enable: true
08-03 17:34:45.875   956   986 D chip.BluetoothManager: MATTER-3840.onDescriptorWrite: 00002902-0000-1000-8000-00805f9b34fb -> 253
08-03 17:34:45.876   956   986 E AndroidChipStack: onDescriptorWrite for 00002902-0000-1000-8000-00805f9b34fb failed with status: 253
08-03 17:34:45.876   956   986 D BLE     : subscribe complete, ep = 0x7a059745f0

08-03 17:35:00.823   956  1019 E BLE     : connect handshake timed out, closing ep 0x7a059745f0
08-03 17:35:00.823   956  1019 E IN      : Failed to establish BLE connection: Ble Error 6021 (0x00001785): Connect handshake timed out
08-03 17:35:00.823   956  1019 D IN      : Clearing BLE pending packets.
08-03 17:35:00.823   956  1019 D CTL     : Received UnsubscribeCharacteristic
08-03 17:35:00.824   956  1019 D CTL     : Calling Java UnsubscribeCharacteristic
08-03 17:35:00.826   956  1019 D BluetoothGatt: setCharacteristicNotification() - uuid: 18ee2ef5-263d-4559-959f-4f9c429f9d12 enable: false
08-03 17:35:00.914   671   865 I nanohub : osLog: [AR_CHRE] still: 49
08-03 17:35:00.915   671   865 I nanohub : osLog: [AR_CHRE] ON => IDLE
08-03 17:35:00.920   671   865 I nanohub : osLog: [AR_CHRE] IDLE => ON
08-03 17:35:01.070   956   986 D chip.BluetoothManager: MATTER-3840.onDescriptorWrite: 00002902-0000-1000-8000-00805f9b34fb -> 0
08-03 17:35:01.070   956   986 D BLE     : Auto-closing end point's BLE connection.
08-03 17:35:01.071   956   986 D CTL     : Received CloseConnection
08-03 17:35:01.071   956   986 D CTL     : Calling Java CloseConnection
08-03 17:35:01.071   956   986 D ChipDeviceController: Closing GATT and removing connection for 1
08-03 17:35:01.071   956   986 D BluetoothGatt: close()
08-03 17:35:01.071   956   986 D BluetoothGatt: unregisterApp() - mClientIf=8
08-03 17:35:01.075   956   986 D ChipDeviceController: Calling onCloseBleComplete()
```

**Relevant output from lighting app**

```
[1628028783.418380][107509:107512] CHIP:DL: Device 7C:A2:E7:D1:DA:A8 (Path: /org/bluez/hci0/dev_7C_A2_E7_D1_DA_A8) Connected
[1628028785.956335][107509:107512] CHIP:DL: BluezCharacteristicAcquireWrite is called, conn: 0xffff8002fb90
[1628028785.958532][107509:107512] CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 517
[1628028785.958747][107509:107512] CHIP:BLE: Write request received debug 0xffff8002fb90
[1628028785.958990][107509:107509] CHIP:BLE: local and remote recv window sizes = 3
[1628028785.959062][107509:107509] CHIP:BLE: selected BTP version 3
[1628028785.959107][107509:107509] CHIP:BLE: using BTP fragment sizes rx 128 / tx 20.
[1628028800.961313][107509:107509] CHIP:BLE: receive handshake timed out, closing ep 0xaaaacf09a318
[1628028800.961407][107509:107509] CHIP:IN: Clearing BLE pending packets.
[1628028800.961458][107509:107509] CHIP:BLE: Releasing end point's BLE connection back to application.
[1628028800.961510][107509:107509] CHIP:BLE: Got notification regarding chip connection closure
[1628028802.336948][107509:107512] CHIP:DL: Bluez disconnected
[1628028802.337035][107509:107512] CHIP:DL: Bluez notify CHIPoBluez connection disconnected
```

**Notes**

Tested also with chip-tool running on macbook and lighting-app on Raspberry-PI, the commission succeeded.

 #### Proposed Solution
N/A

austinh0, 2021-08-04:
This is the same as #7078, but for Linux (Bluez).

```
08-03 17:34:45.875   956   986 D chip.BluetoothManager: MATTER-3840.onDescriptorWrite: 00002902-0000-1000-8000-00805f9b34fb -> 253
08-03 17:34:45.876   956   986 E AndroidChipStack: onDescriptorWrite for 00002902-0000-1000-8000-00805f9b34fb failed with status: 253
```
253 = 0xFD. In the [bluedroid source](https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-cts-5.1_r17/stack/include/gatt_api.h#67) we can see that this represents `Client Characteristic Configuration Descriptor Improperly Configured`. I guess we need some Linux platform changes?

jmartinez-silabs, 2021-08-04:
Android chip-tool was not working with efr32 at the time of TE4 branch. 
#8666 fixed it on master. For TE4 use python controller or chip-tool on your computer

fengli, 2021-08-04:
@jmartinez-silabs will the fix for EFR32 also fix lighting-app on Linux? I saw there's some platform specific change for EFR32 in the fix for #8666

jmartinez-silabs, 2021-08-05:
@fengli the fix i linked only applies to efr32. I do not know if it is the same issue with linux app

austinh0, 2021-08-05:
Looking at the code now. Unfortunately I don't have a Linux machine to test on; I ordered a raspberry pi but it will take a couple days to arrive.

TE-N-ElvenWang, 2021-08-30:
Hi, is there any progress with this issue?
I see this still not fixed on Aug 30's master tree.

Thanks.

austinh0, 2021-08-30:
@Damian-Nordic Do you have any update here?

xylophone21, 2021-09-06:
I meet the same issue, any update on this?

Damian-Nordic, 2021-09-07:
Hi, I tried the following changes yesterday to support BLE notifications on the Linux peripheral side:
```
diff --git a/src/platform/Linux/bluez/Helper.cpp b/src/platform/Linux/bluez/Helper.cpp
index f784eb56c..8929f12b7 100644
--- a/src/platform/Linux/bluez/Helper.cpp
+++ b/src/platform/Linux/bluez/Helper.cpp
@@ -1227,8 +1227,8 @@ exit:
 static void BluezPeripheralObjectsSetup(gpointer apClosure)
 {
 
-    static const char * const c1_flags[] = { "write", nullptr };
-    static const char * const c2_flags[] = { "read", "indicate", nullptr };
+    static const char * const c1_flags[] = { "write", "write-without-response", nullptr };
+    static const char * const c2_flags[] = { "read", "indicate", "notify", nullptr };
     static const char * const c3_flags[] = { "read", nullptr };
 
     BluezEndpoint * endpoint = static_cast<BluezEndpoint *>(apClosure);
```

Unfortunately, it's not enough and although the processing goes a little bit further, Bluez somehow doesn't deliver the second message to the app. I'm afraid I won't have time to investigate it further this week, but I'm pasting the diff in case someone else has spare time to take a look into that.

TE-N-ElvenWang, 2021-09-08:
Get the application work if revert below change.
Tested on Linux and K32W Lock.

commit bc3d6bf593ed27592a939a58a056776b6a5407bb
Author: Robert Farnum <robert_farnum@comcast.com>
Date:   Tue Apr 6 21:52:32 2021 -0500

    Android: use ENABLE_NOTIFICATION_VALUE instead of ENABLE_INDICATION_VALUE (#5784)

    * Fix Issue#5593

    * Fix Issue#5593

    * Merge with master

    * issue#5593 merge with master

    * issue#5593 do not reformat code

    * issue#5593 apply patch failure on second part

    * Restyled by whitespace

    Co-authored-by: Restyled.io <commits@restyled.io>

TE-N-ElvenWang, 2021-09-28:
Hi everyone,
    Looks there is still no formal fix of this issue. The indication and notification in BLE seems still not align in every platforms.

yunhanw-google, 2023-08-09:
not reproduced

closed 2023-08-09 by yunhanw-google