project-chip/connectedhomeip #8791: [build] Build failure 2021-08-04

turon, 2021-08-04:
#### Problem

Pulling latest version of tree fails to build:

```
$ ./gn_build.sh 

Traceback (most recent call last):
  File "/home/user/src/vendor/connectedhomeip/master/third_party/pigweed/repo/pw_cli/py/pw_cli/plugins.py", line 87, in from_name
    module = importlib.import_module(module_name)
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/user/src/vendor/connectedhomeip/master/third_party/pigweed/repo/pw_watch/py/pw_watch/watch.py", line 50, in <module>
    import httpwatcher  # type: ignore
ModuleNotFoundError: No module named 'httpwatcher'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/user/src/vendor/connectedhomeip/master/.environment/pigweed-venv/bin/pw", line 11, in <module>
    load_entry_point('pw-cli', 'console_scripts', 'pw')()
  File "/home/user/src/vendor/connectedhomeip/master/third_party/pigweed/repo/pw_cli/py/pw_cli/__main__.py", line 41, in main
    pw_command_plugins.register(args.directory)
  File "/home/user/src/vendor/connectedhomeip/master/third_party/pigweed/repo/pw_cli/py/pw_cli/pw_command_plugins.py", line 57, in register
    _register_builtin_plugins(_plugin_registry)
  File "/home/uer/src/vendor/connectedhomeip/master/third_party/pigweed/repo/pw_cli/py/pw_cli/pw_command_plugins.py", line 37, in _register_builtin_plugins
    registry.register_by_name('watch', 'pw_watch.watch', 'main')
  File "/home/user/src/vendor/connectedhomeip/master/third_party/pigweed/repo/pw_cli/py/pw_cli/plugins.py", line 261, in register_by_name
    Plugin.from_name(name, module_name, member_name, source))
  File "/home/user/src/vendor/connectedhomeip/master/third_party/pigweed/repo/pw_cli/py/pw_cli/plugins.py", line 89, in from_name
    raise Error(f'Failed to import module "{module_name}"') from err
pw_cli.plugins.Error: Failed to import module "pw_watch.watch" (ModuleNotFoundError: No module named 'httpwatcher')

```

saurabhst, 2021-08-04:
Did you pull in all the submodules as well? I got the same error last night but after pulling in submodules I was able to build .

turon, 2021-08-04:
Ok.  I thought `./gn_build.sh` is supposed to do that for you.

turon, 2021-08-05:
Chatting offline, both saurab and I ended up doing a fresh clone to resolve this.  

We are unblocked and this bug is low priority. However it would be good to know whether there is a command sequence that can fully update an existing tree to the latest and to know what sorts of operations cause this state to occur.  It vaguely seems related to submodules going out of sync.

yinyihu-silabs, 2022-03-08:
Similar issue see again since commit ea177792c.
Forced submodule upgrade didn't fix the issue. The only solution was similar to above comment to delete and re-clone the repo.

closed 2022-09-22 by stale[bot]

nmtoan91, 2024-01-15:
The latest version of chip-tool didn't work for me. I tried to clone an old old version and I got this error.