project-chip/connectedhomeip #8808: the iOS demo in Chiptool can not be builded, why create a so complex project for iOS! not a good way to spread CHIP

tigline, 2021-08-05:
#### Problem
<what's wrong or missing, please include any applicable:
 1. can not build the iOS project
 
[image]
2. even remove git submodule update --init
 it will need the ninja and gn tools. that really hard to install. I have tried many times.
 >

 #### Proposed Solution
<suggested fix, suggested enhancement>
I hope the iOS project should be recreated by the default way. all need files should be in the project. not need run shell scripts when build. that real stops many people to use it.

tigline, 2021-08-05:
So anyone who can run the iOS demo? how to fix

tewarid, 2021-08-06:
I am able to build and use ChipTool for iOS without resorting to gn/ninja command line. Open src/darwin/Darwin.xcworkspace in Xcode (from command line `open src/darwin/Darwin.xcworkspace`) and build CHIP Framework first (selected by default) then build or run CHIP Tool App.

woody-apple, 2022-01-26:
We're in progress of reviewing older issues. Is this still an issue given how old the SDK this was tested against? Please re-open if so. Thanks!

closed 2022-01-26 by woody-apple

phil-margetson, 2022-11-18:
make sure that you've cloned the repo rather than just downloading the zip