project-chip/connectedhomeip #8810: Run 'source scripts/activate.sh' failed, failed to access 'chrome-infra-packages.appspot.com'

KenneyQu, 2021-08-05:
#### Problem
I cloned the matter repo, ready the submodules, then run 'source scripts/activate.sh', but always got error:
**OSError: [Errno 101] Network is unreachable**, After analyisis, I found that I can nor access 'chrome-infra-packages.appspot.com', anyone can reproduce this or meet this issue before.

Env:
    Commit:
    commit 203b25d94211952e52b2bcd991218151dd0c9777 (HEAD -> master, origin/master, origin/HEAD)
    Author: Song GUO <songguo@google.com>
    Date:   Thu Aug 5 15:11:33 2021 +0800

    [cloudbuild] Add compress flag to compress built artifacts (#8781)

[image]

KenneyQu, 2021-08-05:
I checked that it seems the address 'chrome-infra-packages.appspot.com' can not work for China network. Anyone can help?

KenneyQu, 2021-08-06:
I double checked this again, finally get a proxy from NXP IT to fix this.
This may bring a trouble for Chinese customer, if we can have a default proxy in the code for China network, it will be great.
Comments?

hainahouwei, 2021-08-07:
FQ吧，老兄

woody-apple, 2022-01-26:
@andy31415 ?

niulei0514, 2022-06-17:
# Put CIPD client in tools so that users can easily get it in their PATH.
CIPD_HOST = 'chrome-infra-packages.appspot.com'
源码中的 这个地址确实对中国用户很不友好........  要么找代理  要么翻墙...  否则没办法构建.......

closed 2022-12-23 by stale[bot]