project-chip/connectedhomeip #8829: Sort out our usage of operator new

bzbarsky-apple, 2021-08-06:
#### Problem
When doing a non-overridden non-placement `new` without passing `noexcept`, while compiling with `-fno-exceptions` (as we do), OOM will ... do something.  Probably not something we would like, like crash.

 #### Proposed Solution
Need to figure out what the options are here.  Maybe we can have all CHIP code ensure that we're passing `noexcept`, though I'm not sure how best to do that.  Or we can forbid use of global non-placement `operator new` (and require people to use `Platform::New` or class-specific `new` that never throws.  But enforce this how?

@msandstedt @mspang @andy31415

msandstedt, 2021-08-06:
This will touch things like https://github.com/project-chip/connectedhomeip/pull/8299.  I gather this could be addressed by adding `noexcept` new operators to objects and calling these in the allocator.

Alternatively, I suppose we could just understand that if something like `#define CHIP_SYSTEM_CONFIG_POOL_USE_HEAP` and `-fno-exceptions` are used together, you just need to accept that program abort is (I assume) the result of a failed allocation.  

I can also imagine having `#define CHIP_SYSTEM_CONFIG_POOL_USE_HEAP` and not using `-fno-exceptions` as being a reasonable configuration for some more capable targets.

kpschoedel, 2021-08-09:
Base CHIP should never use non-placement `new`. Platforms can choose their own mistakes :-)

> But enforce this how?

`nm -Cu libCHIP.a | grep 'operator new'`

Also catches indirect uses (e.g. `std::string`).

bzbarsky-apple, 2021-08-11:
> Base CHIP should never use non-placement new

I think it should be fine to use non-placement new with classes that have an overridden operator new that uses a pool or heap or whatever it wants.

> `nm -Cu libCHIP.a | grep 'operator new'`

Ah, interesting.  If we changed that to ignore `something::operator new` we could still allow overridden operators....

bzbarsky-apple, 2021-08-11:
Of course right now libCHIP _does_ pull in `operator new`...

kpschoedel, 2021-08-11:
`src/setup_payload/` and `src/app/server/OnboardingCodesUtil.cpp` use `std::string`.

mrjerryjohns, 2022-01-31:
Running 'nm' still reveals a number of users of operator new.

mrjerryjohns, 2022-01-31:
Requesting secondary triage since I'm not sure how critical this is for v1.0.

woody-apple, 2022-02-07:
Agree, moving post v1.