project-chip/connectedhomeip #8831: Building this example over ARM 32-bit architecture not working?

saumildmehta, 2021-08-06:
I have Raspberry pi 4B+ with ubuntu 21.04 32-bit server image running.
But chip compilation gives error and it looks like it supports arm64 architecture only.

Setting up CIPD package manager...skipped (0.1s)
    UnsupportedPlatform('linux-armv7l')
    abandoning CIPD setup

I don't have option to switch my OS to 64-bit. Any guide to build for 32-bit?

bzbarsky-apple, 2021-08-10:
We absolutely compile for all sorts of 32-bit ARM architectures.

@mspang do you know offhand what that ` UnsupportedPlatform('linux-armv7l')` bit from CIPD means and what needs to be done there?

turon, 2022-02-01:
We should support 32-bit raspi for v1, though not blocking.
Removed tag `v1_triage_split_4`.

woody-apple, 2022-02-07:
The base case works on pi, moving post v1.

woody-apple, 2023-06-24:
Is this still happening for you? Please re-open if so.

closed 2023-06-24 by woody-apple