project-chip/connectedhomeip #8844: Device identities not derived from Certificates

samadDotDev, 2021-08-06:
#### Problem

During commissioning, commissionees are inferring nodeId (for accessories themselves) from destination nodeId of incoming packets, as in `src/transport/SecureSessionMgr.cpp`
```cpp
// TODO: Remove temporary code once AddOptCert is implemented
if (packetHeader.GetDestinationNodeId().HasValue())
{
    localNodeId = packetHeader.GetDestinationNodeId().Value();
    if (localNodeId != kUndefinedNodeId && fabric->GetNodeId() != localNodeId)
    {
        fabric->SetNodeId(localNodeId);
        ChipLogProgress(Inet, "Setting nodeID %" PRIX64 " on fabric.", fabric->GetNodeId());
        modifiedFabric = true;
    }
}
```

This causes inconsistency especially with the fact that [some part of SDK](https://github.com/project-chip/connectedhomeip/blob/81a9fe6a71ee58cd5d29a6d038833746f2a68d51/src/protocols/secure_channel/CASESession.cpp#L1058) is now deriving the nodeIds from certificates (the right way): `nodeOperationalCertificate.mSubjectDN.GetCertChipId(nodeId)`

This doesn't break the in-tree examples since the destination nodeId is set by commissioner and is same as the one (unreasonably) expected to be incorporated in certificates during NOC request, even with the new API [`setNodeIdForNextNextNOCRequest(nodeId)` then `generateNOCChain(csr, ...)` has the same behavior in that sense as `generateNOCChain(nodeId, csr, ...)`].

This issue is also intented to highlight [the previous comment at a related issue](https://github.com/project-chip/connectedhomeip/issues/8227#issuecomment-887096419).

With the [recent changes](https://github.com/project-chip/connectedhomeip/pull/8137) and inconsistencies in nodeId for devices, this issue causes inconsistent DestinationIDs (for SigmaR1) and breaks CASE establishment for infrastructures that generate unique nodeIds during NOC allocation (instead of incorporating nodeIds provided by the commissioner into NOC).

Also, fabricId is not retrieved from certificates either and devices use a hard-coded value (or a test identity when flag `DEVICE_CONFIG_ENABLE_TEST_DEVICE_IDENTITY` is set) from configuration manager.

 #### Proposed Solution

Derive identities from certificates once provisioned, and use the same for advertising the device.

bzbarsky-apple, 2021-08-09:
@pan-apple is actively working on this.  See https://github.com/project-chip/connectedhomeip/pull/8826, which I expect to fix this...  @samadDotDev please check whether it does?

closed 2021-08-12 by bzbarsky-apple

samadDotDev, 2021-08-24:
This ticket was intended to also cover the case of fabric identity out of certificates, but yeah the mentioned PR solved the node identity part which was more of a blocker as discussed in the issue details. We can create a separate ticket for fabric identity/reference but I suspect there are more pre-requisites to that issue which are being dealt separately.