project-chip/connectedhomeip #8856: ESP32 All Clusters Example attribute reporting using chip-tool not working

Ache17, 2021-08-09:
I'm new to the chip/matter project and this is my first issue, so I don't know whether attribute reporting is implemented and  I might have misunderstood some concepts ( If I did I would be grateful for an explanation of what I'm doing wrong ). 
I'm trying to make attribute reporting of "measured-value" of "temperature measurement" cluster to work ( for mock of temperature sensor ). Firstly I'm trying to do it on ESP32-DevKitC but my main interest is to implement it on the Silicon Labs BRD4161A. I would appreciate any tips/advice/instructions on how to do it ( I'm fairly new to the repository, and so far I managed only to play with few examples using ESP32 and BRD4161A and modify some stuff).

## Problem
The attribute reporting for measured-value of "temperaturemeasurement" doesn't work using chip-tool and the software from the "all-clusters-app"/"ESP32" for ESP32-DevKitC.

## Expected behavior
My understanding is, that after I send the report command I should be able to listen for the scheduled data reports from the device ( like reading temperature every few seconds ).  

## Actual behavior 
I'm sending the report command to flashed device ( paired over ble-wifi command from chip-tool ) and getting ZCL_STATUS_SUCCESS. 
It seems the command is applied because in logs for the device there is a log every 2 seconds that "sched report event in 2000 ms". 
When ran the "chip-tool reporting listen" I'm getting the ZCL_STATUS_FAILURE ( also the cluster IDs doesn't match, I though it would be the 0x402 for "temperature cluster" rather than 0x28 for "ping" ).
All logs are attached ( also attaching reading temperature command just to prove the app is working normally ).
I also checked on ZAP and in Endpoint 1, Temperature Measurement the "measured value" from "Attribute Reporting" is toggled on.

## steps to reproduce 
Compiling the chip-tool ( examples/chip-tool)
```
source third_party/connectedhomeip/scripts/activate.sh 
gn gen out/debug
ninja -C out/debug
```
Compiling the ESP32 app ( examples/all-clusters-app/esp32 ). Board connected via USB is ESP32-DevKitC
```
idf.py set-target esp32
idf.py build
idf.py -p /dev/ttyUSB0 flash monitor
```
Communicating with the device ( from examples/chip-tool folder)
```
/out/debug/chip-tool pairing ble-wifi my-wifi-SSID my-wifi-password 0 20202021 3840
out/debug/chip-tool temperaturemeasurement report measured-value 1 2 10 1  [ log attached, ZCL_SUCCESS ]
out/debug/chip-tool reporting listen 1 [log attached, ZCL_FAILURE]
```

## System configuration
Ubuntu 20.4 laptop 

## screenshots, images, logs, etc. 
just reading temperature, just to show it's normally working 
[read-temperature.txt](https://github.com/project-chip/connectedhomeip/files/6954242/read-temperature.txt)

sending the report command
[request-report.txt](https://github.com/project-chip/connectedhomeip/files/6954243/request-report.txt)

running the listening command
[report-listening.txt](https://github.com/project-chip/connectedhomeip/files/6954349/report-listening.txt)

esp32 log
[esp32-log.txt](https://github.com/project-chip/connectedhomeip/files/6954350/esp32-log.txt)

 # Proposed Solution
Could you provide information on whether what I'm doing is possible and good with the CHIP specification? 
If I'm doing anything wrong, could you provide some instructions/information about how to make it work? 
I would be also grateful if you could provide information/instructions about the openThread version ( I added using ZAP the temperature cluster for the lighting example for efr32 BRD4161A but there I'm using python controller, and I couldn't figure any way of reading the reports ( same for ESP32, just reading the temperature is working )).

bzbarsky-apple, 2021-08-09:
@Ache17 A few things:

1.  Reporting in the codebase is not based on the CHIP spec yet; it's based on ZCL.  So the CHIP specification is not really relevant here yet.
2. To properly set up reporting for now you need to do two things: configure reporting of the attribute in terms of the min/max reporting intervals, min-change, etc (which the above code snippet is doing) and define where to send the reports via the bindings cluster, since that's how this works in ZCL.  That is not done by the above code.

In practical terms, doing `out/debug/chip-tool binding bind 112233 0 1 1026 1` before doing the `listen` command should get you useful reports, I would expect.

As far as using the python controller, if I set up the `chip-all-clusters-app` with node id 17 when I `connect`, the following seem to work:
```
zclconfigure TemperatureMeasurement MeasuredValue 17 1 1 2 10
zcl Binding Bind 17 1 0 nodeId=112233 groupId=0 endpointId=1 clusterId=1026
```
and then the reports come in.

Ache17, 2021-08-10:
Many thanks for fast reply, help and clarifying concepts. 
Reporting temperature using ESP32 worked after applying the bind command so thank you very much for resolving this.

if the below is another issue then I will make one but I thought I would ask since it's closely related : 
When using python controller, is there some command I can use to listen for the responses ( or overall how could I get the data) ? 
I switched now to EFR32 Lighting example and I configured EFR32 example simillarly to the all-clusters-app using ZAP, I got successes in all messages, but I don't receive any data really ( setup and log attached below ).

I'm communicating over Bluetooth for now just for making the testing faster ( my understanding is that it should work the same as when device communicate using border router, although I'm not sure about that ).

### enabling the cluster for temperature measurement
```
1. ./scripts/tools/zap/run_zaptool.sh examples/lighting-app/lighting-common/lighting-app.zap
2. "Endpoint - 1" -> "Measurement & Sensing" -> set to "Server" in options for Temperature Measurement and and enable the  "measured value" from "attribute reporting" -> enable the "Binding" cluster ( and enable "Bind" and "Unbind" command)
3. Save the file
4. ./scripts/tools/zap/generate.py examples/lighting-app/lighting-common/lighting-app.zap
```
### compiling and flashing the efr32 board
```
./scripts/examples/gn_efr32_example.sh ./examples/lighting-app/efr32/ ./out/lighting-app BRD4161A
python ./out/lighting-app/BRD4161A/chip-efr32-lighting-example.flash.py
```
### communication with the device and configuring the reports.
from chip-device-ctrl
```
1. > connect -ble 3840 73141520 135246
2. > zclread TemperatureMeasurement MeasuredValue 135246 1 0 [ that's working, I got success and reading ]
3. > zclconfigure TemperatureMeasurement MeasuredValue 135246 1 1 2 0 [ also working, got success ]
4. > zcl Binding Bind 135246 1 0 nodeId=112233 groupId=0 endpointId=1 clusterId=1026 [ also success but after that I got nothing else ]
```
I'm sending also the log, every message is successful but I don't see any data.
[msg2.txt](https://github.com/project-chip/connectedhomeip/files/6963079/msg2.txt)

closed 2022-01-26 by woody-apple