project-chip/connectedhomeip #8860: [ZAP] Allocate only the necessary bytes for read-only attributes

vivien-apple, 2021-08-09:
#### Problem

Currently, for attributes, codegen generates the maximum possible amount of bytes into `GENERATED_DEFAULTS` and some other macros such as `ATTRIBUTE_MAX_SIZE` and `ATTRIBUTE_SINGLETONS_SIZE`.

Those macros are used for the size of some global arrays: 
 * https://github.com/project-chip/connectedhomeip/blob/a55a0aa5e6b711dc1f39323352aac73a4e881e33/src/app/util/attribute-storage.cpp#L63
 * https://github.com/project-chip/connectedhomeip/blob/a55a0aa5e6b711dc1f39323352aac73a4e881e33/src/app/util/attribute-storage.cpp#L70

 #### Proposed Solution
 For `read-only` attributes, we could generate only the necessary amount of bytes to store the data

It would require to fix `helper-endpoint_config.js` (https://github.com/project-chip/zap/blob/master/src-electron/generator/helper-endpointconfig.js) and notably the part that do thing for attributes larger than 2 bytes: https://github.com/project-chip/zap/blob/master/src-electron/generator/helper-endpointconfig.js#L474

erjiaqing, 2021-08-11:
Just a reminder: the "ReadOnly" flag in ZCL is only a flag, and this check can be bypassed: https://github.com/project-chip/connectedhomeip/blob/4dd523067f9c3c47107efab90f2e52930fa9e248/examples/wifi-echo/server/esp32/main/attribute-table.c#L509-L524

bzbarsky-apple, 2022-06-13:
For the generated defaults, we did this in #11768.