project-chip/connectedhomeip #8872: iOS demo can't connect device via BLE by QRCode scan.  allways CHIP Error Time out

tigline, 2021-08-10:
#### Problem
<what's wrong or missing, please include any applicable:
I run the iOS demo on iPhone, use QR code scan and connect by ble. then always fail

Here is the Log:
2021-08-10 16:31:03.441346+0800 CHIPTool[26554:6541382] Found an existing keypair in the keychain
2021-08-10 16:31:03.441505+0800 CHIPTool[26554:6541382] Deserializing the key
2021-08-10 16:31:03.442673+0800 CHIPTool[26554:6541382] CHIPOperationalCredentialsDelegate::init returning ../../../../../../../../../../../connectedhomeip/src/crypto/CHIPCryptoPALmbedTLS.cpp:707: Success
2021-08-10 16:31:03.442732+0800 CHIPTool[26554:6541382] PersistentStorageDelegate Sync Get Value for Key: com.zigbee.chip.commissioner.device_id
2021-08-10 16:31:03.442832+0800 CHIPTool[26554:6541382] CHIPPersistentStorageDelegate Get Value for Key: com.zigbee.chip.commissioner.device_id, value yQzKK6n2nsU=
2021-08-10 16:31:03.442899+0800 CHIPTool[26554:6541382] Found c59ef6a92bca0cc9 node ID for the controller
2021-08-10 16:31:03.444829+0800 CHIPTool[26554:6541382] [all] 🟢 [1628584263444] [26554:6541382] CHIP: [DL] _Init
2021-08-10 16:31:03.444950+0800 CHIPTool[26554:6541382] [all] 🟢 [1628584263444] [26554:6541382] CHIP: [IN] UDP::Init bind&listen port=5540
2021-08-10 16:31:03.445068+0800 CHIPTool[26554:6541382] [all] 🟢 [1628584263445] [26554:6541382] CHIP: [IN] UDP::Init bind&listen port=5540
2021-08-10 16:31:03.445113+0800 CHIPTool[26554:6541382] [all] 🟢 [1628584263445] [26554:6541382] CHIP: [IN] TransportMgr initialized
2021-08-10 16:31:03.445133+0800 CHIPTool[26554:6541382] [all] 🟢 [1628584263445] [26554:6541382] CHIP: [DIS] Init fabric pairing table with server storage
2021-08-10 16:31:03.445169+0800 CHIPTool[26554:6541382] PersistentStorageDelegate Sync Get Value for Key: Fabric0
2021-08-10 16:31:03.445206+0800 CHIPTool[26554:6541382] CHIPPersistentStorageDelegate Get Value for Key: Fabric0, value AAAAAAAAAADJDMorqfaexQAAAAAAAAAAAADvAAAA/AAEcRITIIL8VMjwGHYnZkkSSoPSgwReQuGmXF012pLvc5e7Pn2Y5mFnAksfRapMiAiCJoURHTVKWgyERFYcuq0vTjZyXpPWaCoPziLzFSoOfKEEtIvH+ygUeLd8m53Wf6NzAAAAAAAAAGEAAAAAAAAAFTABAQAkAgE3AyUU0gQkFQAYJgSu/6QoJgUuM4YqNwYlFNIEJBUAGCQHASQIATAJQQQz1oAq56XiAWtPYDR9BZSRbpZ5HT5IkgzsK6aWiwTQ6bt96AAUwVPkITV0GA3OMCCrfUk3YaXZO5t/yrKbeQebNwo1ASkBGCQCYDAEFDysx9jLyi+yY/Oy2YWzsZHuvSaGMAUUPKzH2MvKL7Jj87LZhbOxke69JoYYMAtAgZfQUnBen5hLEAohHbawx25aZ8d31VbKFZBDo8KyF+cW/6pZZMbf96Pe4zzi2QyDvABDB1pbqF32UpFy7Qg35BgAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAVMAEBASQCATcDJRTSBCQVABgmBK7/pCgmBS4zhio3BicRyQzKK6n2nsUkFQAYJAcBJAgBMAlBBHESEyCC/FTI8Bh2J2ZJEkqD0oMEXkLhplxdNdqS73OXuz59mOZhZwJLH0WqTIgIgiaFER01SloMhERWHLqtL043CjUBKAEYJAIBNgMEAgQBGDAEFOSNKG542p83oUVtqdJAG9/+6tlpMAUUPKzH2MvKL7Jj87LZhbOxke69JoYYMAtAklBzVF0No9MvmeOPhJKZ1b9tmivxc/MbMzKf+n3M0j2+llBxrZz1+s18KccPujfmng5XdFGtlB65gOP87JG0JRgAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
2021-08-10 16:31:03.445828+0800 CHIPTool[26554:6541382] [all] 🔵 [1628584263445] [26554:6541382] CHIP: [IN] Loading certs from KVS
2021-08-10 16:31:03.445987+0800 CHIPTool[26554:6541382] [all] 🔵 [1628584263445] [26554:6541382] CHIP: [ZCL] Using ZAP configuration...
2021-08-10 16:31:03.446104+0800 CHIPTool[26554:6541382] [all] 🔵 [1628584263446] [26554:6541382] CHIP: [ZCL] deactivate report event
2021-08-10 16:31:03.446128+0800 CHIPTool[26554:6541382] [all] 🔵 [1628584263446] [26554:6541382] CHIP: [CTL] Getting operational keys
2021-08-10 16:31:03.446144+0800 CHIPTool[26554:6541382] [all] 🔵 [1628584263446] [26554:6541382] CHIP: [CTL] Generating credentials
2021-08-10 16:31:03.446338+0800 CHIPTool[26554:6541382] [all] 🔵 [1628584263446] [26554:6541382] CHIP: [CTL] Loaded credentials successfully
2021-08-10 16:31:03.446385+0800 CHIPTool[26554:6541382] PersistentStorageDelegate Sync Get Value for Key: StartKeyID
2021-08-10 16:31:03.446423+0800 CHIPTool[26554:6541382] CHIPPersistentStorageDelegate Get Value for Key: StartKeyID, value LgA=
2021-08-10 16:31:03.446451+0800 CHIPTool[26554:6541382] PersistentStorageDelegate Sync Get Value for Key: ListPairedDevices0
2021-08-10 16:31:03.446471+0800 CHIPTool[26554:6541382] CHIPPersistentStorageDelegate Get Value for Key: ListPairedDevices0, value (null)
2021-08-10 16:31:03.446496+0800 CHIPTool[26554:6541382] PersistentStorageDelegate Sync Get Value for Key: ListPairedDevices0
2021-08-10 16:31:03.446556+0800 CHIPTool[26554:6541382] CHIPPersistentStorageDelegate Get Value for Key: ListPairedDevices0, value (null)
2021-08-10 16:31:03.446722+0800 CHIPTool[26554:6541382] PersistentStorageDelegate Sync Get Value for Key: ListPairedDevices0
2021-08-10 16:31:03.446863+0800 CHIPTool[26554:6541382] CHIPPersistentStorageDelegate Get Value for Key: ListPairedDevices0, value (null)
2021-08-10 16:31:03.447016+0800 CHIPTool[26554:6541382] PersistentStorageDelegate Sync Get Value for Key: ListPairedDevices0
2021-08-10 16:31:03.447217+0800 CHIPTool[26554:6541382] CHIPPersistentStorageDelegate Get Value for Key: ListPairedDevices0, value (null)
2021-08-10 16:31:03.447352+0800 CHIPTool[26554:6541382] PersistentStorageDelegate Sync Get Value for Key: ListPairedDevices0
2021-08-10 16:31:03.447491+0800 CHIPTool[26554:6541382] CHIPPersistentStorageDelegate Get Value for Key: ListPairedDevices0, value (null)
2021-08-10 16:31:03.447661+0800 CHIPTool[26554:6541382] PersistentStorageDelegate Sync Get Value for Key: ListPairedDevices0
2021-08-10 16:31:03.447863+0800 CHIPTool[26554:6541382] CHIPPersistentStorageDelegate Get Value for Key: ListPairedDevices0, value (null)
2021-08-10 16:31:03.447984+0800 CHIPTool[26554:6541382] PersistentStorageDelegate Sync Get Value for Key: ListPairedDevices0
2021-08-10 16:31:03.448107+0800 CHIPTool[26554:6541382] CHIPPersistentStorageDelegate Get Value for Key: ListPairedDevices0, value (null)
2021-08-10 16:31:03.448220+0800 CHIPTool[26554:6541382] PersistentStorageDelegate Sync Get Value for Key: ListPairedDevices0
2021-08-10 16:31:03.448383+0800 CHIPTool[26554:6541382] CHIPPersistentStorageDelegate Get Value for Key: ListPairedDevices0, value (null)
2021-08-10 16:31:03.448579+0800 CHIPTool[26554:6541382] PersistentStorageDelegate Sync Get Value for Key: ListPairedDevices0
2021-08-10 16:31:03.448733+0800 CHIPTool[26554:6541382] CHIPPersistentStorageDelegate Get Value for Key: ListPairedDevices0, value (null)
2021-08-10 16:31:03.448819+0800 CHIPTool[26554:6541382] PersistentStorageDelegate Sync Get Value for Key: ListPairedDevices0
2021-08-10 16:31:03.449022+0800 CHIPTool[26554:6541382] CHIPPersistentStorageDelegate Get Value for Key: ListPairedDevices0, value (null)
2021-08-10 16:31:03.449162+0800 CHIPTool[26554:6541382] PersistentStorageDelegate Sync Get Value for Key: ListPairedDevices0
2021-08-10 16:31:03.449292+0800 CHIPTool[26554:6541382] CHIPPersistentStorageDelegate Get Value for Key: ListPairedDevices0, value (null)
2021-08-10 16:31:03.449417+0800 CHIPTool[26554:6541382] PersistentStorageDelegate Sync Get Value for Key: ListPairedDevices0
2021-08-10 16:31:03.449618+0800 CHIPTool[26554:6541382] CHIPPersistentStorageDelegate Get Value for Key: ListPairedDevices0, value (null)
2021-08-10 16:31:03.449783+0800 CHIPTool[26554:6541382] PersistentStorageDelegate Sync Get Value for Key: ListPairedDevices0
2021-08-10 16:31:03.449905+0800 CHIPTool[26554:6541382] CHIPPersistentStorageDelegate Get Value for Key: ListPairedDevices0, value (null)
2021-08-10 16:31:03.450104+0800 CHIPTool[26554:6541382] PersistentStorageDelegate Sync Get Value for Key: ListPairedDevices0
2021-08-10 16:31:03.450258+0800 CHIPTool[26554:6541382] CHIPPersistentStorageDelegate Get Value for Key: ListPairedDevices0, value (null)
[... cut ...]
2021-08-10 16:32:05.918318+0800 CHIPTool[26554:6541610] DevicePairingDelegate Pairing complete. Status ../../../../../../../../../../../connectedhomeip/src/protocols/secure_channel/PASESession.cpp:333: CHIP Error 0x00000032: Timeout
2021-08-10 16:32:05.919090+0800 CHIPTool[26554:6541610] Got pairing error back Error Domain=CHIPErrorDomain Code=1 "Undefined error:50." UserInfo={NSLocalizedDescription=Undefined error:50.}

 >

 #### Proposed Solution
<suggested fix, suggested enhancement>
So where should I find the issue?

cjandhyala, 2021-08-10:
Same thing happens. Python controller not able to connect to the device over BLE with ToT (73f5cf16059e5a9e4586083e42a7b42ac56e62e0) 

Steps to reproduce. 
1. pull ToT both on controller and accessory. 
2. Build accessory using 'scripts/examples/gn_build_example.sh examples/lighting-app/linux/ out/lighting-app' and on controller raspi build controller using './scripts/build_python.sh --clusters_for_ip_commissioning true --chip_detail_logging true --chip_mdns platform'
3. launch accesoosy using 'sudo ./out/lighting-app/chip-lighting-app --wifi' , launch controller using 'source ./out/python_env/bin/activate' and 'chip-device-ctrl '

Python controller log
---------------------
chip-device-ctrl > connect -ble 3840 20202021 12344321
Device is assigned with nodeid = 12344321
[1628617625.929040][3940377:3940385] CHIP:SC: Assigned local session key ID 1
[1628617625.929279][3940377:3940385] CHIP:IN: Message appended to BLE send queue
[1628617625.929346][3940377:3940385] CHIP:SC: Sent PBKDF param request
[1628617625.929453][3940377:3940385] CHIP:CTL: SyncSetKeyValue on StartKeyID
[1628617625.930380][3940377:3940502] CHIP:DL: TRACE: Bluez mainloop starting Thread
[1628617625.930699][3940377:3940381] CHIP:DL: TRACE: Bus acquired for name C-2019
[1628617625.942634][3940377:3940385] CHIP:DL: PlatformBlueZInit init success
[1628617625.950090][3940377:3940381] CHIP:BLE: BLE removing known devices.
[1628617625.950183][3940377:3940381] CHIP:BLE: BLE initiating scan.
[1628617626.089317][3940377:3940381] CHIP:BLE: Device 6A:3C:2E:1B:4E:E4 does not look like a CHIP device.
[1628617626.106071][3940377:3940381] CHIP:BLE: Device 70:BF:92:00:85:66 does not look like a CHIP device.
[1628617626.110951][3940377:3940381] CHIP:BLE: Device 0F:FA:98:D8:F7:D3 does not look like a CHIP device.
[1628617626.175183][3940377:3940381] CHIP:BLE: Device 7B:93:FC:6E:F7:5A does not look like a CHIP device.
[1628617626.200187][3940377:3940381] CHIP:BLE: Device 52:06:BB:B9:F8:DF does not look like a CHIP device.
[1628617626.214733][3940377:3940381] CHIP:BLE: Device 61:13:75:E0:54:35 does not look like a CHIP device.
[1628617626.216312][3940377:3940381] CHIP:BLE: Device 47:87:B7:48:CC:63 does not look like a CHIP device.
[1628617626.260760][3940377:3940381] CHIP:BLE: Device 75:FB:3D:7E:D9:FC does not look like a CHIP device.
[1628617626.300483][3940377:3940381] CHIP:BLE: Device 40:F4:D8:22:74:C2 does not look like a CHIP device.
[1628617626.312678][3940377:3940381] CHIP:BLE: Device 37:12:C7:56:26:8B does not look like a CHIP device.
[1628617626.415916][3940377:3940381] CHIP:BLE: Device 45:A7:68:29:6A:BA does not look like a CHIP device.
[1628617626.418414][3940377:3940381] CHIP:BLE: Device 47:0D:75:0C:65:D6 does not look like a CHIP device.
[1628617626.419935][3940377:3940381] CHIP:BLE: Device 58:A4:E3:EE:F5:C8 does not look like a CHIP device.
[1628617626.583997][3940377:3940381] CHIP:BLE: Device FB:67:AC:AE:05:8A does not look like a CHIP device.
[1628617626.586333][3940377:3940381] CHIP:BLE: Device B2:22:7A:63:AE:F5 does not look like a CHIP device.
[1628617627.244491][3940377:3940381] CHIP:BLE: Device 6F:08:02:EB:DD:66 does not look like a CHIP device.
[1628617627.314837][3940377:3940381] CHIP:BLE: Device 2B:EF:62:F3:20:19 does not look like a CHIP device.
[1628617627.414609][3940377:3940381] CHIP:BLE: Device D6:C0:95:E5:C2:29 does not look like a CHIP device.
[1628617629.258300][3940377:3940381] CHIP:BLE: Device D1:34:C5:E6:77:D8 does not look like a CHIP device.
[1628617635.204608][3940377:3940381] CHIP:BLE: Device 28:CD:C4:49:E8:D4 does not look like a CHIP device.
[1628617635.956057][3940377:3940381] CHIP:IN: Clearing BLE pending packets.
[1628617635.956170][3940377:3940381] CHIP:IN: BleConnection Error: ../../src/platform/Linux/BLEManagerImpl.cpp:821: CHIP Error 0x00000032: Timeout
[1628617655.930539][3940377:3940385] CHIP:CTL: SyncSetKeyValue on StartKeyID
Failed to establish secure session to device: 50
[1628617655.930991][3940377:3940385] CHIP:SC: PASESession timed out while waiting for a response from the peer. Expected message type was 33
CHIP Error 0x00000032: Timeout
[1628617655.931085][3940377:3940385] CHIP:CTL: SyncSetKeyValue on StartKeyID
Failed to establish secure session to device: 50

Accessory Log
--------------
$ sudo ./out/lighting-app/chip-lighting-app --wifi
[1628617619.094800][12995:12995] CHIP:DL: writing settings to file (/tmp/chip_factory.ini.tmp)
[1628617619.094950][12995:12995] CHIP:DL: renamed tmp file to file (/tmp/chip_factory.ini)
[1628617619.095150][12995:12995] CHIP:DL: writing settings to file (/tmp/chip_config.ini.tmp)
[1628617619.095226][12995:12995] CHIP:DL: renamed tmp file to file (/tmp/chip_config.ini)
[1628617619.095425][12995:12995] CHIP:DL: writing settings to file (/tmp/chip_counters.ini.tmp)
[1628617619.095502][12995:12995] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1628617619.095700][12995:12995] CHIP:DL: Device Configuration:
[1628617619.095759][12995:12995] CHIP:DL:   Serial Number: TEST_SN
[1628617619.095782][12995:12995] CHIP:DL:   Vendor Id: 9050 (0x235A)
[1628617619.095802][12995:12995] CHIP:DL:   Product Id: 65279 (0xFEFF)
[1628617619.095824][12995:12995] CHIP:DL:   Product Revision: 0
[1628617619.095846][12995:12995] CHIP:DL:   Setup Pin Code: 20202021
[1628617619.095869][12995:12995] CHIP:DL:   Setup Discriminator: 3840 (0xF00)
[1628617619.095918][12995:12995] CHIP:DL:   Manufacturing Date: (not set)
[... cut ...]
[1628617619.127797][12995:12995] CHIP:DL: CHIPoBLE advertising config complete
[1628617619.137362][12995:12998] CHIP:DL: RegisterAdvertisement complete
[1628617619.137512][12995:12995] CHIP:DL: HandlePlatformSpecificBLEEvent 16394
[1628617633.611180][12995:12995] CHIP:DIS: Replying to DNS-SD service listing request
[1628617633.611282][12995:12995] CHIP:DIS: Replying to DNS-SD service listing request
[1628617633.611343][12995:12995] CHIP:DIS: Replying to DNS-SD service listing request
[1628617633.611397][12995:12995] CHIP:DIS: Replying to DNS-SD service listing request

andy31415, 2021-08-10:
Per chat with Chaitanya, it seems #7994 is the commit that first makes this happen. Assigning this to @erjiaqing  and @yunhanw-google

andy31415, 2021-08-10:
Yunhan noticed that #7994 was merged *AFTER* this issue was reported ... so root cause is likely diffrent. 
@tingline could you bisect to find the root PR that causes this?

andy31415, 2021-08-10:
Guessing python controller is new, iOS thing existed before?

If that is the case, it may be worth it  to split the issues. Would also try to split the logs into files as it seems messages get lost when long logs are inside comments.

andy31415, 2021-08-10:
I edited the comments to place the logs in a collapsible section.

yunhanw-google, 2021-08-10:
use python chip device controller in macbook
one rapsi before song's commit for im write enablement, the ble connection is good to go
another raspi after song's commit for im write enablement,  the ble connection is good to go,
@cjandhyala it may be caused by your setup?

the issue in original ticket happens in ios demo app, which could be another different story

yunhanw-google, 2021-08-10:
@pan-apple maybe you can take a look this ios demo issue for this timeout error?

andy31415, 2021-08-11:
I would suggest to separate out the python controller issue from the iOS one as they seem different.

Since they are both regressions,  bisecting and finding what causes them would help a lot.

tigline, 2021-08-12:
when I use Chip tool run in terminal, also occur this error. CHIP Error 0x00000032: Timeout

tigline, 2021-08-13:
@andy31415  additional information: when I checkout the code at the timeline Jun 15. the App can work normally,
[image]

and the time maybe after July 1. all can not work.

tewarid, 2021-08-31:
I am able to pair using qrcode with a Linux device running all-clusters-app - commit hash at f07ff95183d33af17bc833c92c64e7409c6fd2eb.

tewarid, 2021-09-01:
On the other hand, pairing does not complete successfully - on the device side the following appear in the log, ad infinitum

```text
[1630500676.438793][303:306] CHIP:BLE: Write request received debug 0xb63b0240
[1630500678.990717][303:306] CHIP:BLE: Indication confirmation, 0xb63b0240
```

closed 2022-01-26 by woody-apple