project-chip/connectedhomeip #8894: Client-side command response dispatch is unsafe/broken

mrjerryjohns, 2021-08-11:
#### Problem
The current client-side command request + response pathways have a number of sizable issues:
* The success and error callbacks are type-erased to generic Cancelables when stored in gCallbackMgr. Later when a response comes in, they are forcibly re-interpret casted back to a typed response callback signature with a specific set of command arguments. This expects the 'other side' to be well behaved and send back the right kind of response. However, if the response coming back is different (bug on the sender), it will result in really egregious casting of those response fields into a mismatched signature, causing all kind of havoc.
* When a successful data response is received, it goes through a completely different path than when status codes are sent back. In the former case, it goes through `DispatchSingleClusterResponseCommand` --> `emberAf<CommandReponse>()`, gCallbackMgr and finally, OnSuccess callback. The latter goes through a delegate registered to the CommandSender object, which then goes to CHIPDeviceController, before finally getting back through gCallbackMgr. This is highly asymmetrical and makes debugging very difficult.
* When a status code indicating success is received, it eventually dispatches to the OnSuccess callback reinterpret casted to a DefaultSuccessCallback. This means that if you were expecting a data response but got a success status response instead, you will have complete gibberish passed into your onSuccess callback as the args. 
* When sending out a command request, to correctly dispatch the ensuing response back to the right set of success/error callbacks, the `CommandSender` object is forcibly typecast to a 64-bit 'NodeId'. This is not intrinsically broken (unless we go to a 128-bit arch), but it is definitely is not a right thing to do.

mrjerryjohns, 2021-08-11:
Proposed solution: 

Creation of a new `SingleInvokeInitiator` class that replaces `CommandSender` and does the following:
* Application managed lifetime, i.e not allocated through InteractionModelEngine. This gives the application flexibility to control the allocation strategy (heap, stack).
* Only permits single command request in invoke + single command response in invoke.
* Template class that is specialized by the type of the response object that adheres to #8891.
* Automatic typedef of a success callback function that has an argument a reference to the response of type T.
* Automatic instantiation of the response object on stack, decoding of TLV into that object.
* Automatic checking of the received response against the Command ID + Cluster ID of the specialized response object, ensuring safety.
* Compresses the function call-chain above from 4-5 frames down to 1, by immediately invoking the response callback from within the initiator object's exchange delegate callback.
* Allows for us to completely avoid generating IMClusterCommandHandler.cpp and CHIPClientCallbacks.cpp, a net savings of over 10K lines of code.

The above strategy has been roughly prototyped already in the IM Prototype PR (#8313) in the `DemuxedInvokeInitiator` class.

msandstedt, 2021-09-15:
> The success and error callbacks are type-erased to generic Cancelables when stored in gCallbackMgr. Later when a response comes in, they are forcibly re-interpret casted back to a typed response [...]

Related to unresolved #1350.   Fixing this here would also fix that.

mrjerryjohns, 2021-09-16:
While my original post here mentioned creating a new `InvokeInitiator` class, I think we should just focus instead of fixing the existing `CommandSender` class based on what's mentioned above instead.

closed 2021-10-12 by mrjerryjohns