project-chip/connectedhomeip #8906: app::CommandHandler should provide an accessor for the corresponding ExchangeContext

pan-apple, 2021-08-11:
Add an accessor to find the exchange context on which a particular command was received.

_Originally posted by @msandstedt in https://github.com/project-chip/connectedhomeip/pull/8826#discussion_r686956918_

bzbarsky-apple, 2021-08-11:
What information do consumers actually need here?  I'd rather not expose exchange contexts directly; they are too easy to misuse.

@mrjerryjohns

mrjerryjohns, 2021-08-11:
Is this to retrieve the fabric associated with the interaction, to then correctly enforce privileges for the executed command? (e.g UpdateNOC can only update op-certs on the fabric associated with the underlying session through which that command came through)

mrjerryjohns, 2021-08-11:
Instead of providing access to the EC directly, I'd highly recommend instead an insulated method like "GetFabricInfo" that internally, gets that through the EC...

msandstedt, 2021-08-11:
> Is this to retrieve the fabric associated with the interaction, to then correctly enforce privileges for the executed command? (e.g UpdateNOC can only update op-certs on the fabric associated with the underlying session through which that command came through)

Yes it is.

> Instead of providing access to the EC directly, I'd highly recommend instead an insulated method like "GetFabricInfo" that internally, gets that through the EC.
> 
> Just giving access to the underlying EC should be avoided _if possible_, since mis-use by applications can wreck havoc on the true owner of that EC, which is the command object.

That sounds fine.  Per the PR description, I had pointed this out in https://github.com/project-chip/connectedhomeip/pull/8826#discussion_r686956918 where we see code using `emberAfCurrentCommand()->source` to get this information.  Whatever context is needed ought to be accessible from `app::CommandHandler`.

bzbarsky-apple, 2022-08-26:
This accessor exists now: `CommandHandler:GetExchangeContext`.

closed 2022-08-26 by bzbarsky-apple