project-chip/connectedhomeip #8909: Need clarity on requirements and design of AdvertiseOperational() API

pan-apple, 2021-08-11:
#### Problem
The API is being called when the device get provisioned with new operational certificates (via AddNOC or UpdateNOC) commands. However, it's not clear how long the API is supposed to take during the synchronous part of the operation. Given, the API could have multiple implementations, for different platforms, it's important to document the timing requirements, and expected design. Based on that the call to `AdvertiseOperational` could be moved to the more appropriate part of the code. 

Reference: https://github.com/project-chip/connectedhomeip/pull/8826#discussion_r686973301

bzbarsky-apple, 2021-08-11:
@andreilitvin @cecille

andy31415, 2021-08-12:
Add as a comment. I believe that 'Advertise' should be 'initiate advertising' and will do any calculation inline that it can (all name and property calculation) however if something is significantly blocking, it would be deferred async to not make  'advertise' block.

I am thinking of these potential blocking operations:
  - mDNS RFC requires detecting of conflicting hostnames (you query first your desired name, if nobody answers, you take it. Otherwise you pick a new name and try again). minmdns does *not* do this, stating that 'chip picks unique enough names' (and it is not mDNS RFC compliant here, but we need to pick if this is important enough to fix vs other things). Avahi and darwin will do this (very likely)
  - Setting up advertisement may rely on storage, which is still 'sync' for us, however reading from flash is also somewhat fast. So we are not real time, but nor are we blocking for  a long/undeterminated amount of time either.
  - minmdns does a 'enqueue broadcast advertise when we start advertising' however it will not wait for it to be actually sent out over the network.
  
  What  are the requirements for Add/Update NOC? Do we need  to block until 'advertisement is officially done/available' or it this a concern of add/update callback to not block for long times?

bzbarsky-apple, 2021-08-12:
>  Do we need to block until 'advertisement is officially done/available' 

Not at all.  Ideally we would do the advertisement completely in the background or something... and definitely in parallel with sending the "we succeeded" response to the caller.