project-chip/connectedhomeip #8924: CASE allocates memory based on wire-provided sizes without sound bounds

tcarmelveilleux, 2021-08-12:
#### Problem
In CASESession.cpp, there exists several usage of the following pattern:

`VerifyOrExit(msg_R2_Encrypted.Alloc(tlvReader.GetLength()), err = CHIP_ERROR_NO_MEMORY);`

This assumes in many places that `tlvReader.GetLength()` is not a huge value (or contextually large value), even though only certain values are permissible.

 #### Proposed Solution

All usage of `tlvReader.GetLength()` for allocations should be audited and each one should be validated against a valid upper bound for the size of the value being read, within practical limits.

holbrookt, 2022-02-01:
@tcarmelveilleux I see that a few PRs were merged referencing this. Is the issue fixed?

woody-apple, 2022-02-07:
Marking as closed, please re-open if still an issue.

bzbarsky-apple, 2022-03-31:
This is still an issue, as far as I can see.  For example, CASESession::HandleSigma2 still does this:
```
    SuccessOrExit(err = tlvReader.Next());
    VerifyOrExit(TLV::TagNumFromTag(tlvReader.GetTag()) == ++decodeTagIdSeq, err = CHIP_ERROR_INVALID_TLV_TAG);
    VerifyOrExit(msg_R2_Encrypted.Alloc(tlvReader.GetLength()), err = CHIP_ERROR_NO_MEMORY);
```

PR #17296 (2022-04-13): Stop allocating CASE buffer based on wire sizes
#### Problem
- CASE was using direct lengths of TLV values to allocate heap values,
  without validations.
Fixes #8924

#### Change overview
- Added validations on lengths of Sigma2/3 TBE, with some room to grow
- Updated tag order validation logic to be more explicit and easier to
  read.

#### Testing
- Cert tests still pass
- Unit tests still pass
commit 213869f6 (2022-04-14): Stop allocating CASE buffer based on wire sizes (#17296)

* Stop allocating CASE buffer based on wire sizes

- CASE was using direct lengths of TLV values to allocate heap values,
  without validations.
- Added validations on lengths of Sigma2/3 TBE, with some room to grow
- Updated tag order validation logic to be more explicit and easier to
  read.

Fixes #8924

Test done:
- Cert tests still pass
- Unit tests still pass

* Restyled by clang-format
closed 2022-04-13 by tcarmelveilleux