project-chip/connectedhomeip #8932: Error occur calling SyncDeleleKeyValue

sweetymhaiske, 2021-08-12:
#### Problem
For all-clusters app, after flashing, on the device side these error logs are seen. However, commissioning and Cluster control works fine.

**I (1526) chip[SVR]: Delete value in server storage
E (1526) chip[DIS]: Error occurred calling SyncDeleteKeyValue
I (1536) chip[SVR]: Delete value in server storage
E (1536) chip[DIS]: Error occurred calling SyncDeleteKeyValue**

```
I (1336) chip[DL]: WIFI_EVENT_STA_START
W (1336) wifi:Haven't to connect to a suitable AP now!
I (1346) chip[DL]: Done driving station state, nothing else to do...
I (1346) app-devicecallbacks: Current free heap: 117260

I (1256) chip[ZCL]: DEVICE WILL NOT FUNCTION PROPERLY AFTER REBOOTING!!
I (1366) app-devicecallbacks: PostAttributeChangeCallback - Cluster ID: '0x0500', EndPoint ID: '0x01', Attribute ID: '0x0001'
I (1386) app-devicecallbacks: Unhandled cluster ID: 1280
I (1386) app-devicecallbacks: Current free heap: 117260

I (1396) app-devicecallbacks: PostAttributeChangeCallback - Cluster ID: '0x0500', EndPoint ID: '0x01', Attribute ID: '0x0002'
I (1406) app-devicecallbacks: Unhandled cluster ID: 1280
I (1406) app-devicecallbacks: Current free heap: 117260

I (1416) chip[ZCL]: Failed to send IAS Zone update. Err 0x70
I (1416) app-devicecallbacks: PostAttributeChangeCallback - Cluster ID: '0x0406', EndPoint ID: '0x02', Attribute ID: '0x0001'
I (1426) app-devicecallbacks: Unhandled cluster ID: 1030
I (1436) app-devicecallbacks: Current free heap: 117260

I (1436) app-devicecallbacks: PostAttributeChangeCallback - Cluster ID: '0x0406', EndPoint ID: '0x02', Attribute ID: '0x0002'
I (1456) app-devicecallbacks: Unhandled cluster ID: 1030
I (1466) app-devicecallbacks: Current free heap: 117260

V (1476) chip[DIS]: Init fabric pairing table with server storage
V (1476) chip[IN]: UDP::Init bind&listen port=5540
V (1476) chip[IN]: UDP::Init bind&listen port=5540
V (1486) chip[IN]: TransportMgr initialized
I (1486) chip[DL]: NVS set: chip-counters/GlobalMCTR = 1000 (0x3E8)
I (1496) chip[SVR]: Delete value in server storage
E (1496) chip[DIS]: Error occurred calling SyncDeleteKeyValue
I (1516) chip[SVR]: Delete value in server storage
E (1516) chip[DIS]: Error occurred calling SyncDeleteKeyValue
I (1526) chip[SVR]: Delete value in server storage
E (1526) chip[DIS]: Error occurred calling SyncDeleteKeyValue
I (1536) chip[SVR]: Delete value in server storage
E (1536) chip[DIS]: Error occurred calling SyncDeleteKeyValue
I (1546) chip[SVR]: Delete value in server storage
E (1546) chip[DIS]: Error occurred calling SyncDeleteKeyValue
I (1556) chip[SVR]: Delete value in server storage
E (1556) chip[DIS]: Error occurred calling SyncDeleteKeyValue
I (1566) chip[SVR]: Delete value in server storage
E (1576) chip[DIS]: Error occurred calling SyncDeleteKeyValue
I (1586) chip[SVR]: Delete value in server storage
E (1586) chip[DIS]: Error occurred calling SyncDeleteKeyValue
I (1596) chip[SVR]: Delete value in server storage
E (1596) chip[DIS]: Error occurred calling SyncDeleteKeyValue
I (1606) chip[SVR]: Delete value in server storage
E (1606) chip[DIS]: Error occurred calling SyncDeleteKeyValue
I (1616) chip[SVR]: Delete value in server storage
E (1616) chip[DIS]: Error occurred calling SyncDeleteKeyValue
I (1626) chip[SVR]: Delete value in server storage
E (1626) chip[DIS]: Error occurred calling SyncDeleteKeyValue
I (1646) chip[SVR]: Delete value in server storage
E (1646) chip[DIS]: Error occurred calling SyncDeleteKeyValue
I (1656) chip[SVR]: Delete value in server storage
E (1656) chip[DIS]: Error occurred calling SyncDeleteKeyValue
I (1666) chip[SVR]: Delete value in server storage
E (1666) chip[DIS]: Error occurred calling SyncDeleteKeyValue
I (1676) chip[SVR]: Delete value in server storage
E (1676) chip[DIS]: Error occurred calling SyncDeleteKeyValue
V (1686) chip[SC]: Assigned local session key ID 0
V (1686) chip[SC]: Waiting for PBKDF param request
I (1686) chip[IN]: CASE Server enabling CASE session setups
I (1696) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable, device name MATTER-3840)
I (1706) chip[SVR]: Server Listening...
I (1716) app-devicecallbacks: PostAttributeChangeCallback - Cluster ID: '0x0402', EndPoint ID: '0x01', Attribute ID: '0x0000'
I (1716) chip[DL]: Device already advertising, stop active advertisement and restart
I (1726) app-devicecallbacks: Unhandled cluster ID: 1026
GAP procedure initiated: stop advertising.
I (1746) app-devicecallbacks: Current free heap: 115892

I (1746) app-devicecallbacks: PostAttributeChangeCallback - Cluster ID: '0x0101', EndPoint ID: '0x01', Attribute ID: '0x0000'
GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=2 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (1756) app-devicecallbacks: Unhandled cluster ID: 257
I (1786) app-devicecallbacks: Current free heap: 115772

I (1786) chip[ZCL]: Failed to read AutoRelockTime attribute: 0x86
I (1796) app-devicecallbacks: PostAttributeChangeCallback - Cluster ID: '0x0008', EndPoint ID: '0x01', Attribute ID: '0x0000'
I (1806) app-devicecallbacks: Current free heap: 114992

I (1806) chip[ZCL]: 0x3c0f9f2cep 2 clus  8 attr  0 not supported
I (1816) all-clusters-app: Setup discriminator: 3840 (0xf00)
I (1826) all-clusters-app: Setup PIN code: 20202021 (0x1344225)
I (1836) all-clusters-app: Short Manual(decimal) setup code: 34970112332
I (1846) all-clusters-app: Long Manual(decimal) setup code:  749701123309050177298
I (1846) all-clusters-app: QR CODE Text: 'MT:C8XA0SRB00KA0648G00'
I (1856) all-clusters-app: Copy/paste the below URL in a browser to see the QR CODE:
	https://dhrishi.github.io/connectedhomeip/qrcode.html?data=MT%3AC8XA0SRB00KA0648G00
I (31686) chip[DL]: Configuring CHIPoBLE advertising (interval 500 ms, connectable, device name MATTER-3840)
```

andy31415, 2021-08-13:
Error seems reported from FabricInfo::DeleteFromKVS(

dhrishi, 2021-09-09:
@sweetymhaiske Is this fixed now? If yes, please close the issue

sweetymhaiske, 2021-09-09:
Fixed.

closed 2021-09-09 by sweetymhaiske