project-chip/connectedhomeip #8934: iOS project Issue

Ben21hao, 2021-08-12:
#### Problem
Run the IOS project under Src / Darwin directory and an error is reported.The following is the error information：

+ gn --root=/Users/xmiles-bo/Downloads/homeIp/connectedhomeip gen --check out '--args=default_configs_cosmetic=[] chip_crypto="mbedtls" chip_build_tools=false chip_build_tests=false chip_ble_project_config_include="" chip_device_project_config_include="" chip_inet_project_config_include="" chip_system_project_config_include="" target_cpu="arm64" target_defines=["CHIP_PROJECT_CONFIG_INCLUDE=<CHIPProjectConfig.h>","CHIP_PLATFORM_CONFIG_INCLUDE=<platform/Darwin/CHIPPlatformConfig.h>","CHIP_DEVICE_PLATFORM_CONFIG_INCLUDE=<platform/Darwin/CHIPDevicePlatformConfig.h>","DEBUG=1"] target_cflags=["-target","arm64-apple-ios13.4","-arch","arm64","-flto"] target_os="ios" import("//config/ios/args.gni")'
./chip_xcode_build_connector.sh: line 158: gn: command not found
Command PhaseScriptExecution failed with a nonzero exit code

 #### Proposed Solution
Expect to be able to run the IOS project

Ben21hao, 2021-08-13:
How can I run Darwin.xcworkspace project in the Src / Darwin folder?

Ben21hao, 2021-08-13:
@mspang

Ben21hao, 2021-08-16:
@pan-apple @mkardous-silabs @sagar-apple @anush-apple Hello, everyone. I just started the fork project, How can I run Darwin.xcworkspace project in the Src / Darwin folder?

tigline, 2021-08-17:
@Ben21hao  you should install ninja and gn build tool in you mac for CHIP.

Ben21hao, 2021-08-19:
> @Ben21hao you should install ninja and gn build tool in you mac for CHIP.

Thank you. I can start the project.

woody-apple, 2022-01-26:
We're in progress of reviewing older issues. Is this still an issue given how old the SDK this was tested against? Please re-open if so. Thanks!

closed 2022-01-26 by woody-apple

Lee6804, 2022-04-07:
@woody-apple Hello, I installed ninja and gn, why does it still report this error when running?