project-chip/connectedhomeip #8955: Sort out what Advertise should actually be doing

bzbarsky-apple, 2021-08-12:
I spent some time staring at `Advertise(bool commissionableNode)` and I don't understand what this code is even trying to do....  I _think_ the `additionalPairing` boolean can basically be set to `HaveOperationalCredentials()`, but what does the `modeEnabled` arg to `CommissionAdvertisingParameters::SetCommissioningMode` even mean?  Should it just be set to `commissionableNode`?  Something else?  None of this stuff is documented and the naming doesn't match up to spec concepts very well.  :(

@cecille @andy31415 @chrisdecenzo 

_Originally posted by @bzbarsky-apple in https://github.com/project-chip/connectedhomeip/pull/8923#discussion_r687917074_

cecille, 2021-08-12:
What is this issue trying to address? Are you looking for more documentation? A re-write? Or are you saying there is a bug?

bzbarsky-apple, 2021-08-12:
The context is that the existing code is clearly not correct, because it uses `IsFullyProvisioned`, which does not correspond to useful provisioning state in practice.

What neither I nor @msandstedt could figure out is exactly what this code is trying to do, which makes it hard to make it do it.

So:
1.  The code is buggy.
2. There is not enough documentation about what this code is meant to do and what the various APIs involved are meant to do, so we don't know exactly how to fix the bugginess.

bzbarsky-apple, 2022-08-26:
We no longer use `IsFullyProvisioned` here, and the API surface has been changed some and better documented.

closed 2022-08-26 by bzbarsky-apple