project-chip/connectedhomeip #8994: Accessor functions do not set default values for attributes

mkardous-silabs, 2021-08-13:
#### Problem
* Accessor functions do not set default values for attributes in case the ember read function fails.
`EmberAfStatus GetColorLoopTime(chip::EndpointId endpoint, uint16_t * colorLoopTime)
{
    return emberAfReadServerAttribute(endpoint, ColorControl::Id, Ids::ColorLoopTime, (uint8_t *) colorLoopTime,
                                      sizeof(*colorLoopTime));
}`

 #### Proposed Solution
* Set attributes in accessor function to default value in case ember read fails

bzbarsky-apple, 2021-08-13:
@vivien-apple

turon, 2022-02-01:
Zap support for spec defined default values required for v1.
Removing tag `v1_triage_split_4`

bzbarsky-apple, 2022-02-01:
`GetColorLoopTime` no longer exists.

It's not clear to me what this issue is about, at this point....

closed 2022-02-07 by woody-apple