project-chip/connectedhomeip #9017: Sort out what is going on with InitEntropy and RandUtils

bzbarsky-apple, 2021-08-13:
#### Problem
We have InitEntropy methods that call `srand` but are never called, and we have `RandUtils`, which calls `rand` and seems like a footgun waiting to happen as a result; see https://github.com/project-chip/connectedhomeip/pull/8989

 #### Proposed Solution
Figure out what should be going on here.  Either remove RandUtils, or make sure we call `srand` as needed, or make RandUtils use our DRBG.

@tcarmelveilleux @pan-apple @jpk233

tcarmelveilleux, 2021-08-13:
`InitEntropy()` should be called in early system init, before/during `GenericPlatformManagerImpl<ImplClass>::_InitChipStack()`

I don't know why it's not called, but it's not good.

bzbarsky-apple, 2021-09-11:
Fixed by #9420

closed 2021-09-11 by bzbarsky-apple