project-chip/connectedhomeip #9035: chip-tool pairing ble-wifi Failure

GoldSora, 2021-08-16:
~/Desktop/connectedhomeip/examples/chip-tool/out/debug$ **./chip-tool pairing ble-wifi LONGAN rockysheep057 0 20202021 3840**
[1629098589.680014][150721:150721] CHIP:CTL: Read local id 0x000000000001B669, remote id 0x0000000000BC5C01
[1629098589.680685][150721:150721] CHIP:CTL: Generating NOC
[1629098589.680820][150721:150721] CHIP:CTL: Generating ICAC
[1629098589.685534][150721:150721] CHIP:IN: UDP::Init bind&listen port=5541
[1629098589.685615][150721:150721] CHIP:IN: UDP::Init bind&listen port=5541
[1629098589.685641][150721:150721] CHIP:IN: TransportMgr initialized
[1629098589.685653][150721:150721] CHIP:DIS: Init fabric pairing table with server storage
[1629098589.685916][150721:150721] CHIP:ZCL: Using ZAP configuration...
[1629098589.685975][150721:150721] CHIP:ZCL: deactivate report event
[1629098589.686277][150721:150721] CHIP:DIS: Added new fabric at index: 1, Initialized: 1
[1629098589.686326][150721:150721] CHIP:DIS: Assigned fabric ID: 0x0000000000000000, node ID: 0x000000000001B669
[1629098589.686339][150721:150721] CHIP:CTL: Joined new fabric at index 1
[1629098589.686704][150721:150726] CHIP:DL: CHIP task running
[1629098589.686927][150721:150726] CHIP:SC: Assigned local session key ID 12
[1629098589.687037][150721:150726] CHIP:IN: Message appended to BLE send queue
[1629098589.687075][150721:150726] CHIP:SC: Sent PBKDF param request
[1629098589.687892][150721:150727] CHIP:DL: TRACE: Bluez mainloop starting Thread
[1629098589.688785][150721:150724] CHIP:DL: TRACE: Bus acquired for name C-4cc1
[1629098589.694797][150721:150726] CHIP:DL: PlatformBlueZInit init success
[1629098589.702715][150721:150724] CHIP:BLE: BLE removing known devices.
[1629098589.702799][150721:150724] CHIP:BLE: BLE initiating scan.
[1629098590.936984][150721:150724] CHIP:BLE: Device BC:7F:A4:33:26:BC does not look like a CHIP device.
[1629098592.652116][150721:150724] CHIP:BLE: Device 50:EB:71:FA:D2:95 does not look like a CHIP device.
[1629098599.707401][150721:150724] CHIP:IN: Clearing BLE pending packets.
[1629098599.707465][150721:150724] CHIP:IN: BleConnection Error: ../../third_party/connectedhomeip/src/platform/Linux/BLEManagerImpl.cpp:821: CHIP Error 0x00000032: Timeout
[1629098619.692969][150721:150726] CHIP:TOO: Pairing Failure: ../../third_party/connectedhomeip/src/controller/CHIPDeviceController.cpp:1487: CHIP Error 0x00000032: Timeout
[1629098619.693134][150721:150726] CHIP:SC: PASESession timed out while waiting for a response from the peer. Expected message type was 33
**[1629098619.693148][150721:150726] CHIP:TOO: Secure Pairing Failed
[1629098619.693430][150721:150726] CHIP:TOO: Pairing Failure: ../../third_party/connectedhomeip/src/protocols/secure_channel/PASESession.cpp:333: CHIP Error 0x00000032: Timeout
[1629098619.693842][150721:150721] CHIP:TOO: Run command failure: ../../third_party/connectedhomeip/src/protocols/secure_channel/PASESession.cpp:333: CHIP Error 0x00000032: Timeout**
[1629098619.694494][150721:150721] CHIP:CTL: Shutting down the commissioner
[1629098619.694546][150721:150721] CHIP:CTL: Shutting down the controller
[1629098619.694575][150721:150721] CHIP:DL: Inet Layer shutdown
[1629098619.694634][150721:150721] CHIP:DL: BLE layer shutdown
[1629098619.694750][150721:150721] CHIP:DL: System Layer shutdown

tewarid, 2021-08-25:
Did you per chance try the same command a second time? I've seen intermittent failure during the first attempt at pairing.

bekencorp, 2022-01-07:
@GoldSora Did you resolve this issue?

tewarid, 2022-01-07:
Issue has resurfaced on macOS monterey 12.1 as #13303.

woody-apple, 2022-01-26:
Does look like a dup of #13303

closed 2022-01-26 by woody-apple