project-chip/connectedhomeip #9140: Stack overflow in m5stack upon factory reset

pan-apple, 2021-08-19:
#### Problem
The m5stack device is crashing due to stack overflow on factory reset.
The device eventually recovers, but that's most likely a fluke (i.e. the stack could still be overflowing, but maybe the memory may not be overwritten yet).

```
I (2276) chip[DIS]: Start dns-sd server - no current nodeId
V (2296) chip[DIS]: Using wifi MAC for hostname
V (2296) chip[D
***ERROR*** A stack overflow in task main has been detected.

Backtrace:0x4009189a:0x3ffbb450 0x40091fc5:0x3ffbb470 0x40095462:0x3ffbb490 0x40093bc4:0x3ffbb510 0x400920bc:0x3ffbb530 0x4009206e:0x3ffbb52c |<-CORRUPTED
0x4009189a: panic_abort at ../tools/esp-idf/components/esp_system/panic.c:367
0x40091fc5: esp_system_abort at ../tools/esp-idf/components/esp_system/system_api.c:112
0x40095462: vApplicationStackOverflowHook at ../tools/esp-idf/components/freertos/port/xtensa/port.c:489
0x40093bc4: vTaskSwitchContext at ../tools/esp-idf/components/freertos/tasks.c:3289
0x400920bc: _frxt_dispatch at ../tools/esp-idf/components/freertos/port/xtensa/portasm.S:432
0x4009206e: _frxt_int_exit at ../tools/esp-idf/components/freertos/port/xtensa/portasm.S:231
```

 #### Proposed Solution
`acb4ddf1108a4735bbf0df40aebaf3d1af9194a5` is able to boot up fine.
`0bdeccb54bfaeca043e47f854c65e19297160a1e` is showing the issue.
There are a few intermediate commits, which did not build for m5stack. So I wasn’t able to narrow it down to a particular commit.

cecille, 2021-08-19:
Backtrace:
0x4009095e: panic_abort at /opt/espressif/esp-idf/components/esp_system/panic.c:341
0x40090f35: esp_system_abort at /opt/espressif/esp-idf/components/esp_system/system_api.c:106
0x4009765e: abort at /opt/espressif/esp-idf/components/newlib/abort.c:46
0x4013915a: esp_sha_read_digest_state at /opt/espressif/esp-idf/components/mbedtls/port/esp32/sha.c:272
0x40136a62: mbedtls_sha256_finish_ret at /opt/espressif/esp-idf/components/mbedtls/port/esp32/esp_sha256.c:395
0x400eaf9c: chip::Crypto::Hash_SHA256_stream::Finish(chip::Span<unsigned char>&) at /workspaces/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/crypto/CHIPCryptoPALmbedTLS.cpp:261
0x400ef0e1: chip::AdditionalDataPayloadGenerator::generateRotatingDeviceId(unsigned short, char const*, unsigned int, char*, unsigned int, unsigned int&) at /workspaces/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/setup_payload/AdditionalDataPayloadGenerator.cpp:101
0x400d7fa2: chip::app::Mdns::GenerateRotatingDeviceId(char*, unsigned int) at /workspaces/connectedhomeip/examples/all-clusters-app/esp32/build/../third_party/connectedhomeip/src/app/server/Mdns.cpp:292
0x400d816a: chip::app::Mdns::Advertise(bool) at /workspaces/connectedhomeip/examples/all-clusters-app/esp32/build/../third_party/connectedhomeip/src/app/server/Mdns.cpp:194
0x400d8295: chip::app::Mdns::AdvertiseCommissionableNode() at /workspaces/connectedhomeip/examples/all-clusters-app/esp32/build/../third_party/connectedhomeip/src/app/server/Mdns.cpp:131
0x400d82c8: chip::app::Mdns::StartServer() at /workspaces/connectedhomeip/examples/all-clusters-app/esp32/build/../third_party/connectedhomeip/src/app/server/Mdns.cpp:265
0x400e10d8: DeviceCallbacks::DeviceEventCallback(chip::DeviceLayer::ChipDeviceEvent const*, int) at /workspaces/connectedhomeip/examples/all-clusters-app/esp32/build/../main/DeviceCallbacks.cpp:71
0x401a006b: chip::DeviceManager::CHIPDeviceManager::CommonDeviceEventHandler(chip::DeviceLayer::ChipDeviceEvent const*, int) at /workspaces/connectedhomeip/examples/all-clusters-app/esp32/build/../main/CHIPDeviceManager.cpp:47
0x401a0d25: chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::DispatchEventToApplication(chip::DeviceLayer::ChipDeviceEvent const*) at /workspaces/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl.cpp:285 (discriminator 2)
0x400f38ab: chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::_DispatchEvent(chip::DeviceLayer::ChipDeviceEvent const*) at /workspaces/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl.cpp:233
0x400f3aa3: chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::_RunEventLoop() at /workspaces/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/include/platform/PlatformManager.h:363
 (inlined by) chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::_RunEventLoop() at /workspaces/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.cpp:179
0x400f3acd: chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::EventLoopTaskMain(void*) at /workspaces/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/include/platform/PlatformManager.h:295
 (inlined by) chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::EventLoopTaskMain(void*) at /workspaces/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.cpp:20
0x40090f3d: vPortTaskWrapper at /opt/espressif/esp-idf/components/freertos/xtensa/port.c:143

andy31415, 2021-08-20:
Could we disable rotating device id for now? I thought it is an optional feature.

pan-apple, 2021-08-20:
I'll try it out.

pan-apple, 2021-08-20:
I tried disabling the rotating device id by setting `chip_enable_rotating_device_id` to `false`. It didn't solve the issue.

cecille, 2021-08-20:
I also have a device that was bootlooping on latest master. for me, setting chip_enable_rotating_device_id to false caused linker errors when compiling the all-clusters-app although I'm not sure where the link errors are coming from (looks like a staticly linked lib from esp). Regardless, if I just comment out the part in Mdns.cpp where the rotating device ID is generated, I no longer get boot loops.

mspang, 2021-08-23:
This started with #8888

mspang, 2021-08-23:
Here's another stack

```
abort() was called at PC 0x401388ae on core 0
0x401388ae: esp_sha_read_digest_state at /home/spang/tools/esp-idf/components/mbedtls/port/esp32/sha.c:272

Backtrace:0x400908da:0x3ffbb5c0 0x40090ec1:0x3ffbb5e0 0x40097562:0x3ffbb600 0x401388ae:0x3ffbb670 0x4013621e:0x3ffbb690 0x400eaf94:0x3ffbb6c0 0x400ef11e:0x3ffbb6e0 0x400d811a:0x3ffbb7e0 0x400d82c2:0x3ffbb840 0x400d842d:0x3ffbbb50 0x400d850e:0x3ffbbb70 0x400e2d76:0x3ffbbbd0 0x400d8bc5:0x3ffbbc30 0x400d8e7d:0x3ffbbd30 0x400d6d2b:0x3ffbbd90 0x400d3f2c:0x3ffbbfb0
0x400908da: panic_abort at /home/spang/tools/esp-idf/components/esp_system/panic.c:341

0x40090ec1: esp_system_abort at /home/spang/tools/esp-idf/components/esp_system/system_api.c:106

0x40097562: abort at /home/spang/tools/esp-idf/components/newlib/abort.c:46

0x401388ae: esp_sha_read_digest_state at /home/spang/tools/esp-idf/components/mbedtls/port/esp32/sha.c:272

0x4013621e: mbedtls_sha256_finish_ret at /home/spang/tools/esp-idf/components/mbedtls/port/esp32/esp_sha256.c:395

0x400eaf94: chip::Crypto::Hash_SHA256_stream::Finish(chip::Span<unsigned char>&) at /ssd/src/connectedhomeip5/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/crypto/CHIPCryptoPALmbedTLS.cpp:265

0x400ef11e: chip::AdditionalDataPayloadGenerator::generateRotatingDeviceId(unsigned short, char const*, unsigned int, char*, unsigned int, unsigned int&) at /ssd/src/connectedhomeip5/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/setup_payload/AdditionalDataPayloadGenerator.cpp:101

0x400d811a: chip::app::Mdns::GenerateRotatingDeviceId(char*, unsigned int) at /ssd/src/connectedhomeip5/examples/all-clusters-app/esp32/build/../third_party/connectedhomeip/src/app/server/Mdns.cpp:339

0x400d82c2: chip::app::Mdns::Advertise(bool, chip::app::Mdns::CommissioningMode) at /ssd/src/connectedhomeip5/examples/all-clusters-app/esp32/build/../third_party/connectedhomeip/src/app/server/Mdns.cpp:206

0x400d842d: chip::app::Mdns::AdvertiseCommissionableNode(chip::app::Mdns::CommissioningMode) at /ssd/src/connectedhomeip5/examples/all-clusters-app/esp32/build/../third_party/connectedhomeip/src/app/server/Mdns.cpp:268

0x400d850e: chip::app::Mdns::StartServer(chip::app::Mdns::CommissioningMode) at /ssd/src/connectedhomeip5/examples/all-clusters-app/esp32/build/../third_party/connectedhomeip/src/app/server/Mdns.cpp:309

0x400e2d76: chip::RendezvousServer::WaitForPairing(chip::RendezvousParameters const&, unsigned int, chip::Span<unsigned char const> const&, unsigned short, chip::Messaging::ExchangeManager*, chip::TransportMgrBase*, chip::SecureSessionMgr*) at /ssd/src/connectedhomeip5/examples/all-clusters-app/esp32/build/../third_party/connectedhomeip/src/app/server/RendezvousServer.cpp:99

0x400d8bc5: OpenBasicCommissioningWindow(chip::ResetFabrics, unsigned short, chip::PairingWindowAdvertisement) at /ssd/src/connectedhomeip5/examples/all-clusters-app/esp32/build/../third_party/connectedhomeip/src/app/server/Server.cpp:281

0x400d8e7d: InitServer(AppDelegate*) at /ssd/src/connectedhomeip5/examples/all-clusters-app/esp32/build/../third_party/connectedhomeip/src/app/server/Server.cpp:412

0x400d6d2b: app_main at /ssd/src/connectedhomeip5/examples/all-clusters-app/esp32/build/../main/main.cpp:637

0x400d3f2c: main_task at /home/spang/tools/esp-idf/components/esp32/cpu_start.c:609

```

dhrishi, 2021-08-29:
Please give a try with the ToT. This should fix it https://github.com/project-chip/connectedhomeip/pull/9263

mspang, 2021-09-08:
This is still an issue.

shubhamdp, 2022-07-14:
Tested on 37bb2df9941fcff64657cc9a0ca25ad176aeb42a
example: all-clusters-app/esp32

Commissioned the device using chip-tool: `./chip-tool pairing ble-wifi 0x2222 (SSID) (PASSPHRASE) 20202021 3840`
Performed some cluster control.
Factory reset the device using ESP32 console `> matter device factoryreset`

No crash were seen and device rebooted and I was able to commission and could perform cluster control.

closed 2023-01-20 by stale[bot]