project-chip/connectedhomeip #9141: mDNS advertisements are not seen on the network

pan-apple, 2021-08-19:
#### Problem
Steps
1. Commission the device (m5stack in this case)
2. The device adds mDNS advertisement correctly, and can be seen by running `dns-sd -B _matter._tcp`
3. Open the commissioning window, and commission the second fabric
4. The device gets the operational identity, and claims to start advertisement for both fabrics. However, `dns-sd -B _matter._tcp` doesn't show the advertisements.
5. Reboot the device. On boot up, it again claims to start advertisement for both fabrics. Now `dns-sd -B _matter._tcp` can see the advertisements.

After step 2, `dns-sd -B _matter._tcp` shows the following update.
```
12:46:08.219  Add        2  11 local.               _matter._tcp.        0000000000000000-000000000000000E
```

After step 4, these are the logs on the device.
```
V (103085) chip[DIS]: Using wifi MAC for hostname
I (103085) chip[DIS]: Advertise operational node 0000000000000000-000000000000000E
I (103085) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'.
V (103105) chip[DIS]: Using wifi MAC for hostname
I (103115) chip[DIS]: Advertise operational node 0000000000000000-8499EEDE4D510E5C
I (103115) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'.
```

But, `dns-sd -B _matter._tcp` is showing no updates.

On Reboot, the device logs look pretty much the same
```
V (8395) chip[DIS]: Using wifi MAC for hostname
I (8395) chip[DIS]: Advertise operational node 0000000000000000-000000000000000E
I (8395) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'.
V (8415) chip[DIS]: Using wifi MAC for hostname
I (8425) chip[DIS]: Advertise operational node 0000000000000000-8499EEDE4D510E5C
I (8425) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'.
```

And, `dns-sd -B _matter._tcp` is showing the following.
```
12:57:59.919  Add        3  11 local.               _matter._tcp.        0000000000000000-000000000000000E
12:57:59.919  Add        2  11 local.               _matter._tcp.        0000000000000000-8499EEDE4D510E5C
```

 #### Proposed Solution
Debug and fix. This is needed for multi-admin functionality to work.

woody-apple, 2022-01-26:
@pan-apple ?

woody-apple, 2022-02-07:
This appears to be working now.

closed 2022-02-07 by woody-apple