project-chip/connectedhomeip #9173: Controller/BUILD: Cannot add controller tests for certain platforms (Darwin, clang)

sharadb-amazon, 2021-08-20:
#### Problem

This PR (#8824) attempts to add tests for src/controller: https://github.com/project-chip/connectedhomeip/pull/8824/files#diff-8b9a0a48a2fb362c82aef9724b90a5476c4b80358995f69c83324546c0c96648

This builds with gcc, but fails for clang, Darwin with the following errors:

`Undefined symbols for architecture x86_64:
  "chip::app::ReadSingleClusterData(chip::app::ClusterInfo&, chip::TLV::TLVWriter*, bool*)", referenced from:
      chip::app::reporting::Engine::RetrieveClusterData(chip::app::AttributeDataElement::Builder&, chip::app::ClusterInfo&) in libCHIPDataModel.a(libCHIPDataModel.Engine.cpp.o)
  "chip::app::ServerClusterCommandExists(unsigned int, unsigned int, unsigned short)", referenced from:
      chip::app::CommandHandler::ProcessCommandDataElement(chip::app::CommandDataElement::Parser&) in libCHIPDataModel.a(libCHIPDataModel.CommandHandler.cpp.o)
  "chip::app::DispatchSingleClusterCommand(unsigned int, unsigned int, unsigned short, chip::TLV::TLVReader&, chip::app::CommandHandler*)", referenced from:
      chip::app::CommandHandler::ProcessCommandDataElement(chip::app::CommandDataElement::Parser&) in libCHIPDataModel.a(libCHIPDataModel.CommandHandler.cpp.o)
  "chip::app::DispatchSingleClusterResponseCommand(unsigned int, unsigned int, unsigned short, chip::TLV::TLVReader&, chip::app::CommandSender*)", referenced from:
      chip::app::CommandSender::ProcessCommandDataElement(chip::app::CommandDataElement::Parser&) in libCHIPDataModel.a(libCHIPDataModel.CommandSender.cpp.o)
  "chip::app::WriteSingleClusterData(chip::app::ClusterInfo&, chip::TLV::TLVReader&, chip::app::WriteHandler*)", referenced from:
      chip::app::WriteHandler::ProcessAttributeDataList(chip::TLV::TLVReader&) in libCHIPDataModel.a(libCHIPDataModel.WriteHandler.cpp.o)
ld: symbol(s) not found for architecture x86_64
`

Expectation is that src/controller/tests can be built and run for all platforms.

 #### Proposed Solution
TBD

mspang, 2021-08-20:
@yunhanw-google Why do we have these undefined symbols?

erjiaqing, 2021-08-24:
You need to add these files to XCode project.

sharadb-amazon, 2021-08-24:
Not sure if I understood. The build failure happens for clang as well (if I include the controller tests for all platforms), not just Darwin. Which files/libs need to be added to the XCode project? Can you explain why that would fix this problem?

erjiaqing, 2021-08-24:
> Not sure if I understood. The build failure happens for clang as well (if I include the controller tests for all platforms), not just Darwin. Which files/libs need to be added to the XCode project? Can you explain why that would fix this problem?

Seems I misunderstood your need, can you try adding `${chip_root}/src/controller/data_model` to `src/controller/tests/BUILD.gn`

sharadb-amazon, 2021-08-24:
Tried that on a brand with [this change](https://github.com/sharadb-amazon/connectedhomeip/commit/764db2e19c0c165e8ef6127cb0783a6ae0fa66e5). Build on Darwin succeeds, but then all-clusters-app build starts to fail on esp32 with the following type of error. Full logs: https://github.com/sharadb-amazon/connectedhomeip/runs/3413744436 This happens only when I make the above change.

> ../../../../../../config/esp32/third_party/connectedhomeip/src/controller/data_model/zap-generated/CHIPClientCallbacks.cpp: In function 'void ApplicationLauncherClusterApplicationLauncherListListAttributeFilter(chip::TLV::TLVReader*, chip::Callback::Cancelable*, chip::Callback::Cancelable*)':
> ../../../../../../config/esp32/third_party/connectedhomeip/src/controller/data_model/zap-generated/CHIPClientCallbacks.cpp:376:6: error: stack usage might be unbounded [-Werror=stack-usage=]
>  void ApplicationLauncherClusterApplicationLauncherListListAttributeFilter(TLV::TLVReader * tlvData,
> ...
> ...

mspang, 2021-08-24:
Hold on, why would we need to add src/controller/data_model ?

bzbarsky-apple, 2021-09-04:
It looks like #8824 merged.  @sharadb-amazon is there still a problem here?  If so, what are the steps to reproduce?

bzbarsky-apple, 2021-09-11:
It looks like #8824 merged by just not building these tests on various OSes, and #9374 disabled them on still others where they were failing.

Fixing the build on Darwin here is pretty simple: just use the right spelling of `"darwin"` in the `chip_device_platform` tests in `src/controller/tests/BUILD.gn` (and why are there platform-specific bits here to start with???).  But then the test crashes:
```
    frame #0: 0x00007fff67383143 libdispatch.dylib`dispatch_source_create + 167
  * frame #1: 0x00000001000c6c9c TestDevice`chip::Inet::UDPEndPoint::Bind(this=0x000000010012fbd8, addrType=kIPAddressType_IPv4, addr=0x000000010012e6c0, port=5540, intfId=0) at UDPEndPoint.cpp:254:27
   ...
frame #1: 0x00000001000c6c9c TestDevice`chip::Inet::UDPEndPoint::Bind(this=0x000000010012fbd8, addrType=kIPAddressType_IPv4, addr=0x000000010012e6c0, port=5540, intfId=0) at UDPEndPoint.cpp:254:27
   251 	    {
   252 	        unsigned long fd = static_cast<unsigned long>(mSocket);
   253 	
-> 254 	        mReadableSource = dispatch_source_create(DISPATCH_SOURCE_TYPE_READ, fd, 0, dispatchQueue);
   255 	        ReturnErrorCodeIf(mReadableSource == nullptr, CHIP_ERROR_NO_MEMORY);
   256 	
   257 	        dispatch_source_set_event_handler(mReadableSource, ^{
```
where `dispatchQueue` is a garbage value because this test never initialized the platform manager.

closed 2021-09-13 by bzbarsky-apple