project-chip/connectedhomeip #9176: Compilation failure (libChipCrypto) while compiling examples/lock-app/cc13x2x7_26x2x7

kaushik6, 2021-08-21:
#### Problem
Getting compilation error when i try to compile examples/lock-app/cc13x2x7_26x2x7. Followed instructions in https://github.com/project-chip/connectedhomeip/blob/master/examples/lock-app/cc13x2x7_26x2x7/README.md
Environment: Ubuntu 20.04 LTS

Command:
```
$ cd ~/connectedhomeip/examples/lock-app/cc13x2x7_26x2x7
$ export TI_SIMPLELINK_SDK_ROOT=$HOME/ti/simplelink_cc13x2_26x2_sdk_4_40_05_02_eng
$ export TI_SYSCONFIG_ROOT=$HOME/ti/sysconfig_1.7.0
$ gn gen out/debug --args="ti_simplelink_sdk_root=\"${TI_SIMPLELINK_SDK_ROOT}\" ti_sysconfig_root=\"${TI_SYSCONFIG_ROOT}\""
$ ninja -C out/debug
```

Error:
```
../../third_party/connectedhomeip/src/crypto/CHIPCryptoPALmbedTLS.cpp:215:45: error: static assertion failed: kMAX_Hash_SHA256_Context_Size is too small for the size of underlying mbedtls_sha256_context
  215 | static_assert(kMAX_Hash_SHA256_Context_Size >= sizeof(mbedtls_sha256_context),
      |               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ../../third_party/connectedhomeip/src/crypto/CHIPCryptoPALmbedTLS.cpp:48:
../../third_party/connectedhomeip/src/lib/support/SafePointerCast.h: In instantiation of 'std::enable_if_t<std::is_pointer<_Tp>::value, ToPtr> chip::SafePointerCast(From*) [with ToPtr = mbedtls_sha256_context*; From = chip::Crypto::HashSHA256OpaqueContext; std::enable_if_t<std::is_pointer<_Tp>::value, ToPtr> = mbedtls_sha256_context*]':
../../third_party/connectedhomeip/src/crypto/CHIPCryptoPALmbedTLS.cpp:220:61:   required from here
../../third_party/connectedhomeip/src/lib/support/SafePointerCast.h:36:32: error: static assertion failed: Casting to a bigger type
   36 |     static_assert(sizeof(From) >= sizeof(To), "Casting to a bigger type");
      |                   ~~~~~~~~~~~~~^~~~~~~~~~~~~
```

tcarmelveilleux, 2021-08-21:
What is the value for `sizeof(mbedtls_sha256_context)` on your platform? The error is clear: the underlying impl is using a custom context that appears larger than what is accounted-for.

Raqbit, 2021-09-06:
I can reproduce this error, `sizeof(mbedtls_sha256_context) == 296`. The sha256 implementation is swapped by TI for hardware acceleration.